4IceG/luci-app-3ginfo-lite #1: add support for Huawei e5372 please

lukjod, 2021-07-18:
cat /sys/kernel/debug/usb/devices
`T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.04
S:  Manufacturer=Linux 5.4.128 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.0.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  3 Spd=480  MxCh= 0
D:  Ver= 2.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=12d1 ProdID=1506 Rev= 1.02
S:  Manufacturer=HUAWEI_MOBILE
S:  Product=HUAWEI_MOBILE
C:* #Ifs= 3 Cfg#= 1 Atr=c0 MxPwr=  2mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=02 Prot=12 Driver=option
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 1 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=02 Prot=16 Driver=huawei_cdc_ncm
E:  Ad=84(I) Atr=03(Int.) MxPS=  16 Ivl=2ms
I:* If#= 1 Alt= 1 #EPs= 3 Cls=ff(vend.) Sub=02 Prot=16 Driver=huawei_cdc_ncm
E:  Ad=84(I) Atr=03(Int.) MxPS=  16 Ivl=2ms
E:  Ad=83(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=125us

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.04
S:  Manufacturer=Linux 5.4.128 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.0.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.04
S:  Manufacturer=Linux 5.4.128 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.04
S:  Manufacturer=Linux 5.4.128 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms`

4IceG, 2021-07-18:
@lukjod show what you get after executing AT commands:
ATI
AT ^ HCSQ?
AT ^ CHIPTEMP =?
AT ^ CHIPTEMP?
It may be that commands except "ATI" are blocked and nothing will be read, there will only be an "ERROR" message.

lukjod, 2021-07-18:
ATI
Manufacturer: huawei
Model: E5786s-32a
Revision: 21.313.17.00.00
IMEI: 864417020920360
+GCAP: +CGSM,+DS,+ES

AT ^ HCSQ?
ERROR

AT ^ CHIPTEMP =?
ERROR

AT ^ CHIPTEMP?
ERROR

no, usb1 is not valid.
Anything else that I can do to make it work?

4IceG, 2021-07-18:
We will do nothing if the modem does not respond to commands, or it is
blocked or the commands have changed. I can't do anything more at the
moment. I don't know if it can be unlocked by command or by other software.

4IceG, 2021-07-19:
@lukjod Have you tested the normal 3ginfo, does it show something? It is more universal, complex, so maybe it will show something more.

lukjod, 2021-07-19:
Nope, nothing beside basic ATI response. Will try to search for valid at commands on net, maybe just different logic of sentence

closed 2021-07-20 by 4IceG

4IceG, 2021-08-16:
@lukjod  jak będziesz miał chwilę to zerknij czy nowa wersja pakietu wypluje coś dla Twojego e5786.

4IceG, 2021-08-18:
@lukjod jak będziesz miał chwilę to zerknij czy nowa wersja pokaże wersję FW i pełniejszą nazwę modemu.

lukjod, 2021-08-19:
@4IceG ta nic nie pokazuje przy takich samych ustawieniach jak poprzednio

4IceG, 2021-08-20:
No to lipa, tak to jest robic wsparcie na sucho bez fizycznego sprzetu :).
Jeszcze zerkne na to dzis, moze ja gdzies sie pomylilem w logice/kodzie, a
jak nie no to zostanie to co jest..

4IceG, 2021-08-20:
Wygląda na to że skrypty z założenia są poprawne, jeszcze trochę je zmianiłem, może teraz złapią więcej. @lukjod pokaż mi jak to aktualnie będzie wyglądało (zrób screena). Wyczyść przed zdjęciem cache przeglądarki i cache Luci z tmp. Ostatnie podejście.

lukjod, 2021-08-20:
tak jak poprzednio, nie dziala, znacznie zwieksza obciazenie procesora i tworzy mnostwo kolo 20 plikow  tmp.xxxxxx w /tmp. Wyraznie mu ta wersja nie sluzy. wracam do ostatniej dzialajacej z 18.06.2021

4IceG, 2021-08-20:
No nic probowalem. Jeszcze wgram u siebie wieczorem i zerkne na to
obciazenie procesora.

Swoja droga na tym to polega ze tworzone sa pliki w tmp, z ktorych
pobierane sa dane. Posprawdzam to jeszcze pozniej.