4IceG/luci-app-3ginfo-lite #3: Model unknown after firmware upgrade...

erdoukki, 2021-08-03:
Before upgrade : 
```
Modem Information
Modem type:
Quectel EP06
Revision / Firmware:
EP06ELAR03A08M4G
Communication Port:
/dev/ttyUSB2
Protocol:
QMI
```

After firmware upgrade with Quectel QFirehose, the model is garbage : 
```
Modem Information
Modem type:
ision: EP06ELAR04A04M4G
Revision / Firmware:
EP06ELAR04A04M4G
Communication Port:
/dev/ttyUSB2
Protocol:
QMI
```

How can I help to fix this ?

erdoukki, 2021-08-03:
`# /usr/share/3ginfo-lite/3ginfo.sh `
```
{
"modem":"ision: EP06ELAR04A04M4G",
"mtemp":"",
"firmware":"EP06ELAR04A04M4G",
"cport":"/dev/ttyUSB2",
"protocol":"QMI",
"csq":"-",
"signal":"0",
"operator_name":"-",
"operator_mcc":"-",
"operator_mnc":"-",
"mode":"No Service /",
"registration":"0",
"lac_dec":"0",
"lac_hex":"0",
"cid_dec":"0",
"cid_hex":"0",
"rsrp":"",
"rsrq":"",
"rssi":"-",
"sinr":"",
"addon":[]
}
```

erdoukki, 2021-08-03:
fixed by modifying the file : 2c7c0306
`# cat /usr/share/3ginfo-lite/3ginfo-addon/2c7c0306 | grep MODEL`
`MODEL=$(echo $DEVV | awk -F 'ati|Rev' '{print $1}'| xargs)`

erdoukki, 2021-08-03:
# /usr/share/3ginfo-lite/3ginfo.sh 
```
{
"modem":"Quectel EP06",
"mtemp":"",
"firmware":"EP06ELAR04A04M4G",
"cport":"/dev/ttyUSB2",
"protocol":"QMI",
"csq":"-",
"signal":"0",
"operator_name":"-",
"operator_mcc":"-",
"operator_mnc":"-",
"mode":"No Service /",
"registration":"0",
"lac_dec":"0",
"lac_hex":"0",
"cid_dec":"0",
"cid_hex":"0",
"rsrp":"",
"rsrq":"",
"rssi":"-",
"sinr":"",
"addon":[]
}
```

4IceG, 2021-08-03:
They warned me against uploading the firmware from the firehouse directory, I do not know what mode the modem enters then, but you can see that it does not work normally / correctly. It is best to post on the modem manufacturer's forum how to restore the modem operation.

erdoukki, 2021-08-03:
For me there is no problem with the card itself...
`# sms_tool -d /dev/ttyUSB2 at "ati"`
```

Quectel
EP06
Revision: EP06ELAR04A04M4G

```

erdoukki, 2021-08-03:
The upgrade process work fine and finished without errors...
But the code testing the model may look strange, print $2 is not correct for me and it works fine with print $1

4IceG, 2021-08-03:
If the modem works then OK, so other commands collecting data from the modem should work. Tomorrow I will check and correct the script from EP06-E to make it work well for the ATI command.

closed 2021-08-03 by 4IceG

4IceG, 2021-08-05:
> 
> 
> For me there is no problem with the card itself...
> `# sms_tool -d /dev/ttyUSB2 at "ati"`
> 
> ```
> 
> Quectel
> EP06
> Revision: EP06ELAR04A04M4G
> ```

The correct output of the "ati" command for this modem is:
"ati
Quectel
EP06
Revision: EP06ELAR01A02M4G "

If we add this command output to the script
"# Modem
# OF = $ (sms_tool -d $ DEVICE at "ati")
OF = "ati
Quectel
EP06
Revision: EP06ELAR01A02M4G ""
it is all read well, nothing needs to be edited or corrected.

erdoukki, 2021-08-09:
This was a correct display with old firmware, the latest shown i differently, so to get the always working we may modify the file as this way : 
```
# Modem
OF=$(sms_tool -d $DEVICE at "ati")
DEVV=$(echo $OF | tr -s "\n"  | xargs)
FW=$(echo $DEVV | awk -F 'Revision:' '{print $2}'| xargs)
MODEL=$(echo $DEVV | awk -F 'ati|Revision:' '{print $1}'| xargs)

```
then : 
```
root@ULTRA-5G:~# /usr/share/3ginfo-lite/3ginfo.sh 
{
"modem":"Quectel EP06",
"mtemp":"",
"firmware":"EP06ELAR04A04M4G",
"cport":"/dev/ttyUSB2",
"protocol":"MBIM",
"csq":"14",
"signal":"45",
"operator_name":"Free",
"operator_mcc":"208",
"operator_mnc":"15",
"mode":"LTE B7 (2600 MHz)",
"registration":"0",
"lac_dec":"0",
"lac_hex":"0",
"cid_dec":"0",
"cid_hex":"0",
"rsrp":"-113",
"rsrq":"-11",
"rssi":"-82",
"sinr":"13",
"addon":[{"RSRP":"-113 dBm"},{"RSRQ":"-11 dB"},{"RSSI":"-82 dBm"},{"SINR":"13 dB"}]
}

```

4IceG, 2021-08-09:
@erdoukki Check if such a record will work?
```
# Modem
OF=$(sms_tool -d $DEVICE at "ati")
DEVV=$(echo $OF | tr -s "\n"  | xargs)
FW=$(echo $DEVV | awk -F 'Revision:' '{print $2}'| xargs)
MODEL=$(echo $DEVV | awk -F 'ati|Revision:' '{print $1}'| xargs)
if [ -z "$MODEL" ]
then
      MODEL=$(echo $DEVV | awk -F 'ati|Revision:' '{print $2}'| xargs)
fi
```

erdoukki, 2021-08-11:
> I have a second Quectel EP06, untouched (not upgraded), and will test also on it...

root@ultra:~# export DEVICE=/dev/ttyUSB2
root@ultra:~# # Modem
root@ultra:~# OF=$(sms_tool -d $DEVICE at "ati")
root@ultra:~# DEVV=$(echo $OF | tr -s "\n"  | xargs)
root@ultra:~# FW=$(echo $DEVV | awk -F 'Revision:' '{print $2}'| xargs)
root@ultra:~# MODEL=$(echo $DEVV | awk -F 'ati|Revision:' '{print $1}'| xargs)
root@ultra:~# if [ -z "$MODEL" ]
> then
>       MODEL=$(echo $DEVV | awk -F 'ati|Revision:' '{print $2}'| xargs)
> fi
root@ultra:~# echo $MODEL
 EP06el

erdoukki, 2021-08-11:
look worrect with : 
```
sms_tool -d $DEVICE at "ati"| tr -d "\r"|xargs
```
```
# Modem
OF=$(sms_tool -d $DEVICE at "ati"| tr -d "\r"|xargs)
DEVV=$(echo $OF | tr -s "\n"  | xargs)
FW=$(echo $DEVV | awk -F 'Revision:' '{print $2}'| xargs)
MODEL=$(echo $DEVV | awk -F 'ati|Revision:' '{print $1}'| xargs)
if [ -z "$MODEL" ]
then
      MODEL=$(echo $DEVV | awk -F 'ati|Revision:' '{print $2}'| xargs)
fi
```
both give : 
```
root@ultra:~# echo $MODEL
Quectel EP06
```
```
root@ULTRA-5G:~# echo $MODEL
Quectel EP06
```