4IceG/luci-app-3ginfo-lite #4: luasrc?

kyliezero, 2021-12-29:
I was wondering, why does this package not appear in the LuCi menu, it turns out that there is no "luasrc" file from this package, do you really not need it?

4IceG, 2021-12-29:
Hi, the package is written for the current Luci JS so there won't be any
"luasrc" files there because the package doesn't need them. Log out, clear
browser cache or restart the router to refresh the cache. Everything should
work if Openwrt is up to date and the dependencies are satisfied.

kyliezero, 2021-12-29:
here I use lede, are there other dependants that need to be installed, so that it can work smoothly..

4IceG, 2021-12-30:
Lede is an old system already, I don't know what modem you have, but maybe
it will be supported by the usual 3ginfo package along with the Luci
package.

closed 2022-01-07 by 4IceG