4IceG/luci-app-3ginfo-lite #8: luci-app-3ginfo-lite

Pirateek, 2022-06-01:
Cześć
Usunąłem stary pakiet luci-app-3ginfo-lite
Zaktualizowałem pakiet luci-app-sms-tool

dalej mam to samo
Uncaught (in promise) SyntaxError: JSON.parse: bad control character in string literal at line 5 column 16 of the JSON data
    render http://192.168.3.1/luci-static/resources/view/modem/3gdetail.js?v=git-22.083.69105-af8e91c:31
    promise callback*render/< http://192.168.3.1/luci-static/resources/view/modem/3gdetail.js?v=git-22.083.69105-af8e91c:31
    step http://192.168.3.1/luci-static/resources/luci.js?v=git-22.083.69105-af8e91c:92

4IceG, 2022-06-01:
Siema
@Pirateek pokaz mi wynik polecenia:
`/usr/share/3ginfo-lite/3ginfo.sh`

Pewnie gdzies dodawany jest cudzyslow i js glupieje. 
Uroki dodawania "wsparcia" kiedy sie nie ma fizycznie sprzetu i nie widzi efektow/bledow.

Pirateek, 2022-06-01:
root@P1G:~# /usr/share/3ginfo-lite/3ginfo.sh
{
"connt":"0d, 00:44:17",
"conntx":"771.5 KiB",
"connrx":"1.4 MiB",
",odem":" F286R
"mtemp":"",
",irmware":"LMF286RMODULEV1.0.1B03
"cport":"/dev/ttyACM0",
"protocol":"ECM",
"csq":"19",
"signal":"61",
"operator_name":"Aero2",
"operator_mcc":"260",
"operator_mnc":"01",
"mode":"LTE-A | B7 (2600 MHz) / B3 (1800 MHz)",
"registration":"1",
"lac_dec":"11020",
"lac_hex":"2B0C",
"tac_dec":"",
"tac_hex":"-",
"cid_dec":"2896664",
"cid_hex":"2C3318",
"pci":"181",
"earfcn":"2850",
"pband":"B7 (2600 MHz) @20 MHz",
 MHz",":"B3 (1800 MHz) @20
"spci":"280",
"searfcn":"1300",
"rsrp":"-107",
"rsrq":"-13.0",
"rssi":"-74",
",inr":"1
 MHz"},{"(S) PCI":"280"},{"(S) EARFCN":"1300"}]{"PCI":"181"},{"EARFCN":"2850"},{"(S) band":"B3 (1800 MHz), 20
}

4IceG, 2022-06-02:
Wczoraj już nie miałem czasu, więc działam dopiero dziś.

@Pirateek uruchom sobie WinSCP i podmień zawartość plików w routerze:
`/usr/share/3ginfo-lite/3ginfo-addon/19d21489`
na tą zawartość z githuba
`https://github.com/4IceG/luci-app-3ginfo-lite/tree/main//luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/3ginfo-addon/19d21489`

oraz

`/usr/share/3ginfo-lite/3ginfo.sh`
na tą zawartość
`https://github.com/4IceG/luci-app-3ginfo-lite/tree/main/luci-app-3ginfo-lite/root/usr/share)/3ginfo-lite/3ginfo.sh`

bo wygląda to tak jakby dalej uruchamiały się stare skrypty.

Potem pozostaje nam już tylko edycja/korekta w pliku
`/usr/share/3ginfo-lite/3ginfo-addon/19d21489`

4IceG, 2022-06-02:
U mnie jest różnie z czasem, nie zawsze mam chwilę i ochotę aby po pracy jeszcze siadać do komputera. 
Może jakoś się dojdzie co tu się z tymi modemami pomerdało, bo z historii na forum wiemy już że wersje R potrafią mieć rożne modemy i pewnie tu jest problem. Moje skrypty nie są wystarczająco uniwersalne.

Ok @Pirateek na routerze tym co nie działa w pliku
`/usr/share/3ginfo-lite/3ginfo-addon/19d21489`
usuń poniższe linijki i sprawdź czy chociaż pozostałe dane pokazuje poprawnie.
```
# OF=$(sms_tool -d $DEVICE at "ati")
# DEVV=$(echo $OF | tr -s "\n"  | xargs)
FWA=$(sms_tool -d $DEVICE at "AT+GMR" | tr -s "\n"  | xargs)
FW=$(echo $FWA | cut -c 8-)

MODELB=$(sms_tool -d $DEVICE at "AT+CGMI" | tr -s "\n"  | xargs)
MODELBCUT=$(echo $MODELB | awk -F'CGMI: ' '{print $2}' | sed 's/"//g' | sed 's/ CORPORATION//g')

MODELD=$(sms_tool -d $DEVICE at "AT+CGMM" | tr -s "\n"  | xargs)
MODELDCUT=$(echo $MODELD | awk -F'CGMM: ' '{print $2}' | sed 's/"//g')
MODEL="$MODELBCUT "$MODELDCUT

# Protocol
# DRIVER=QMI_WWAN & DRIVER=CDC_MBIM & DRIVER=CDC_ETHER
PV=$(cat /sys/kernel/debug/usb/devices 2>/dev/null)
PVCUT=$(echo $PV | awk -F 'Vendor=19d2 ProdID=1489' '{print $2}' | cut -c-1450)
if echo "$PVCUT" | grep -q "cdc_ncm"
then
    PROTO="ncm"
fi
```

Pirateek, 2022-06-02:
dostaję taki błąd

Uncaught (in promise) SyntaxError: JSON.parse: bad control character in string literal at line 26 column 27 of the JSON data
    render http://192.168.3.1/luci-static/resources/view/modem/3gdetail.js?v=git-22.083.69105-af8e91c:31
    promise callback*render/< http://192.168.3.1/luci-static/resources/view/modem/3gdetail.js?v=git-22.083.69105-af8e91c:31
    step http://192.168.3.1/luci-static/resources/luci.js?v=git-22.083.69105-af8e91c:92

Pirateek, 2022-06-02:
root@P1G:/usr/share/3ginfo-lite# ./3ginfo.sh
{
"connt":"0d, 07:48:59",
"conntx":"18.2 MiB",
"connrx":"99.4 MiB",
"modem":"",
"mtemp":"",
"firmware":"",
"cport":"/dev/ttyACM0",
"protocol":"",
"csq":"20",
"signal":"64",
"operator_name":"Aero2",
"operator_mcc":"260",
"operator_mnc":"01",
"mode":"LTE-A | B7 (2600 MHz) / B3 (1800 MHz)",
"registration":"1",
"lac_dec":"11020",
"lac_hex":"2B0C",
"tac_dec":"",
"tac_hex":"-",
"cid_dec":"2896664",
"cid_hex":"2C3318",
"pci":"181",
"earfcn":"2850",
"pband":"B7 (2600 MHz) @20 MHz",
 MHz",":"B3 (1800 MHz) @20
"spci":"280",
"searfcn":"1300",
"rsrp":"-106",
"rsrq":"-14.0",
"rssi":"-72",
",inr":"-1
 MHz"},{"(S) PCI":"280"},{"(S) EARFCN":"1300"}]{"PCI":"181"},{"EARFCN":"2850"},{"(S) band":"B3 (1800 MHz), 20
}

4IceG, 2022-06-02:
@Pirateek widzisz, że tu jest jeszcze od groma złych danych i js ma problem aby pokazać niewłaściwe wartości.

Na ten moment głupoty są jeszcze przy sband i sinr. Pytanie czy ja coś pomerdałem czy składnia w poleceniach at jest inna.
Ok, czyli ciąg dalszy śledztwa.

@Pirateek podmień zawartość pliku:
`/usr/share/3ginfo-lite/3ginfo-addon/19d21489`
na tą zawartość z githuba Cezarego
`https://github.com/obsy/packages/blob/master/easyconfig/addon/3ginfo-addon/19d21489`
wyjdzie czy więcej danych wyciągnie niezmieniony skrypt Cezarego. Luci nic nie pokaże, ale będzie wiadomo czy pomerdałem coś ja.

Kolejna sprawa, pokaż proszę wynik poleceń at:
```
at+zrssi
at+zcellinfo?
at+zcainfo?
```

Pirateek, 2022-06-02:
po podmianie
root@P1G:/usr/share/3ginfo-lite# ./3ginfo.sh
{
"connt":"0d, 09:12:22",
"conntx":"19.3 MiB",
"connrx":"103.2 MiB",
"modem":"",
"mtemp":"",
"firmware":"",
"cport":"/dev/ttyACM0",
"protocol":"",
"csq":"20",
"signal":"64",
"operator_name":"Aero2",
"operator_mcc":"260",
"operator_mnc":"01",
"mode":"LTE_A B7 (2600 MHz) / B3 (1800 MHz)",
"registration":"1",
"lac_dec":"11020",
"lac_hex":"2B0C",
"tac_dec":"",
"tac_hex":"-",
"cid_dec":"2896664",
"cid_hex":"2C3318",
"pci":"",
"earfcn":"",
"pband":"",
"sband":"",
"spci":"",
"searfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":"",
 MHz"},{"(S) PCI":"280"},{"(S) EARFCN":"1300"}]{"PCI":"181"},{"EARFCN":"2850"},{"(S) band":"B3 (1800 MHz), 20
}

+ZRSSI: -107,-13.5,-73,4
+ZCELLINFO: 2896664,181,LTE B7,2850
+ZCAINFO: 181,7,11020,2850,20;280,1,3,1300,20

4IceG, 2022-06-02:
Pewnie wget traktuje to jako odnośnik do strony, nie do pliku i tak pobiera. Nie pobieraj plików wget-em tylko pobieraj zawartość pliku Raw i podmieniaj ręcznie w WinSCP, usuwając ewentualną ostatnią pustą linijkę.

Posiedziałem teraz trochę i na sucho podstawiałem Twoje przesłane wyniki poleceń pod fragmenty skryptów wycinające dane i u mnie wszystko działa, ładnie wycina i dane są poprawnie pokazywane jako json i w LuCI, więc skrypty są OK, robią swoją robotę.
Teraz wypadałoby szukać linijka po linijce (skrypt po skrypcie od danej wartości) w pliku 19d21489 do którego momentu pokazuje poprawnie, a od którego już są błędy i źle wycina. Ja routera nie mam wiec musiałbyś @Pirateek sam sobie chwilę posiedzieć.

Pirateek, 2022-06-02:
Pliki pobierałem przez RAW

Na chwilę obecną to mi nic nie wyświetla, mam nazwy bez wartości.
Może prościej zzrobię ikopiei iw gram od mowa wersję poczekam tylko na rc2 po niedzieli @Cezary pisał , że będzie. Jak nie ruszy będę grzebał, chociaż w skryptogi jestem cieniki. Przyznam się.

4IceG, 2022-06-02:
@Pirateek luzik, można poczekać na wersję rc2 potem wgrać najnowszą wersję pakietu i walczyć dalej. Skryptologi tu jakiejś nie ma, skrypt (polecenia) do wycinania są. Teraz tylko pozostają testy i sprawdzanie gdzie/na czym się wykłada. Bedę najwyżej pisał co i jak.

Pirateek, 2022-06-02:
Tak jak na grupie pisałem, mam 2 routery jeden z Play i uptime teraz mam prawie 10 dni i jeden restart Wan-u firmware modemu B11 z tego co pamiętam. Zaś drugi z T-mobile firmware B04 i mam restart routera co 2 dzien i restart Wan-u nawet 4x dziennie. Chyba to zależy od modemu.

4IceG, 2022-06-11:
Cześć @Pirateek 
I jak, próbowałeś coś ruszyć z tematem we własnym zakresie?

Pirateek, 2022-06-13:
Cześć   Zainstalowałem najnowszą wersję openwrt   OpenWrt 22.03-SNAPSHOT r19441-3cfe050c4a  Niestety nic nie pokazuje  W załączeniu zrzut  A to pokazuje 3ginfo.sh  ***@***.***:/usr/share/3ginfo-lite# ./3ginfo.sh  ./3ginfo.sh: line 51: getpath: not found  ./3ginfo.sh: line 54: getpath: not found  ./3ginfo.sh: line 54: getpath: not found  ./3ginfo.sh: line 54: getpath: not found  {  &#34;connt&#34;:&#34;0d, 00:18:07&#34;,  &#34;conntx&#34;:&#34;2.8 MiB&#34;,  &#34;connrx&#34;:&#34;11.9 MiB&#34;,  &#34;,odem&#34;:&#34; F286R  &#34;mtemp&#34;:&#34;&#34;,  &#34;,irmware&#34;:&#34;LMF286RMODULEV1.0.1B03  &#34;cport&#34;:&#34;/dev/ttyACM0&#34;,  &#34;protocol&#34;:&#34;ECM&#34;,  &#34;csq&#34;:&#34;23&#34;,  &#34;signal&#34;:&#34;74&#34;,  &#34;operator_name&#34;:&#34;Aero2&#34;,  &#34;operator_mcc&#34;:&#34;260&#34;,  &#34;operator_mnc&#34;:&#34;01&#34;,  &#34;mode&#34;:&#34;LTE B8 (900 MHz)&#34;,  &#34;registration&#34;:&#34;1&#34;,  &#34;lac_dec&#34;:&#34;11020&#34;,  &#34;lac_hex&#34;:&#34;2B0C&#34;,  &#34;tac_dec&#34;:&#34;&#34;,  &#34;tac_hex&#34;:&#34;-&#34;,  &#34;cid_dec&#34;:&#34;2896653&#34;,  &#34;cid_hex&#34;:&#34;2C330D&#34;,  &#34;pci&#34;:&#34;341&#34;,  &#34;earfcn&#34;:&#34;3526&#34;,  &#34;pband&#34;:&#34;B8 (900 MHz) @5 MHz&#34;,  &#34;sband&#34;:&#34;&#34;,  &#34;spci&#34;:&#34;&#34;,  &#34;searfcn&#34;:&#34;&#34;,  &#34;rsrp&#34;:&#34;-97&#34;,  &#34;rsrq&#34;:&#34;-14.0&#34;,  &#34;rssi&#34;:&#34;-67&#34;,  &#34;,inr&#34;:&#34;-3  dB&#34;},{&#34;Primary band&#34;:&#34;B8 (900 MHz), 5 MHz&#34;},{&#34;PCI&#34;:&#34;341&#34;},{&#34;EARFCN&#34;:&#34;3526&#34;}]:&#34;-14.0 dB&#34;},{&#34;SINR&#34;:&#34;-3  }   Dnia 11 czerwca 2022 17:51 Rafał Wabik ***@***.***&gt; napisał(a):    Cześć  github.com @Pirateek   I jak, próbowałeś coś ruszyć z tematem we własnym zakresie?

4IceG, 2022-06-13:
To było do przewidzenia, że nie będzie działać. 
Jak się nie przysiądzie i nie znajdzie błędu/problemu dla tego modemu w skryptach to tak zostanie.

@Pirateek , a na Mifi od Cezarego jest wszystko OK?

Pirateek, 2022-06-13:
Mifi nie instalowałem.  Mogę jutro zobaczyć jak będę mieć chwilkę, dzisiaj nie dam już rady.   Dnia 13 czerwca 2022 12:21 Rafał Wabik ***@***.***&gt; napisał(a):   To było do przewidzenia, że nie będzie działać.   Jak się nie przysiądzie i nie znajdzie błędu/problemu dla tego modemu w skryptach to tak zostanie.  github.com @Pirateek  , a na Mifi od Cezarego jest wszystko OK?

Pirateek, 2022-06-13:
Udało się znaleźć chwilę czasu, na MiFi pokazuje na www parametry modemu.

Pirateek, 2022-06-13:
kopię plików zrobiłem cały katalog /usr/share/3ginfo-lite.
Wynik 3ginfo.sh

root@P1G:~# ./3ginfo.sh
{
"csq":"22",
"signal":"70",
"operator_name":"Aero2",
"operator_mcc":"260",
"operator_mnc":"01",
"mode":"LTE B8 (900 MHz)",
",egistration":"READY
"lac_dec":"11020",
"lac_hex":"2B0C",
"cid_dec":"2896653",
"cid_hex":"2C330D",
 dB"},{"Primary band":"B8 (900 MHz), 5 MHz"},{"PCI":"341"},{"EARFCN":"3526"}]:"-13.5 dB"},{"SINR":"4
}

Pirateek, 2022-06-13:
Zawsze na początku na www w konfiguracji ustawiałem port modemu /devttyACM0
teraz nie ustawiłem i wszedłem w zakładkę Detalis 3G/4G i o dziwo jest pokazanych kilkanaście parametrów.
Sądzę, że jak wpiszę port modemu to zniknie
[image]

Pirateek, 2022-06-13:
zauważyłem też , że przy uruchomieniu modprobe.gcom dostaję komunikat

root@P1G:/usr/share/3ginfo-lite# ./probeport.gcom
./probeport.gcom: line 1: syntax error: unexpected "("

Pirateek, 2022-06-13:
Wyczyściłem cache przeglądarki i mam ECM - protokół

4IceG, 2022-06-14:
Ja się poddaje, jak pokazuje coś czego nie ma w skryptach dla MF286R `/usr/share/3ginfo-lite/3ginfo-addon/19d21489`.

Pirateek, 2022-06-14:
To jest zrzut w/w pliku i jest ECM na samym końcu 
//// PROTO="ECM" ////
Z drugiej strony te modemy są dziwne bo zależy jak je zapytasz.
Raz pokażą wersje modemu B04 a raz B03 (@Cezary to potwierdził)
https://eko.one.pl/forum/viewtopic.php?pid=269619#p269619
Może tu jest problem?

root@P1G:/usr/share/3ginfo-lite/3ginfo-addon# cat 19d21489
# ZTE MF286R

O=$(sms_tool -d $DEVICE at "at+zrssi;+zcellinfo?;+zcainfo?")

case "$MODE_NUM" in
        3*)
                T=$(echo "$O" | awk -F[,:] '/^\+ZRSSI/ {print $2}' | xargs)
                [ -n "$ADDON" ] && ADDON="$ADDON,"
                RSSI="$T"
                ADDON="$ADDON"'{"RSSI":"'$T' dBm"}'
                ;;
        2*|\
        4*|\
        5*|\
        6*|\
        8*)
                T=$(echo "$O" | awk -F[,:] '/^\+ZRSSI/ {print $2}' | xargs)
                [ -n "$ADDON" ] && ADDON="$ADDON,"
                ADDON="$ADDON"'{"RSCP":"'$T' dBm"}'
                T=$(echo "$O" | awk -F[,:] '/^\+ZRSSI/ {print $3}' | xargs)
                [ -n "$ADDON" ] && ADDON="$ADDON,"
                ADDON="$ADDON"'{"ECIO":"'$T' dB"}'
                T=$(echo "$O" | awk -F[,:] '/^\+ZCELLINFO/ {print $4}' | xargs)
                [ -n "$T" ] && MODE="$T"
                ;;

        7*|\
        9*)
                T=$(echo "$O" | awk -F[,:\;] '/^\+ZCAINFO/ {print $4}' | xargs)
                if [ -n "$T" ] && [ "x$T" != "x0" ]; then
                        [ -n "$ADDON" ] && ADDON="$ADDON,"
                        T_HEX=$(printf "%X" $T)
                        ADDON="$ADDON"'{"TAC":"'$T' ('$T_HEX')"}'
                fi

                T=$(echo "$O" | awk -F[,:] '/^\+ZRSSI/ {print $4}' | xargs)
                [ -n "$ADDON" ] && ADDON="$ADDON,"
                RSSI="$T"
                ADDON="$ADDON"'{"RSSI":"'$T' dBm"}'
                T=$(echo "$O" | awk -F[,:] '/^\+ZRSSI/ {print $2}' | xargs)
                [ -n "$ADDON" ] && ADDON="$ADDON,"
                RSRP="$T"
                ADDON="$ADDON"'{"RSRP":"'$T' dBm"}'
                T=$(echo "$O" | awk -F[,:] '/^\+ZRSSI/ {print $3}' | xargs)
                [ -n "$ADDON" ] && ADDON="$ADDON,"
                RSRQ="$T"
                ADDON="$ADDON"'{"RSRQ":"'$T' dB"}'
                T=$(echo "$O" | awk -F[,:] '/^\+ZRSSI/ {print $5}' | xargs)
                [ -n "$ADDON" ] && ADDON="$ADDON,"
                SINR="$T"
                ADDON="$ADDON"'{"SINR":"'$T' dB"}'

                T=$(echo "$O" | awk -F[,:] '/^\+ZCELLINFO/ {print $4}' | xargs)
                case "$T" in
                        *"LTE B1") MODE=$(band 1 "LTE ");;
                        *"LTE B3") MODE=$(band 3 "LTE ");;
                        *"LTE B7") MODE=$(band 7 "LTE ");;
                        *"LTE B8") MODE=$(band 8 "LTE ");;
                        *"LTE B20") MODE=$(band 20 "LTE ");;
                        *) MODE="${T}";;
                esac

[... 59 lines cut ...]
FWA=$(sms_tool -d $DEVICE at "AT+GMR" | tr -s "\n"  | xargs)
FW=$(echo $FWA | cut -c 8-)

MODELA=$(echo $DEVV | awk -F 'Revision: ' '{print $1}'| xargs)
MODELB=$(echo $MODELA | awk -F 'Manufacturer: |CORPORATION' '{print $2}'| xargs)

MODELD=$(sms_tool -d $DEVICE at "AT+CGMM" | tr -s "\n"  | xargs)
MODELDCUT=$(echo $MODELD | cut -c 9-)

MODEL="$MODELB "$MODELDCUT

# Protocol
# DRIVER=QMI_WWAN & DRIVER=CDC_MBIM & DRIVER=CDC_ETHER
PV=$(cat /sys/kernel/debug/usb/devices)
PVCUT=$(echo $PV | awk -F 'Vendor=19d2 ProdID=1489' '{print $2}' | cut -c-1220)
if echo "$PVCUT" | grep -q "Driver=qmi_wwan"
then
    PROTO="QMI"
elif echo "$PVCUT" | grep -q "Driver=cdc_mbim"
then
    PROTO="MBIM"
elif echo "$PVCUT" | grep -q "Driver=cdc_ether"
then
    PROTO="ECM"
fi

4IceG, 2022-06-14:
To co mi pokazales to jest juz nieaktualne i na moim githubie skrypt wyglada inaczej, dlatego sie tak dziwilem na to co pokazuje.

Modemy sa problematyczne poniewaz pod jednymi identyfikatorem VID oraz PID wystepuja dwa lub juz nawet trzy modemy spotykane w jednym routerze. Ja nie majac tych urzadzen fizycznie mam duzy problem w dostosowaniu skryptu, aby byl uniwersalny i radzil sobie poprawnie z pobraniem danych ze wszystkiego. Pewnie znow musialbym przekopac caly temat na forum aby zebrac wszystkie wariacje tych modemow i przepisac skrypt.

Pirateek, 2022-06-14:
Mogę udostępnić router po wireguard do testów - nie mam nic ważnego.  Tak Leo-PL rozkminił u mnie , że jest modem obsługiwany dla 286R   Dnia 14 czerwca 2022 13:10 Rafał Wabik ***@***.***&gt; napisał(a):   To co mi pokazales to jest juz nieaktualne i na moim githubie skrypt wyglada inaczej, dlatego sie tak dziwilem na to co pokazuje.  Modemy sa problematyczne poniewaz pod jednymi identyfikatorem VID oraz PID wystepuja dwa lub juz nawet trzy modemy spotykane w jednym routerze. Ja nie majac tych urzadzen fizycznie mam duzy problem w dostosowaniu skryptu, aby byl uniwersalny i radzil sobie poprawnie z pobraniem danych ze wszystkiego. Pewnie znow musialbym przekopac caly temat na forum aby zebrac wszystkie wariacje tych modemow i przepisac skrypt.

Pirateek, 2022-06-14:
W wolnej chwili przgotuje co trzeba (co bedziesz musiał sam zrobić) , może uda się dzisiaj wieczorem. Jak nie to w długi weekend, bo jutro w pracy mam wyjazd.

4IceG, 2022-06-14:
Oki, W wolnej chwili tez zerkne jak wyglada jakas przykladowa konfiguracja dla Windowsa.

Pirateek, 2022-06-14:
A ja myślałem o zdefiniowaniu wireguarda na routerze z openwrt,  zrobi.y klienta i po ssh się wbijesz

Pirateek, 2022-06-22:
Cześć.
Wyślij mi e-maila,  na który wysłać klucz do WG.
Jakoś czas szybko umyka i nie miałem jak wcześniej napisać

Pozdrawiam

4IceG, 2022-06-22:
Siema,
Spoko, nic się nie stało, nie pali się bardzo z tematem :).
Wysyłaj/pisz jak coś na 4rafalSZYMPANSgmail.com.
Pozdrawiam

closed 2022-07-07 by 4IceG

Pirateek, 2022-10-11:
Cześć.  W pracy dużo roboty  a popołudniu troszkę  zaległości na działce w następnym tygodniu myślę,  że zrobię upgrade.   Dnia 11 czerwca 2022 17:51 Rafał Wabik ***@***.***&gt; napisał(a):   Cześć  github.com @Pirateek   I jak, próbowałeś coś ruszyć z tematem we własnym zakresie?