4IceG/luci-app-3ginfo-lite #9: Request add support Foxconn T77W676

odie84, 2022-06-26:
vid & pid, It has 2 brand vendor HP & Telit

03f0:0a57 (HP lt4220 MBIM)
03f0:0857 (HP lt4220 QMI)
1bc7:1901 (Telit Ln940 MBIM)
1bc7:1900 (Telit Ln940 QMI) 

ATI

ati
Manufacturer: QUALCOMM INCORPORATED
Model: 4105
Revision: T77W676.F0.0.0.4.7.GC.017  1  [Jul 11 2018 00:00:00]
SVN: 03
IMEI: xxx
+GCAP: +CGSM,+DS,+ES

cat /sys/kernel/debug/usb/devices

	T: Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 7 Spd=480 MxCh= 0 	
	D: Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs= 3 	
	P: Vendor=03f0 ProdID=0857 Rev= 3.10 	
	S: Manufacturer=Qualcomm 	
	S: Product=Snapdragon(TM) X12 LTE-A 	
	S: SerialNumber=0123456789ABCDEF

	T: Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 3 Spd=480 MxCh= 0 	
	D: Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs= 1 	
	P: Vendor=1bc7 ProdID=1900 Rev= 3.10 	
	S: Manufacturer=Telit 	
	S: Product=Telit LN940 Mobile Broadband 	
	S: SerialNumber=0123456789ABCDEF

Serial port

/dev/ttyUSB1 & /dev/ttyUSB2

All band supported
4G: 1,2,3,4,5,7,8,12,13,17,18,19,20,21,25,26,28,29,30,38,39,40,41,66

4IceG, 2022-06-26:
Hi,
To add some modem to 3ginfo package you need commands at with documentation/explanation/examples of results. I don't have these modems, so the more information I get it will be easier to add support.

4IceG, 2022-06-26:
If you really need information from the modem, you can use the package [https://github.com/koshev-msk/luci-app-modeminfo](https://github.com/koshev-msk/luci-app-modeminfo) the guys added these modems.

I found documentation with AT commands for telit ln940 but won't add it soon due to lack of time.

odie84, 2022-07-23:
[image]
[image]

saya menggunakan versi terbaru luci-app-3ginfo-lite dan hasilnya seperti itu

4IceG, 2022-07-23:
Hi @odie84
Ok, and the modem is?

odie84, 2022-07-23:
telit ln940

4IceG, 2022-07-23:
I haven't added this modem yet, so it's not surprising that it doesn't show more.
When I add it there will be a mention of it in the readme.

odie84, 2022-07-23:
ok thanks in advance.

odie84, 2022-08-02:
[1bc71900.txt](https://github.com/4IceG/luci-app-3ginfo-lite/files/9239162/1bc71900.txt)

please fix something is still missing

4IceG, 2022-08-02:
Hi @odie84 ,
It's nice to see that the new modems are described. But what am I supposed to do with it? You didn't show the result of these commands, so I don't know what's wrong. I don't have these modems, so it's a waste of time to manually search for command output.

odie84, 2022-09-06:
[image]
[image]
[1bc71900.txt](https://github.com/4IceG/luci-app-3ginfo-lite/files/9493800/1bc71900.txt)

4IceG, 2022-09-06:
Hi @odie84 , I see that there is progress :), I will have some free time this week then I will try to add support for this modem.

4IceG, 2022-09-11:
Hi @odie84 , if you can, check how much my version shows. I don't have this modem and I didn't find results for some commands, so further testing / fixes may be needed.
[1bc71900.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/9542676/1bc71900.zip)

odie84, 2022-09-11:
[image]
[image]

odie84, 2022-09-11:
root@OpenWrt:~# sh -x /usr/share/3ginfo-lite/3ginfo.sh
+ RES=/usr/share/3ginfo-lite
+ uci -q get '3ginfo.@3ginfo[0].device'
+ DEVICE=/dev/ttyUSB1
+ '[' x/dev/ttyUSB1 '=' x ]
+ echo /dev/ttyUSB1
+ '[' x/dev/ttyUSB1 '=' x ]
+ '[' x/dev/ttyUSB1 '=' x ]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wan1
+ '[' -z wan1 ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wan1
+ grep '"up": true'
+ NETUP='       "up": true,'
+ '[' -n '      "up": true,' ]
+ uci -q -P /var/state/ get network.wan1.connect_time
+ CT=
+ '[' -z ]
+ ifstatus wan1
+ awk '-F[:,]' '/uptime/ {print $2}'
+ xargs
+ CT=29911
+ '[' '!' -z 29911 ]
+ expr 29911 / 60 / 60 / 24
+ D=0
+ expr 29911 / 60 / 60 '%' 24
+ H=8
+ expr 29911 / 60 '%' 60
+ M=18
+ expr 29911 '%' 60
+ S=31
+ printf '%dd, %02d:%02d:%02d' 0 8 18 31
+ CONN_TIME='0d, 08:18:31'
+ ifstatus+  wan1awk
 '-F"' '/l3_device/ {print $4}'
+ IFACE=wwan0
+ '[' -n wwan0 ]
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='482.3 MiB'
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='105.3 MiB'
+ sms_tool -D -d /dev/ttyUSB1 at 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='
+CSQ: 24,99

+CPIN: READY

+COPS: 0,0,"XL Axiata XL Axiata",7

+COPS: 0,2,"51011",7

+CREG: 2,1,"FFFE","7CA2803",7

'K
+ echo '
+CSQ: 24,99

+CPIN: READY

+COPS: 0,0,"XL Axiata XL Axiata",7

+COPS: 0,2,"51011",7

+CREG: 2,1,"FFFE","7CA2803",7

[... 540 lines cut ...]
"tac_hex":"",
"tac":"",
"cid_dec":"130689027",
"cid_hex":"",
"pci":"",
"earfcn":"376",
"pband":"B1 (2100 MHz) @10.0 MHz",
"s1band":"B3 (1800 MHz) @20.0 MHz",
"s1pci":"",
"s1earfcn":"",
"s2band":"B8 (900 MHz) @5.0 MHz",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"51005",
"sinr":"-65"
}
+ exit 0

4IceG, 2022-09-11:
OK @odie84 , it should show well what is now.
The pci & earfcn data for the rest of the bands won't be there because we don't get them.

[1bc71900v2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/9543059/1bc71900v2.zip)

closed 2022-09-12 by 4IceG