4IceG/luci-app-3ginfo-lite #10: Add support for MF289F

stich86, 2022-08-02:
Hi @4IceG 

i'm porting OpenWRT on MF289F. This is a "little" brother of MF286D.
ID of modem are the same

`Bus 002 Device 002: ID 19d2:1485 ZTE,Incorporated ZTE Mobile Broadband`

But the one into this router is 4CA, so the command "AT+ZCAINFO?" reporting 4 bands instead of 3 if you are forcing bands or leaving it in AUTO.

Here is an example output:

`at+zcainfo?
+ZCAINFO: 298, 20, 33003, 6400, 10.0;          130, 2, 3, 1850, 20.0;          225, 2, 1, 525, 15.0;          16, 2, 7, 3025, 15.0;
`

Do you think that is possible to add 4th band display, making a check on ATI commands? I've seen your script, should be easy to address.

Thanks :)

4IceG, 2022-08-02:
Hi @stich86 
I was supposed to change 3ginfo, because they already wrote to me on the forum that it could show more, but I don't know how to show it yet, so that it would be still simple.

Scripts are just scripts and we can customize them, change them to show more (more bands).

stich86, 2022-08-04:
> Hi @stich86 I was supposed to change 3ginfo, because they already wrote to me on the forum that it could show more, but I don't know how to show it yet, so that it would be still simple.
> 
> Scripts are just scripts and we can customize them, change them to show more (more bands).

i've sent you a PR, here is how is loking with the other bands added:

[image]

let me know if you like it :)

4IceG, 2022-08-04:
Hi @stich86 , it's cool that you've customized the scripts, but I won't add it as is anyway because I've already modified 3ginfo and it's similar but different (I'll show in a picture later what it looks like). I'm not publishing the changes because I'm still thinking of an elegant way to search for BTS by Cell ID.

4IceG, 2022-08-05:
Hi @stich86 
I have published a new version of the package, you can take a peek to see if it looks better.

closed 2022-08-05 by 4IceG

stich86, 2022-08-05:
looks like mine with the addon of BTS info. Problem of that MF289F doesn't give cell-id into AT+CREG command :(

Now we have a problem with MF289F AT2 rev that expose modem as MF286\286A. The script of MF286\286A is using into AWK field seprator that doesn't contain ";", so all paramerts are wrong.

As you have implemented isn't easy to fix, because you are looking VID\PID thru usb devices..

Any ideas?

4IceG, 2022-08-05:
if the "modems are the same", maybe you need to do device name verification? and depend on the name of the script?

stich86, 2022-08-05:
> if the "modems are the same", maybe you need to do device name verification? and depend on the name of the script?

yes are the same, but the HW rev AT2 expose it as 19d2:1432 instead of 19d2:1485, so you script will use addon "19d21432" that doesn't recognize the output of AT commands

4IceG, 2022-08-05:
So the script should verify which device it is running on, since the modems are the same, you need to add the device name condition.

stich86, 2022-08-05:
> So the script should verify which device it is running on, since the modems are the same, you need to add the device name condition.

yes.. an may be the idea is to parametrize the AWK filter based on device name, so you will avoid to replicate all band stuff on the script

What do you think?

4IceG, 2022-08-05:
There is no other way, since the script should be universal.

stich86, 2022-08-05:
> There is no other way, since the script should be universal.

ok i'll try to refactor your script and send a PR so you can check if can be integrated

stich86, 2022-08-05:
PR create, take a look and let me know :)

4IceG, 2022-08-05:
Okay, it looks unusual, but if it works. Today I'm barely alive after work so tomorrow I'll add it and make a test package. I don't have this router so the guys on the forum will have to check if it works.

stich86, 2022-08-05:
MiG25 has that modem on Eko, so he can also test it.

I've already tested on a router of a friend and it's working