4IceG/luci-app-3ginfo-lite #16: CSS support?

bam80, 2022-09-19:
Hi, thank you for the project.
I'm not very familiar with Web technologies, but I see no CSS in the language list:
[image]

Do you think it can be added?

4IceG, 2022-09-19:
Hi @bam80 ,
CSS file is not required, the package is based on the standard LuCI appearance, if something needs to be changed dynamically js script does it and that's it. The package was supposed to look simple.

closed 2022-09-19 by 4IceG

bam80, 2022-09-19:
OK.
It's just because I asked some modification from alternative project:
https://github.com/koshev-msk/modemfeed/tree/master/luci/applications/luci-app-modeminfo
and I was replied the visual change I wanted is a job for CSS.

Ideally, I would like the interface something like this, for pointing antenna on roof :)
[image]

4IceG, 2022-09-20:
My package does not go into space :), I do not need so much data, and not every modem provides that much.