4IceG/luci-app-3ginfo-lite #18: installation from the sources is not documented

bam80, 2022-09-26:
While trying to solve #17, I came to the question if installation from sources is possible instead?
It turned out the repo here can be used as openwrt feed, but it's not documented anywhere.

Based on my own experience, I think it's not very obvious the app can be installed this way, so it's probably better to have a note about it in Readme.
Here is the example:
https://github.com/koshev-msk/modemfeed#how-to-add-repo-and-compile-packages

4IceG, 2022-09-26:
Thanks for the suggestion. As soon as I have a free moment, I will test the option from the link.

Earlier, I did not think about adding such information. My packages are rather unpopular so I didn't have many compilation questions.

closed 2022-09-26 by 4IceG

bam80, 2022-09-26:
> I will test the option from the link

Hmm, isn't `feed` a way you compile this app for yourself? I wonder if it's possible to do it without.

As a side note, if you know a more popular app for monitoring modem signal - please let me know :)