4IceG/luci-app-3ginfo-lite #22: upstream the project?

bam80, 2022-10-06:
Did you think about upstreaming this to OpenWrt luci repo?
As we don't have similar package there anyway, this might be a good idea.

4IceG, 2022-10-06:
I didn't think about moving the package to the repository for two reasons. 
The first is the matter of adding the sms-tool package, and I do not know what Cezary has to say.
Second, I don't know if my package would pass developer verification. In addition, my package is not very universal, it supports very few modems.

But yes, I agree the packages for modems are missing in Openwrt.

bam80, 2022-10-06:
Could we tag Cezary here (I don't know if he's present on Github) or ask him by any other means?

For the second, I really wish if we try.
I can say I'm a hobbyist for myself but still, my PR was accepted into the OpenWrt project:
https://github.com/openwrt/openwrt/pull/4791
It's not as scary as it seems to be.
I'm sure your packages are still very useful for many people, that should justify the upstreaming.
 
Not having these things in the upstream OpenWrt or at least in a dedicated packages repository makes the whole OpenWrt project hardly usable for modems, unfortunately.

4IceG, 2022-10-06:
Cezary has his github (the link is in thanks on the Readme page), his forum (eko.one.pl), so you can  @bam80 ask him / try to persuade him to submit a sms-tool to the repository.

obsy, 2022-10-06:
sms_tools has its own repository: https://github.com/obsy/sms_tool, there is also a makefile for OpenWrt: https://github.com/obsy/packages/tree/master/sms-tool. If anyone wants to submit PR to packages, let them do so.

bam80, 2022-10-09:
https://github.com/openwrt/packages/pull/19565

4IceG, 2022-10-09:
With sms-tool there should be no problems with adding, We all know that the application is very useful. 

@bam80 If you want you can also try adding a package for LuCI (but here it will probably be much more difficult and .. with a lot of corrections in the code).

bam80, 2022-10-10:
> @bam80 If you want you can also try adding a package for LuCI (but here it will probably be much more difficult and .. with a lot of corrections in the code).

I could try to make a PR but I probably won't be able to do any corrections myself as I don't follow any Web-based code

4IceG, 2022-10-11:
I'm not a programmer either, so it can be funny :).

closed 2022-12-26 by 4IceG