4IceG/luci-app-3ginfo-lite #23: Provide a repository index to the Assets

bam80, 2022-10-06:
I just tried and a [stable link to the latest release](https://stackoverflow.com/a/54836319/1063363) is actually works!:
https://github.com/4IceG/luci-app-3ginfo-lite/releases/latest/download/luci-app-3ginfo-lite_all.ipk

This opens the doors for making the Assets usable as OpenWrt package repository!
To make it happen, we just probably need to add a repository index to the Assets.

The index is generated with following command:
>  `make package/index` - build a repository index to make the output directory usable as local opkg source

https://openwrt.org/docs/guide-developer/toolchain/using_the_sdk#compile_packages

It produces 3 files for the repo, not sure if all of them are essential:
```
Packages
Packages.gz
Packages.manifest
```

After publishing these files to the Assets, we should be able to add the following to the repositories config, and have hassle-free installation/updates :)
```
src/gz luci-app-3ginfo-lite https://github.com/4IceG/luci-app-3ginfo-lite/releases/latest/download

bam80, 2022-11-01:
@4IceG ping, please?

4IceG, 2022-11-01:
@bam80 I'm listening to you, what is it?

bam80, 2022-11-01:
How about the proposal above?

4IceG, 2022-11-01:
At the moment I have added one repository, just wait for the sms-tool package to be added to Openwrt, which would also make installing my packages a lot easier. It's all new to me and I'm learning these things.

bam80, 2022-11-01:
You mean a packages repository? Where?

4IceG, 2022-11-01:
Currently I have made the packages available here
https://github.com/4IceG/Modem-extras

4IceG, 2022-11-01:
I did not want to change the description yet, I thought that the sms-tool would be added and then..

bam80, 2022-11-01:
How about adding sms-tool in your repo too until it doesn't happened?

4IceG, 2022-11-01:
The problem is that the sms-tool must be compiled for a given architecture, and I will not add this for all supported routers because it does not compile them.

bam80, 2022-11-01:
I think it still worth to be mentioned in Readme, because it's really important.
We can add that `sms-tool` needs to be installed manually one way or another, for now..

Alternatively, try the approach suggested here.

4IceG, 2022-11-01:
I think the current description is optimal. Informs about the need to install sms-tool and compile your own if the package is not available.

If it says that it can be downloaded and installed manually, there will be confusion. How download? from where? why is there no packet on my router? will you compile me on ..? I'd rather avoid it.

bam80, 2022-11-01:
And what about adding the index to the Assets? Is it hard to do? (I just don't know)

4IceG, 2022-11-01:
I created the repository according to the guide, I honestly did not think it could be so simple.

https://eko.one.pl/?p=openwrt-kompilacja#tworzeniewasnegorepozytorium

closed 2022-11-02 by 4IceG

4IceG, 2022-11-07:
Hi :), I will try to do it this week, I will build new images, repository, improve my packages (package description will be improved to be more accurate when installing in LuCI).

bam80, 2022-11-25:
Hi, just wondered if it can happen any time soon?

4IceG, 2022-11-25:
Hi, I will try to update all the descriptions together with the new release of the repository tomorrow, Sunday at the latest. Sorry for the delay, but I don't always have as much time for OpenWrt as I would like.