4IceG/luci-app-3ginfo-lite #25: Provide more information for Huawei E3372 via AT commands

patrakov, 2022-10-16:
Huawei E3372 reflashed to stick mode is already supported and has its own 3ginfo-addon file:

```
Bus 001 Device 003: ID 12d1:1506 HUAWEI_MOBILE HUAWEI_MOBILE
```

Currently, it is missing the display of IMEI, IMSI, ICCID, TAC, and the active band.

It can provide IMEI either as a part of its response to `ATI`, or as a response to `AT+GSN`.

```
# sms_tool -d /dev/ttyUSB1 at 'AT+GSN'
AT+GSN
861XXXXXXXXXXXX

# sms_tool -d /dev/ttyUSB1 at 'ATI'
ATI
Manufacturer: huawei
Model: E3372
Revision: 21.200.07.00.00
IMEI: 861XXXXXXXXXXXX
+GCAP: +CGSM,+DS,+ES

```

Regarding IMSI:

```
# sms_tool -d /dev/ttyUSB1 at 'AT+CIMI'
AT+CIMI
51566XXXXXXXXXX

```

Regarding ICCID:

```
# sms_tool -d /dev/ttyUSB1 at 'AT^ICCID?'
AT^ICCID?
^ICCID: 896XXXXXXXXXXXXXXXXX

```

I could not find how to request TAC.

There is no dedicated command to display the band used, but there is `AT^HFREQINFO?` which displays frequency information which can be a (more detailed) substitute. I could not find any documentation specific to E3372, but http://download-c.huawei.com/download/downloadCenter?downloadId=100630 contains this command.

```
# sms_tool -d /dev/ttyUSB1 at 'AT^HFREQINFO?'
AT^HFREQINFO?
^HFREQINFO:0,6,1,425,21525,15000,18425,19625,15000

```

...which means: no unsolicited reporting, LTE mode, LTE band 1, downlink frequency point 425, downlink frequency 2152.5 MHz, downlink bandwidth 15 MHz, uplink frequency point 18425, downlink frequency 1962.5 MHz, downlink bandwidth 15 MHz.

4IceG, 2022-10-16:
Ok, I'll add this when I have some free time for the week. TAC will probably double with LAC if it can't be read.

patrakov, 2022-10-16:
LAC is actually retrieved successfully, but TAC is missing.

patrakov, 2022-10-20:
Oops, the chip temperature is missing.

```
# sms_tool -d /dev/ttyUSB1 at "AT^CHIPTEMP?"
AT^CHIPTEMP?
^CHIPTEMP: 473,473,65535,35,65535

```

I have manually copy-pasted the awk command, and it prints the correct temperature. So the bug must be not in the parsing of the AT command response, but in how it is propagated to the json.

patrakov, 2022-10-20:
The problem is due to the inappropriate reuse of the `$O` variable:

```
O=$(sms_tool -d $DEVICE at "at^chiptemp?;^hcsq?")                                                                                                       
...
O=$(sms_tool -d $DEVICE at "AT+CIMI" | xargs)                                  
...
T=$(echo "$O" | awk ...)   # of course it won't find the temperature in the AT+CIMI response
```

4IceG, 2022-10-20:
Ok thanks for the tests. I'll fix it tomorrow.

Can't find a command for reading PCI so left blank.

4IceG, 2022-10-21:
It should be OK.

closed 2022-10-21 by 4IceG