4IceG/luci-app-3ginfo-lite #26: problem z wersja 1.0.31

lukjod, 2022-11-27:
Nie dziala po aktualizacji do najnowszej wersji, znika menu modem, nie pomaga restart ani czyszcenie cache w /tmp
Po instalce 1.0.30 wszystko ok. Openwrt od Cezarego  OpenWrt 22.03-SNAPSHOT r19862-615dbec223
architektura mediatek/mt7622 ruter Linksys e8450 modem Huawei e5762
Na porzedniej wersji ustawienia:
`config 3ginfo
	option website 'http://www.btsearch.pl/szukaj.php?mode=std&search='
	option network '@WAN_USB'
	option device '192.168.8.1'`
cat /sys/kernel/debug/usb/devices
`T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.10
S:  Manufacturer=Linux 5.10.152 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1a0c0000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=12d1 ProdID=1506 Rev= 1.02
S:  Manufacturer=HUAWEI_MOBILE
S:  Product=HUAWEI_MOBILE
C:* #Ifs= 3 Cfg#= 1 Atr=c0 MxPwr=  2mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=02 Prot=12 Driver=option
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 1 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=02 Prot=16 Driver=huawei_cdc_ncm
E:  Ad=84(I) Atr=03(Int.) MxPS=  16 Ivl=2ms
I:* If#= 1 Alt= 1 #EPs= 3 Cls=ff(vend.) Sub=02 Prot=16 Driver=huawei_cdc_ncm
E:  Ad=84(I) Atr=03(Int.) MxPS=  16 Ivl=2ms
E:  Ad=83(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=125us

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.10
S:  Manufacturer=Linux 5.10.152 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1a0c0000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms`

lukjod, 2022-11-27:
[[image]](url)

4IceG, 2022-11-27:
Cześć, wypuściłem nową wersję, powinno już być OK z menu.

lukjod, 2022-11-27:
Super, teraz dziala tylko btsearch nie daje zadnych info, zobacz prosze screen

lukjod, 2022-11-27:
[[image]](url)

lukjod, 2022-11-27:
teraz wyglada to tak u mnie na najnowszej wersji:
[[image]](url)

4IceG, 2022-11-27:
Ok, widze że są 2 wartosci nie takie jak być powinny, sprawdz pakiet na innej przegladarce i wyczysc cache.

1. Nie powinno być pustych nawiasow dla TAC, chyba że coś pomerdałem.. powinna być "-" jeżeli wartosci są puste. Pomysle jak to inaczej rozwiązać.
2. Btsearch szuka dla Cellid w postaci dziesietnej, a tu mamy tylko wersje 16 i w tym problem. Zerkne na to później.

lukjod, 2022-11-27:
tak na edge jest tylko -
Poza btsearch wszystko jest ok, dziekuje

closed 2022-11-27 by lukjod

4IceG, 2022-11-27:
Zrobilem aktualizacje, powinno juz normalnie wyszukiwac na btsearch.

lukjod, 2022-11-28:
Tak wszystko działa poprawnie, wyszukuje Stację w  btsearch.
Bardzo dziękuje za włożony czas to jest fajny dodatek dla wszystkich co korzystają z modemów komórkowych w openwrt

4IceG, 2022-11-28:
Luzik. Cieszę się, że dodatek się przydaje i jest używany. Można by dodać "Polak potrafi" :).