4IceG/luci-app-3ginfo-lite #27: Quectel EP06-E SyntaxError

GektorUA, 2022-12-18:
Hi

I have build latest version, and when Quectel EP06-E has CA (2 bands aggregation) 3ginfo has error:
Expected ',' or '}' after property value in JSON at position 269

Without CA there is no errors.

4IceG, 2022-12-18:
Hi, 
Show me the result of the command:
`sh -x /usr/share/3ginfo-lite/3ginfo.sh`

And try removing the package and reinstalling only from my repository, it's the latest possible.

GektorUA, 2022-12-19:
`{
"connt":"-",
"conntx":"-",
"connrx":"-",
"modem":"Quectel EP06",
"mtemp":"34 &deg;C",
"firmware":"EP06ELAR04A05M4G",
"cport":"/dev/ttyUSB7",
"protocol":"ecm",
"csq":"31",
"signal":"100",
"operator_name":"KYIVSTAR",
"operator_mcc":"255",
"operator_mnc":"03",
"mode":""LTE_A BAND 3" / "LTE BAND 7"",
"registration":"1",
"simslot":"",
"imei":"868xxxxxxxxxxxxxx",
"imsi":"255xxxxxxxxxxxxxx",
"iccid":"893800xxxxxxxxxxxxxx",
"lac_dec":"31764",
"lac_hex":"7C14",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"61646112",
"cid_hex":"3ACA520",
"pci":"",
"earfcn":"",
"pband":"",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-85",
"rsrq":"-12",
"rssi":"-52",
"sinr":"-17.4"
}
`

4IceG, 2022-12-20:
Hi, 
Thanks for the answer, but you didn't show me the whole output of the command, so I don't see errors in the scripts. You also sent situations when there is aggregation and there should be no errors. 
I will look for commands for this modem today, maybe I can repeat the error.

GektorUA, 2022-12-20:
Hi, i have make output in a file, maybe something goes wrong and not all output was put in a file. There is a lot of personal information, can i send it to you personally?

4IceG, 2022-12-20:
Hi, 
I know that a lot of data is collected, but the users themselves want to "show/read" as much as possible. You can send me the result by e-mail:
4rafal@gmail.com.

GektorUA, 2022-12-25:
Hi, i have send you detailed information.

4IceG, 2022-12-25:
Hi,
Yes I got the email, thanks. I'll check it in a week.

GektorUA, 2022-12-26:
Hi, i have found that modemband wrongly detects bands of EP06-E

Maybe it can help...

4IceG, 2022-12-26:
Hi @GektorUA ,
I will definitely write about help because I do not have such a modem, and the configuration of the router itself can also be important.

Why wrong? Which bands do not match?
Do you have any command outputs to verify?

GektorUA, 2022-12-26:
I did not touch factory bands of EP06-E, but modemband show that B32 is not supported:

4IceG, 2022-12-26:
It may be that the same modem model in different areas, with different operators, has different settings for the supported bands.

That's why I added the ability to edit the modem settings template to the package, just go to the "Modem settings template" tab and add the missing band to the "_DEFAULT_LTE_BANDS=" variable, save and that's it.

GektorUA, 2022-12-27:
I can't edit, and even get open that page, it hangs forever on "Waiting to read data from modem".

4IceG, 2023-01-06:
I did an update, it should be better.

4IceG, 2023-01-09:
Hi @GektorUA , please check if the version posted here will correctly show SINR.

You have a very non-standard configuration and packages go crazy because they have defined ports for communication with the modem, only that these settings do not match your configuration.

Do you have a list on which ports which modem communicates?
Send a command to change/adjust the port:
```
uci set modemband.@modemband[0].set_port="/dev/ttyUSB7"
uci commit modemband
```

LuCI apps modemband/3ginfo-lite show this message if they can't communicate with the modem, they can't read the data.
This happens for two reasons, either the modem is busy (also by other applications), or the port for communication with the modem is set incorrectly.

Ok, show me the output of the command:
`sms_tool -d /dev/ttyUSB7 at "at+qnwinfo"`

[luci-app-3ginfo-lite_1.0.35-20230109_TEST.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/10370494/luci-app-3ginfo-lite_1.0.35-20230109_TEST.zip)

4IceG, 2023-01-09:
Ok, It will probably soon come out that the script cannot read the modem correctly, so the settings editor correctly shows that it is waiting for the modem.

The output of the commands will show what is happening
```
chmod +x /usr/bin/loaded.sh
/usr/bin/loaded.sh json
```
I will probably have to modify the package for LuCI to get around such situations.

4IceG, 2023-01-09:
Ok, I'm sending another version for testing.

[luci-app-3ginfo-lite_1.0.36-20230109_TEST2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/10373624/luci-app-3ginfo-lite_1.0.36-20230109_TEST2.zip)

4IceG, 2023-01-09:
I know modemband is down, so I sent commands to make sure the script had a problem and the package would be fixable in the future.

3ginfo-lite already working ok? because I don't know whether to add these changes to the package.

4IceG, 2023-01-18:
> > because I don't know whether to add these changes to the package.
> 
> Did you add it in package? I have build fresh 3ginfo-lite and seems that it still with error

Hi @GektorUA 

Use the private version I sent you. I haven't updated yet, everything has to wait. I'm alone, I don't always have time, and there are many users who would like to have support.

4IceG, 2023-01-25:
Ok, I did update the package sources.

closed 2023-01-25 by 4IceG