4IceG/luci-app-3ginfo-lite #28: Add support for multiple modems

GektorUA, 2022-12-18:
Is there any chance that 3ginfo get support for multiple modems at one time?
I have system with 4 modems, mixed types (EP06-E, 3372 and etc), and to get information, i need every time change and save settings and go to information tab.
It will be good to make possibility to setup few configurations and make possible switch modems information with multiple tabs without need to change settings.

4IceG, 2022-12-18:
I don't have a multimodem installation so I never thought about it and I didn't plan on it.
At the moment I want to rewrite the luci-app-sms-tool application to Luci JS. That's what I'd like to do in my free time.

pedroo84, 2023-06-26:
I have a similar setup, would be awesome to have  support for multiple modems at once.

4IceG, 2023-06-27:
I'm planning to make a package that will make it easier to configure the modem(s). 

The principle would be as follows, the user defines the modem/modems in the system and my other packages use that configuration. 

[image]

I'm just learning Javascript and LuCi JS, so I don't know if anything will come out of my concept. The idea seems good, it wouldn't be stupid if it can be described well in code.

4IceG, 2023-07-04:
@GektorUA @pedroo84 
I made something like this, I invite to test and feedback.
[test.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/11947335/test.zip)

pedroo84, 2023-07-04:
Hello @4IceG, thank you for that! 

I'm testing it now, I've successfully installed and was able to navigate to the defined modems tab. There I was able to add two modems just fine. 

I still have to go to 3Ginfo-Lite configuration page to add at least one of the modems in order to get to the status page and be able to switch between the two "defined modems" 

But so far it looks very good. 

I wonder if its possible to have a simpler tab with only the general information section for each modem, side by side, all together   

Thanks again.

4IceG, 2023-07-05:
Hi @pedroo84 
> I still have to go to 3Ginfo-Lite configuration page to add at least one of the modems in order to get to the status page and be able to switch between the two "defined modems"

I will add it in the next version to make 3ginfo-lite search for the settings of the first defined modem at the start.

> But so far it looks very good.

I wanted to add the simplest option to change the modem so that I don't have to rewrite the whole package again.

> I wonder if its possible to have a simpler tab with only the general information section for each modem, side by side, all together

With modems defined, someone would have to write a completely new package with an extensive json object handling all the data. Good luck 😄

pedroo84, 2023-08-22:
Hello @4IceG how are you?

I'm trying to install 3ginfo-lite to a new system and I want the "defined modems" feature to be available, unfortunately I'm not able to use the "test" file you sent previously here, since now 3ginfo-lite is on a newer version than the "test" file.

Is there any ways I can downgrade to that particular version? or maybe you could incorporate "defined Modems" to the oficial releases? It was working great for me.

thanks

4IceG, 2023-08-22:
Hi @pedroo84 , 
I'm just waiting for cooler days. 

Install the latest versions of packages, someone has to test them, and if something does not work, I will fix it with a new version of the package.

pedroo84, 2023-08-29:
Hello @4IceG Thanks for the reply.

I have installed the latest version using "opkg install luci-app-3ginfo-lite"

[image]

But after installing I don't get the "defined modems" tab available on Luci, any idea how to solve this?

[image]

4IceG, 2023-08-29:
Hi @pedroo84 ,
Install the latest versions of packages luci-app-modemdefine and luci-app-3ginfo-lite. If the package is not visible in the menu, log out and clear the browser cache.

pedroo84, 2023-08-29:
Hi @4IceG 

I'm getting this error when trying to "opkg install luci-app-modemdefine" 

[image]

Maybe I shall remove everything and install again? just not sure how to proceed this.

thanks

4IceG, 2023-08-29:
Yes, remove old packages before installing new ones. Configuration files are not compatible between these versions.

pedroo84, 2023-08-29:
Sorry @4IceG I'm really struggling here. 
I've got it all removed, but when I install again I get this:
[image]

Modem Information menu shows up, but defined modems won't show.

Any ideas?

4IceG, 2023-08-29:
Uninstall the packages again and execute commands in the console:
```
rm /etc/config/modemdefine
rm /etc/config/3ginfo
```
Install the packages again.

pedroo84, 2023-08-29:
Great, that worked. Thanks a lot

4IceG, 2023-08-29:
So I already know that the management of configuration files needs to be improved in the next version of the packages.

riccardoferti, 2023-12-28:
Would you like to add support for Foxconn T99W175 also known as Dell DW5930e? 
Device connected to board with PCI MHI and AT port available at /dev/wwan0at0, MBIM at /dev/wwan0mbim0 and QCDM at /dev/wwan0qcdm0

AT Commands available here:
https://www.codico.com/media/productattach/m/v/mv31_w_atc_v00058_2111538_1.pdf

by now only this info are shown
[image]

closed 2024-04-20 by 4IceG