4IceG/luci-app-3ginfo-lite #30: 3gdetail: JSON.parse: unexpected end of data

Jalakas, 2022-12-26:
Page "http://192.168.1.1/cgi-bin/luci/admin/modem/3ginfo-lite/3gdetail"
has error: "SyntaxError
JSON.parse: unexpected end of data at line 1 column 1 of the JSON data"

Git commit: https://github.com/4IceG/luci-app-3ginfo-lite/commit/f5a7f5768081e3eb08af9db6e4400961329af5fa 
Modem: Huawei E3372
3ginfo works fine in ncm mode.

Jalakas, 2022-12-26:
Caused by request:
"/usr/share/3ginfo-lite/3ginfo.sh json"
having error:
"
root@OpenWrt:~# /usr/share/3ginfo-lite/3ginfo.sh json
/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/3ginfo-addon/12d11506: line 89: arithmetic syntax error
"

4IceG, 2022-12-26:
Hi @Jalakas , Ok thanks for the hint.
I'll try to fix it as soon as I get some free time.

4IceG, 2023-01-06:
I did an update, it should be better.

4IceG, 2023-01-25:
It should work ok now (package waiting to be compiled).

closed 2023-01-25 by 4IceG

Jalakas, 2023-02-04:
I have now again access to this device and the JSON error is still present, now caused by line https://github.com/4IceG/luci-app-3ginfo-lite/blob/main/luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/3ginfo-addon/12d11506#L93:
`/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/3ginfo-addon/12d11506: line 93: arithmetic syntax error`

Test in line https://github.com/4IceG/luci-app-3ginfo-lite/blob/main/luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/3ginfo-addon/12d11506#L86 seems to fail, because variable "$XFR" has non empty value "at^hfreqinfo?".

Jalakas, 2023-02-04:
This will fix JSON error:
if [ ! -z "$TDL" ]; then                                                                
DL=$(($TDL/1000))                                                                       
fi                                                                             
                                                                                        
TUL=$(echo "$TBX" | cut -d ',' -f9-9 )                                                  
if [ ! -z "$TUL" ]; then                                                                
UL=$(($TUL/1000))                                                                       
fi
[image]

4IceG, 2023-02-04:
I tested these changes and had no errors. The command returns different values (not always empty) on different modems.

@Jalakas show the result of this command on your modem
`at^hfreqinfo?`

Like now, it can't be either... why should we show "nothing", it's better to hide it completely.

Jalakas, 2023-02-04:
Huawei E3272 modem, using sms_tool:
```
root@OpenWrt:~# sms_tool -d /dev/ttyUSB1 at 'at^chiptemp?'
at^chiptemp?
^CHIPTEMP: 373,373,65535,28,65535

root@OpenWrt:~# sms_tool -d /dev/ttyUSB1 at 'at^hfreqinfo?'
at^hfreqinfo?
root@OpenWrt:~#
```

4IceG, 2023-02-04:
OK thanks. I'll be doing package updates this week, so I'll fix this situation.

Jalakas, 2023-02-25:
As far I see, the fix has not been released yet?

4IceG, 2023-02-25:
I haven't tested it with a modem, but I've added some code that checks the output of the command, and it's supposed to work.
```
XFR=$(sms_tool -d $DEVICE at "at^hfreqinfo?")
XFRCOUNT=$(echo -n "$XFR" | wc -c)

if [ -n "$XFR" ] && [ "$XFRCOUNT" -gt "15" ]; then
```

Jalakas, 2023-02-25:
Tested this patch and it works fine:
[image]