4IceG/luci-app-3ginfo-lite #33: Huawei E3372H (stick): 12d11506: line 88: syntax error: unexpected "else"

patrakov, 2023-01-21:
```
root@OpenWrt:/tmp# /usr/share/3ginfo-lite/3ginfo.sh 
/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/3ginfo-addon/12d11506: line 88: syntax error: unexpected "else"
root@OpenWrt:/tmp# cat /etc/config/3ginfo 

config 3ginfo
	option pbwidth '33'
	option website 'https://lteitaly.it/internal/map.php#bts='
	option network 'wwan0'
	option device '/dev/ttyUSB0'

```

The file in question contains:

```
XFR=$(sms_tool -d $DEVICE at "at^hfreqinfo?")
if [ -z "$XFR" ]
then
      # Huawei E3272 
else
...
fi
# DL / UL / Band
```

The problem is that the shell requires at least one command between "then" and "else", and your code only has a comment, which is not a command. Adding a ":" on a line by itself helps.

4IceG, 2023-01-21:
Ok, thanks @patrakov for reporting the bug. I see that I have to check next time on a working modem, because the interpreter can be wrong. I'll fix it in the next release.

4IceG, 2023-01-25:
It should be ok now, I changed the logic of the condition.

P.S.
I'm waiting to compile the package from sources.

closed 2023-01-25 by 4IceG