4IceG/luci-app-3ginfo-lite #34: Add support for Quectel RM520N-GL

lrdrdn, 2023-01-21:
Please add support [luci-app-3ginfo-lite](https://github.com/4IceG/luci-app-3ginfo-lite) & [luci-app-modemband](https://github.com/4IceG/luci-app-modemband)

Quectel RM520N-GL
vid:pid  `2c7c:0801`

All command same as Quectel RG502Q-EA 
serial port `/dev/ttyUSB3`
Supported band:
```
at+qnwprefcfg="lte_band"
+QNWPREFCFG: "lte_band",1:2:3:4:5:7:8:12:13:14:17:18:19:20:25:26:28:29:30:32:34:38:39:40:41:42:43:48:66:71
```

4IceG, 2023-01-21:
Hi @lrdrdn 
If it's the same, I'll add the modem in the next package release.

lrdrdn, 2023-01-22:
I copied file 2c7c0800 to 2c7c0801, but  modem type & protocol not display properly.

[image]

4IceG, 2023-01-22:
Commands similar, but scripts need fixing anyway. Ok, I will write here when I add the modem.

4IceG, 2023-01-23:
Hi @lrdrdn , replace with this file and see if it will be better.
[2c7c0801.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/10479763/2c7c0801.zip)

lrdrdn, 2023-01-23:
working correctly
[image]

4IceG, 2023-01-25:
The modem is already added to 3ginfo-lite, with the compilation I am waiting for the update of the modemband package by Cezary.

closed 2023-01-25 by 4IceG

pedroo84, 2023-07-25:
Hello guys, recently I got a RM520N-GL, but it is locked to "mbim" this way I can't access the AT commands to change it to ECM, any idea how can I do this? in order to have read outs on 3g-info?

4IceG, 2023-07-25:
Hi @pedroo84 , I don't know in the end if you want to use mbim or ecm modem. My modem works in mbim and I'm not complaining. Do you have the kmod-usb-serial kmod-usb-serial-option packages installed?

pedroo84, 2023-07-25:
> Hi @pedroo84 , I don't know in the end if you want to use mbim or ecm modem. My modem works in mbim and I'm not complaining. Do you have the kmod-usb-serial kmod-usb-serial-option packages installed?

Hi Rafal, actually mbim works with modem manager here too, the problem is that the USBtty port won't show up on my modem, therefore I can't access AT Command and also 3G-Info can't read the data too. Do you think my module needs a firmware update to solve this? 

I own other Quectel modems and never encountered such issue.

4IceG, 2023-07-25:
First of all, uninstall modem manager. Then check if you have the required packages listed above. Finally, set the modem to work in mbim and configure the connection. Important! my packages don't work with MM.