4IceG/luci-app-3ginfo-lite #35: SyntaxError : Bad control character in string literal in JSON at position 52

kuyay, 2023-01-29:
Bad control character in string literal in JSON at position 52

root/usr/share/3ginfo-lite/3ginfo-addon/2c7c0801

# Quectel RM520N-GL

4IceG, 2023-01-29:
Hi @kuyay ,
Thanks for reporting the bug. Show me the result of the command:
`sh -x /usr/share/3ginfo-lite/3ginfo.sh`

kuyay, 2023-01-29:
```
root@FriendlyWrt:~# sh -x /usr/share/3ginfo-lite/3ginfo.sh
+ RES=/usr/share/3ginfo-lite
+ uci -q get '3ginfo.@3ginfo[0].device'
+ DEVICE=/dev/ttyUSB0
+ '[' x/dev/ttyUSB0 '=' x ]
+ echo /dev/ttyUSB0
+ '[' x/dev/ttyUSB0 '=' x ]
+ '[' x/dev/ttyUSB0 '=' x ]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=@modem
+ SUB=@
+ '[[' @modem '==' '*@*' ]]
+ echo @modem
+ sed s/@//
+ SEC=modem
+ '[' -z modem ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus modem
+ grep '"up": true'
+ NETUP='       "up": true,'
+ '[' -n '      "up": true,' ]
+ uci -q -P /var/state/ get network.modem.connect_time
+ CT=
+ '[' -z ]
+ ifstatus modem
+ awk '-F[:,]' '/uptime/ {print $2}'
+ xargs
+ CT=2604
+ '[' '!' -z 2604 ]
+ expr 2604 / 60 / 60 / 24
+ D=0
+ expr 2604 / 60 / 60 '%' 24
+ H=0
+ expr 2604 / 60 '%' 60
+ M=43
+ expr 2604 '%' 60
+ S=24
+ printf '%dd, %02d:%02d:%02d' 0 0 43 24
+ CONN_TIME='0d, 00:43:24'
+ ifstatus modem
+ awk '-F"' '/l3_device/ {print $4}'
+ IFACE=wwan0
+ '[' -n wwan0 ]
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='1.1 GiB'
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='471.9 MiB'
+ sms_tool -D -d /dev/ttyUSB0 at 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
No response from modem.
+ O=
+ echo
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=
+ '[' x '=' x ]
+ CSQ=-1
+ '[' -1 -ge 0 -a -1 -le 31 ]
+ CSQ=-
+ CSQ_PER=0
+ echo
+ awk '-F["]' '/^\+COPS: .,2/ {print $2}'
+ COPS_NUM=
+ '[' x '=' x ]
+ COPS_NUM=-
+ COPS_MCC=-
+ COPS_MNC=-
+ '[' x '=' x ]
+ COPS=-
+ '[' -z  ]
+ echo
+ awk '-F["]' '/^\+COPS: .,0/ {print $2}'
+ T=
+ '[' x '!=' x ]
+ echo -
+ sed ':s;s/\(\<\S*\>\)\(.*\)\<\1\>/\1\2/g;ts'
+ COPZ=-
+ echo -
+ awk '{for(i=1;i<=NF;i++){ $i=toupper(substr($i,1,1)) substr($i,2) }}1'
+ COPS=-
+ echo
+ + xargsawk
 '-F[, ]' '/^\+CPIN:/ {print $0;exit}'
+ T=
+ '[' -n  ]
+ echo
+ awk '-F[, ]' '/^\+CME ERROR:/ {print $0;exit}'
+ T=
+ '[' -n  ]
+ echo
+ awk '-F[,]' '/^\+CREG/ {gsub(/[[:space:]"]+/,"");printf "T=\"%d\";LAC_HEX=\"%X\";CID_HEX=\"%X\";LAC_DEC=\"%d\";CID_DEC=\"%d\";MODE_NUM=\"%d\"", $2, "0x"$3, "0x"$4, "0x"$3, "0x"$4, $5}'
+ eval
+ REG=-
+ CSQ=-
+ CSQ_PER=0
+ '[' -z  ]
+ echo
+ awk '-F[,]' '/^\+COPS/ {print $4;exit}'
+ MODE_NUM=
+ MODE=-
+ sms_tool -d /dev/ttyUSB0 at at+cereg

```

4IceG, 2023-01-29:
Okay, a few things...

Are you using ModemManager? (My package is not compatible with it)

You also have my package misconfigured. Execute the command
```
uci set 3ginfo.@3ginfo[0].device="/dev/ttyUSB3"
uci commit 3ginfo
```

kuyay, 2023-01-29:
Yes sir, i'm using modemmanager, thanks for your answers!

Can you give me some advice, what can I use to connect to my modem (beside ModemManager)?

4IceG, 2023-01-29:
It would be best to uninstall MM (which takes control of the modem and makes it difficult for other packages to access the modem) and configure the connection using a different protocol, such as qmi.

kuyay, 2023-01-29:
alright sir, thanks.

closed 2023-03-12 by 4IceG