4IceG/luci-app-3ginfo-lite #37: Obsługa modemu DW5811e Snapdragon™ X7 LTE

przemekp1, 2023-01-29:
Cześć, 
czy ten pakiet obsługuje modem Dell DW5811e
Pracuję  nim od dwóch lat, działkiem całkiem nieźle - poprzez adpter

https://allegro.pl/oferta/adapter-usb-modemu-lte-ngff-m-2-sim-2x-antena-9278038018

cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.10
S:  Manufacturer=Linux 5.10.161 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.10
S:  Manufacturer=Linux 5.10.161 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=413c ProdID=81b6 Rev= 0.06
S:  Manufacturer=Sierra Wireless, Incorporated
S:  Product=DW5811e Snapdragon™ X7 LTE
S:  SerialNumber=LF61544850011013
C:* #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=504mA
A:  FirstIf#=12 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00
I:* If#=12 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=32ms
I:  If#=13 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
I:* If#=13 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms

4IceG, 2023-01-30:
Siema, z tego co widze to 3ginfo-lite coś by pokazało, ale niewiele więc modem jest do dodania/uzupełnienia o brakujace dane.

4IceG, 2023-02-05:
Cześć @przemekp1 , będę chciał w tym tygodniu zrobić aktualizację pakietu.

Jak możesz to podeślij mi wynik tych poleceń
```
AT+TEMP?
AT+CGMI
AT+CGMM
AT+CGMR
AT+ICCID
AT+CGSN
AT+CIMI 
```
to uzupełnię brakujące rzeczy dla tego modemu.

przemekp1, 2023-03-04:
nie potrafię tego wywołać przy mbim
ale znalazłe stronę grze to wszystko jest - opsiany ten modem
ja go używam dokładnie przez taki adapter kupiony na Aliexpress
USB To NGFF M.2 Key B Adapter Enclosure With SIM Card Slot
https://github.com/danielewood/sierra-wireless-modems

4IceG, 2023-03-05:
Siema @przemekp1 
Ja tą stronę znam i nie ma tam tych danych, o które prosiłem, więc takie odsyłanie mnie do internetów stawia sprawę w miejscu.
Ty masz modem i możesz z nim gadać, a wysłanie tych poleceń i podesłanie mi pełnej odpowiedzi, tak abym widział całą składnię to nie wiem czy 5 minut roboty, tym bardziej, że pakietów od poleceń at nie brakuje.

obsy, 2023-03-21:
Przy takim ustawieniu modemu nie masz interfejsów szeregowych żeby odpytać modem, a przez mbim praktycznie nie ma żadnych użytecznych danych.
Jak byś przełączył modem w inny tryb który udostępnia np. mbim + serial bo by było jak spytać o dane, bo 3giinfo już potrafi ten modem odpytać i wyświetlić dane.

jfrogg, 2023-03-25:
@przemekp1, DW5811e is a Dell branded version of Sierra EM7455, this modem is already supported.
What you need to do:
- switch it to another composition where it will have MBIM (or QMI) and a number of serial interfaces
- de-brand it or make it generic
- upgrade the modem firmware

As already noted, 3ginfo needs a serial interface (/dev/ttyUSBx) to communicate with the modem.

przemekp1, 2023-07-04:
ok, ale jak zrobić aby ten modem przełączyć ten port z „cdc-wdm0” na „ttyUSB0” ?
może ktoś mi podać komendy - jak dla laika

obsy, 2023-07-05:
Dla samego mbim nie będziesz miał portów szeregowych. Przełącz go w inny tryb to się pojawią - służy do tego np. swi_usbcomp.pl  a opis masz np. tutaj: https://zukota.com/posts/sierra-wireless-em7455-how-to-enable-com-ports/

przemekp1, 2023-07-05:
zmiana tego parametru
Current USB composition: 8
nie pomogła

Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.515727] usbcore: registered new interface driver cdc_wdm
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.530032] Loading modules backported from Linux version v6.1.24-0-g0102425ac76b
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.545039] Backport generated by backports.git v5.15.92-1-44-gd6ea70fafd36
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.583746] usbcore: registered new interface driver cdc_ether
Wed Jul  5 20:12:07 2023 kern.notice kernel: [   13.595006] random: crng init done
Wed Jul  5 20:12:07 2023 kern.notice kernel: [   13.602232] random: 31 urandom warning(s) missed due to ratelimiting
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.617818] usbcore: registered new interface driver cdc_ncm
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.830321] pci 0000:00:00.0: enabling device (0000 -> 0003)
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.841655] mt7615e 0000:01:00.0: enabling device (0000 -> 0002)
Wed Jul  5 20:12:07 2023 kern.debug kernel: [   13.867941] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.872830] pci 0000:00:01.0: enabling device (0000 -> 0003)
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.884284] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.884400] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.884400]
Wed Jul  5 20:12:07 2023 kern.debug kernel: [   13.911420] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
Wed Jul  5 20:12:07 2023 kern.debug kernel: [   13.915473] ieee80211 phy1: copying sband (band 1) due to VHT EXT NSS BW flag
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.933153] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.933153]
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.973812] PPP generic driver version 2.4.2
Wed Jul  5 20:12:07 2023 kern.info kernel: [   13.984429] NET: Registered PF_PPPOX protocol family
Wed Jul  5 20:12:07 2023 kern.info kernel: [   14.044645] usbcore: registered new interface driver qmi_wwan
Wed Jul  5 20:12:07 2023 kern.info kernel: [   14.061416] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
Wed Jul  5 20:12:07 2023 kern.info kernel: [   14.077113] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
Wed Jul  5 20:12:07 2023 kern.info kernel: [   14.113941] cdc_mbim 2-1:1.12: cdc-wdm0: USB WDM device
Wed Jul  5 20:12:07 2023 kern.info kernel: [   14.125961] cdc_mbim 2-1:1.12 wwan0: register 'cdc_mbim' at usb-1e1c0000.xhci-1, CDC MBIM, c6:df:d2:86:5b:e9
Wed Jul  5 20:12:07 2023 kern.info kernel: [   14.146271] usbcore: registered new interface driver cdc_mbim

[   13.515727] usbcore: registered new interface driver cdc_wdm
[   13.530032] Loading modules backported from Linux version v6.1.24-0-g0102425ac76b
[   13.545039] Backport generated by backports.git v5.15.92-1-44-gd6ea70fafd36
[   13.583746] usbcore: registered new interface driver cdc_ether

przemekp1, 2023-07-05:
katalog /dev/
bus/             mtd3             ppp              ttyS2
cdc-wdm0         mtd3ro           ptmx             ttyS3
console          mtd4             pts/             ttyS4
cpu_dma_latency  mtd4ro           random           ttyS5
full             mtd5             shm/             ttyS6
gpiochip0        mtd5ro           stderr           ttyS7
gpiochip1        mtd6             stdin            ttyS8
gpiochip2        mtd6ro           stdout           ttyS9
hwrng            mtdblock0        tty              ubi0
kmsg             mtdblock1        ttyS0            ubi0_0
log              mtdblock2        ttyS1            ubi0_1
mtd0             mtdblock3        ttyS10           ubi_ctrl
mtd0ro           mtdblock4        ttyS11           ubiblock0_0
mtd1             mtdblock5        ttyS12           urandom
mtd1ro           mtdblock6        ttyS13           watchdog
mtd2             null             ttyS14           watchdog0
mtd2ro           port             ttyS15           zero

obsy, 2023-07-05:
I pokaż teraz wynik
```
cat /sys/kernel/debug/usb/devices
```

przemekp1, 2023-07-05:
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.118 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.15
S:  Manufacturer=Linux 5.15.118 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=413c ProdID=81b6 Rev= 0.06
S:  Manufacturer=Sierra Wireless, Incorporated
S:  Product=DW5811e Snapdragon™ X7 LTE
S:  SerialNumber=LF61544850011013
C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=504mA
A:  FirstIf#=12 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=83(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=85(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#=12 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim
E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=32ms
I:  If#=13 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
I:* If#=13 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
E:  Ad=86(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms

obsy, 2023-07-05:
Wiec pomogło, nie to co napisałeś wczesniej. Tylko nie masz zainstalowanych sterowników, zainstaluj sobie kmod-usb-serial* i zobacz który z nich to złapie.

przemekp1, 2023-07-05:
po doinstalowaniu sterowników

Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.448231] usbserial: USB Serial support registered for GSM modem (1-port)
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.467338] usbcore: registered new interface driver qcserial
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.479188] usbserial: USB Serial support registered for Qualcomm USB modem
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.494150] qcserial 2-1:1.0: Qualcomm USB modem converter detected
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.507240] usb 2-1: Qualcomm USB modem converter now attached to ttyUSB0
Wed Jul  5 23:56:19 2023 kern.err kernel: [   14.521497] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep3in intr, mpkt:10, interval:9/32ms
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.540463] qcserial 2-1:1.2: Qualcomm USB modem converter detected
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.553643] usb 2-1: Qualcomm USB modem converter now attached to ttyUSB1
Wed Jul  5 23:56:19 2023 kern.err kernel: [   14.567941] xhci-mtk 1e1c0000.xhci: drop_ep_quirk super-speed ep5in intr, mpkt:10, interval:9/32ms
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.586726] qcserial 2-1:1.3: Qualcomm USB modem converter detected
Wed Jul  5 23:56:19 2023 kern.info kernel: [   14.599739] usb 2-1: Qualcomm USB modem converter now attached to ttyUSB2
Wed Jul  5 23:56:19 2023 user.info kernel: [   14.614306] kmodloader: done loading kernel modules from /etc/modules.d/*

ale luci nadal nie widzi portów usb, np przy konfiguracji interfejsu QMI - nadal tylko /dev/cdc-wdm0

obsy, 2023-07-06:
Do qmi używa właśnie interfejsów /dev/cdc* a nie /dev/ttyUSB*

Masz interfejsy, więc teraz możesz używać 3ginfo do wizualizacji parametrów. Wątek można zamykać.

przemekp1, 2023-08-02:
poszło /dev/ttyUSB2
nie pokazuje tylko danych z pól
<html><body>

Modem type |
Revision / Firmware

[image]

a w zakładce Configuration muszę mieć takie ustawienie bo w innych przypasdach wywala mi błąd

[image]
modem pracuje na takich ustawieniach
[image]

[image]

przemekp1, 2023-08-02:
zakładka USB Debug

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.120 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.15
S:  Manufacturer=Linux 5.15.120 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=413c ProdID=81b6 Rev= 0.06
S:  Manufacturer=Sierra Wireless, Incorporated
S:  Product=DW5811e Snapdragon™ X7 LTE
S:  SerialNumber=LF61544850011013
C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=504mA
A:  FirstIf#=12 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=qcserial
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=qcserial
E:  Ad=83(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=qcserial
E:  Ad=85(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#=12 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim
E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=32ms
I:  If#=13 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
I:* If#=13 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
E:  Ad=86(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms

4IceG, 2023-08-03:
Pięknie, czyli coś się ruszyło. Jest komunikacja z modemem więc podeślij mi wynik tych poleceń at:
```
AT+CGMI
AT+CGMM
AT+CGMR
AT+ICCID
AT+CGSN
AT+CIMI
```

przemekp1, 2023-08-03:
dobrze to robię? bo nic mi się nie wyświetla
root@OpenWrt:~# socat - /dev/ttyUSB2
AT+CGMI
AT+CGMI

4IceG, 2023-08-03:
@przemekp1 nie wiem po co instalowałeś dodatkowo socat, polecenia możesz wysłać za pomocą sms_tool
`sms_tool -d /dev/ttyUSB2 at "AT+CGMI"`

Nie powiedziane, że uzyskasz na nie odpowiedź, dlatego chciałem abyś posprawdzał co działa i na co odpowie modem.

przemekp1, 2023-08-03:
a nie wiedziałem tego :)

AT+CGMI
Sierra Wireless, Incorporated

AT+CGMM
EM7455B

AT+CGMR
SWI9X30C_02.24.05.06 r7040 CARMD-EV-FRMWR2 2017/05/19 06:23:09

AT+ICCID
ICCID: 8948032026463302116
AT+CGSN
353990070926303
AT+CIMI
260036446330211

4IceG, 2023-08-05:
@przemekp1
Rozpakuj archiwum [test.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12266010/test.zip) i skopiuj plik na router do tej lokalizacji `usr/share/3ginfo-lite/modem/`

Jeżeli brakujące dane uzupełnią się poprawnie to przygotuję aktualizację pakietu.

przemekp1, 2023-08-05:
jest problem po podmiance
[image]

4IceG, 2023-08-05:
Ok, czyli coś pomerdałem. Pokaz wynik polecenia:
sh -x /usr/share/3ginfo-lite/3ginfo.sh

przemekp1, 2023-08-05:
root@OpenWrt:~# sh -x /usr/share/3ginfo-lite/3ginfo.sh
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB2
+ '[' -z /dev/ttyUSB2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS                                                                             ?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 19,99

+COPS: 0,0,"Orange nju",7

+COPS: 0,2,"26003",7

+CREG: 2,1,"FFFE","23E3E34",7

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=wan
+ SUB=@
+ '[[' wan '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wan
+ '[' -z wan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wan
+ grep '"up": true'
+ NETUP=
+ '[' -n  ]
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 19,99

+COPS: 0,0,"Orange nju",7

+COPS: 0,2,"26003",7

+CREG: 2,1,"FFFE","23E3E34",7

'K
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=19
+ '[' x19 '=' x ]
+ '[' 19 -ge 0 -a 19 -le 31 ]
+ CSQ_PER=61
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 19,99

+COPS: 0,0,"Orange nju",7

+COPS: 0,2,"26003",7

+CREG: 2,1,"FFFE","23E3E34",7

'K
+ awk '-F["]' '/^\+COPS: .,2/ {print $2}'
+ COPS_NUM=26003
+ '[' x26003 '=' x ]
+ COPS_MCC=260
+ COPS_MNC=03
+ awk '-F[;]' '/26003/ {print $2}' /usr/share/3ginfo-lite/mccmnc.dat
+ COPS=Orange
[... 493 lines cut ...]
"tac_h":"D5AC",
"tac_d":"54700",
"cid_dec":"37633588",
"cid_hex":"23E3E34",
"pci":"",
"earfcn":"",
"pband":"B3 (1800 MHz) @10 MHz",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-109/-117",
"rsrq":"-17.4",
"rssi":"-74/-74",
"sinr":"-2.6"
}
+ exit 0

4IceG, 2023-08-05:
Ok, podejście nr 2.
[test2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12266756/test2.zip)

przemekp1, 2023-08-05:
[image]

przemekp1, 2023-08-05:
sms_tool -d /dev/ttyUSB2 at "at!gstatus?"
at!gstatus?
!GSTATUS:
Current Time:  647519           Temperature: 37
Reset Counter: 1                Mode:        ONLINE
System mode:   LTE              PS state:    Attached
LTE band:      B3               LTE bw:      10 MHz
LTE Rx chan:   1749             LTE Tx chan: 19749
LTE CA state:  NOT ASSIGNED
EMM state:     Registered       Normal Service
RRC state:     RRC Connected
IMS reg state: No Srv

PCC RxM RSSI:  -75              RSRP (dBm):  -109
PCC RxD RSSI:  -74              RSRP (dBm):  -118
Tx Power:      0                TAC:         D5AC (54700)
RSRQ (dB):     -17.2            Cell ID:     023E3E34 (37633588)
SINR (dB):     -2.8

4IceG, 2023-08-05:
Czyli bawimy się dalej. Później jeszcze na to zerkne.

4IceG, 2023-08-06:
Odpalam u siebie te skrypty i błędu nie mam, więc ten modem coś jeszcze zwraca co rozwala system. 

Ok.. wróciłem do tej pierwszej wersji skryptów bo one pokazywały najwięcej, poprawiłem to co rzuciło mi się w oczy i tyle. Jak dalej będzie zonk to nic tylko znów pozostaje szukać co może być nie halo.

[413c81b6.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12269751/413c81b6.zip)

@przemekp1 jak wrzucisz tą wersję to pokaż wynik polecenia
`/usr/share/3ginfo-lite/3ginfo.sh`

przemekp1, 2023-08-06:
nadal te same komunikaty
[image]

root@OpenWrt:~# /usr/share/3ginfo-lite/3ginfo.sh
/usr/share/3ginfo-lite/3ginfo.sh: line 193: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
{
"connt":"0d, 03:05:52",
"conntx":"25.0 MiB",
"connrx":"769.2 MiB",
"modem":"Sierra Wireless EM7455B",
 &deg;C", 37
"firmware":"SWI9X30C_02.24.05.06 r7040 CARMD-EV-FRMWR2 2017/05/19 06:23:09",
"cport":"/dev/ttyUSB2",
"protocol":"mbim",
"csq":"16",
"signal":"51",
"operator_name":"Orange Nju",
"operator_mcc":"260",
"operator_mnc":"03",
"mode":"UMTS",
"registration":"1",
"simslot":"",
"imei":"353990070926303",
"imsi":"260036446330211",
"iccid":"8948032026463302116",
"lac_dec":"54700",
"lac_hex":"D5AC",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"54415928",
"cid_hex":"33E5238",
"pci":"",
"earfcn":"",
"pband":"",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":""
}

4IceG, 2023-08-06:
@przemekp1 masz mozliwosc zmienić, poprawic lokalizacje routera? Skrypty operuja na danych z polaczenia lte / lte-a a u Ciebie widze UMTS, pewnie skladnia wyglada wtedy inaczej i rozwala to cale dane wyjsciowe.

Ogolnie skrypty dane zczytuja, to czego brakowalo wczesniej jest uzupelnione. Wypadaloby złapać wynik polecenia przy  UMTS i porównać ze skladnia dla lte, pewnie by się doszlo co rozwala json-a.

przemekp1, 2023-08-06:
a teraz jest w LTE

  /usr/share/3ginfo-lite/3ginfo.sh
/usr/share/3ginfo-lite/3ginfo.sh: line 193: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
{
"connt":"0d, 00:01:10",
"conntx":"159.6 KiB",
"connrx":"4.1 MiB",
"modem":"Sierra Wireless EM7455B",
 &deg;C", 35
"firmware":"SWI9X30C_02.24.05.06 r7040 CARMD-EV-FRMWR2 2017/05/19 06:23:09",
"cport":"/dev/ttyUSB2",
"protocol":"mbim",
"csq":"18",
"signal":"58",
"operator_name":"Orange Nju",
"operator_mcc":"260",
"operator_mnc":"03",
"mode":"LTE B3 (1800 MHz)",
"registration":"1",
"simslot":"",
"imei":"353990073024155",
"imsi":"260036446330211",
"iccid":"8948032026463302116",
"lac_dec":"65534",
"lac_hex":"FFFE",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"D5AC",
"tac_d":"54700",
"cid_dec":"37633588",
"cid_hex":"23E3E34",
"pci":"",
"earfcn":"",
"pband":"B3 (1800 MHz) @10 MHz",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-113",
"rsrq":"-19.3",
"rssi":"-79",
"sinr":"-5.8"
}

4IceG, 2023-08-06:
Ok, poleciała kolejna poprawka.
[413c81b6v2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12270038/413c81b6v2.zip)

Jeżeli dalej będzie miał problem z danymi to poproszę o wynik:
`sh -x /usr/share/3ginfo-lite/3ginfo.sh`

@przemekp1 ustaw też (jeżeli dalej nie będzie działać po podmianie pliku) w konfiguracji port do komunikacji z modemem `/dev/ttyUSB2` bo widzę, że to go też boli choć nie powinno.

przemekp1, 2023-08-06:
👍działa

[image]
[image]

przemekp1, 2023-08-06:
root@OpenWrt:~#  sh -x /usr/share/3ginfo-lite/3ginfo.sh
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB2
+ '[' -z /dev/ttyUSB2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 20,99

+COPS: 0,0,"Orange nju",7

+COPS: 0,2,"26003",7

+CREG: 2,1,"FFFE","23E3E34",7

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=
+ SUB=@
+ '[['  '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=
+ '[' -z  ]
+ getpath /dev/ttyUSB2
/usr/share/3ginfo-lite/3ginfo.sh: line 193: getpath: not found
+ PORIG=
+ getpath /dev/console
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
+ '[' x '=' x ]
+ uci show network
+ grep /dev/console
+ cut -f2 -d.
+ SEC=
+ '[' -n  ]
+ getpath /dev/ptmx
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
+ '[' x '=' x ]
+ uci show network
+ grep /dev/ptmx
+ cut -f2 -d.
+ SEC=
+ '[' -n  ]
+ getpath /dev/tty
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
+ '[' x '=' x ]
+ uci show network
+ grep /dev/tty
+ cut -f2 -d.
+ SEC=
+ '[' -n  ]
+ getpath /dev/ttyS0
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
+ '[' x '=' x ]
+ uci show network
+ grep /dev/ttyS0
+ cut -f2 -d.
+ SEC=
+ '[' -n  ]
+ getpath /dev/ttyS1
/usr/share/3ginfo-lite/3ginfo.sh: line 196: getpath: not found
+ '[' x '=' x ]
+ uci show network
+ grep /dev/ttyS1
+ cut -f2 -d.
+ SEC=
+ '[' -n  ]
[... 705 lines cut ...]
"tac_h":"D5AC",
"tac_d":"54700",
"cid_dec":"37633588",
"cid_hex":"23E3E34",
"pci":"",
"earfcn":"",
"pband":"B3 (1800 MHz) @10 MHz",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-112",
"rsrq":"-20.0",
"rssi":"-74",
"sinr":"-6.4"
}
+ exit 0

przemekp1, 2023-08-06:
przy tych ustawieniach
[image]

[image]

a jak jest tam pustka
[image]

to dane pojawiają się
[image]

przemekp1, 2023-08-06:
w takiej konfiguracji pokazuje ok - bez portu wan
config 3ginfo
	option website 'http://www.btsearch.pl/szukaj.php?mode=std&search='

	option device '/dev/ttyUSB2'