4IceG/luci-app-3ginfo-lite #39: L860-GL modem support

newdanceradio, 2023-02-13:
Hello dear 4IceG
Is it possible to add support for modem L860-GL
I have one, what information I need to send to add support for this modem ?

4IceG, 2023-02-13:
Hi @newdanceradio , if you have a modem it's easier to add support. Now we just need good documentation, examples of at commands and time to adapt the data read from the modem to my package.

Show command result:
```
AT+CSQ?
AT+XCESQ?
AT+RSRP?
AT+RSRQ?
AT+RSCP?
AT+XLEC?
AT+MTSM=1
AT+XMCI=1
AT+GTPKGVER?
AT+CGMI
AT+GMM?
AT+CCID
AT+CIMI
AT+CGSN
```

newdanceradio, 2023-02-14:
Thank you @4IceG for your reply.
Please, here will be output from AT commands

=============================
Output from AT+CSQ? command:
+CSQ: 15,6
OK
=============================

=================================================
Output from AT+XCESQ? command:
+XCESQ: 0,99,99,255,255,26,57,50,255,255,255,255
OK
=================================================

======================================================
Output from AT+RSRP? command:
+RSRP: 502,1820,-82.00,502,6200,-89.00,502,300,-98.00
OK
======================================================

=====================================================
Output from AT+RSRQ? command:
+RSRQ: 502,1820,-9.-50,502,6200,-10.00,502,300,-9.00
OK
=====================================================

===============================
Output from AT+RSCP? command:
OK
===============================

====================================
Output from AT+XLEC? command:
+XLEC: 0,3,5,3,5,BAND_LTE_3,0,0,0,0
OK
====================================

====================================
Output from AT+MTSM=1 command:
+MTSM: 51
OK
====================================

==========================================================================
Output from AT+XMCI=1 command:
+XMCI: 4,247,05,"0x00D5","0x0020EB17","0x01F6","0x0000071C","0x00004D6C",
"0xFFFFFFFF",59,25,50,"0x00000017","0x00000000"
+XMCI: 5,000,000,"0xFFFE","0xFFFFFFFF","0x01F6","0x00001838","0xFFFFFFFF",
"0xFFFFFFFF",52,21,255,"0x7FFFFFFF","0x00000000"
+XMCI: 5,000,000,"0xFFFE","0xFFFFFFFF","0x01F6","0x0000012C","0xFFFFFFFF",
"0xFFFFFFFF",43,24,255,"0x7FFFFFFF","0x00000000"
OK
===========================================================================

====================================================
Output from AT+GTPKGVER? command:
+GTPKGVER: "18600.5001.00.35.01.57_5026.01.001.005"
OK
====================================================

==============================================
Output from AT+CGMI command:
Fibocom
OK
==============================================

==============================================
Output from AT+GMM? command:
+GMM: "L860 LTE Module","L860"
OK
==============================================

==============================================
Output from AT+CCID command:
+CCID: 8937105220100124117F
OK
==============================================

==============================================
Output from AT+CIMI command:
247050102982906
OK
==============================================

=============================================
Output from AT+CGSN command:
352279100355589
OK
=============================================

4IceG, 2023-02-14:
Hi @newdanceradio , thanks for sending commands. If the flu doesn't break me down more, I'll send you a version of the package with this modem on Monday (to test if it works properly).

We don't have an answer to all commands. I'll think what to do with it.
Regards.

newdanceradio, 2023-02-15:
Dear @4IceG, what exactly is missing, what kind of answers are not there? maybe on this modem i can see the necessary information using other commands?

4IceG, 2023-02-15:
Hi @newdanceradio ,
The command below returned nothing:
`AT+RSCP?`
this may mean that the value is shown in some other command.

It's nothing to worry about.

4IceG, 2023-02-20:
Hi @newdanceradio 
Please install the package from my repository (I haven't updated anything, it's still in an older version, but it's OK for testing).

After installation, copy to the router the two files that I sent to the folder
`/usr/share/3ginfo-lite/3ginfo-addon/`
[l860gl.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/10781936/l860gl.zip)

newdanceradio, 2023-02-20:
> Hi @newdanceradio Please install the package from my repository (I haven't updated anything, it's still in an older version, but it's OK for testing).
>
> After installation, copy to the router the two [...]

Thank you @4IceG.
I have copied files, but after that, i have an error
[image]

4IceG, 2023-02-20:
Well, it would be too nice if it worked right away..

@newdanceradio show the result of the commands
```
/usr/share/3ginfo-lite/3ginfo.sh
sh -x /usr/share/3ginfo-lite/3ginfo.sh
```
We'll see where the script has a problem.

newdanceradio, 2023-02-20:
```
+ RES=/usr/share/3ginfo-lite
+ uci -q get '3ginfo.@3ginfo[0].device'
+ DEVICE=/dev/ttyACM2
+ '[' x/dev/ttyACM2 '=' x ]
+ echo /dev/ttyACM2
+ '[' x/dev/ttyACM2 '=' x ]
+ '[' x/dev/ttyACM2 '=' x ]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=eth1
+ SUB=@
+ '[[' eth1 '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=eth1
+ '[' -z eth1 ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus eth1
+ grep '"up": true'
+ NETUP=
+ '[' -n  ]
+ sms_tool -D -d /dev/ttyACM2 at 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=5
+ '[' x5 '=' x ]
+ '[' 5 -ge 0 -a 5 -le 31 ]
+ CSQ_PER=16
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ awk '-F["]' '/^\+COPS: .,2/ {print $2}'
+ COPS_NUM=24705
+ '[' x24705 '=' x ]
+ COPS_MCCB=247
+ echo 247
+ tr '\n' ' '
+ sed 's/ //g'
+ COPS_MCC=247
+ COPS_MNCB=05
+ echo 05
+ tr '\n' ' '
+ sed 's/ //g'
+ COPS_MNC=05
+ awk '-F[;]' '/24705/ {print $2}' /usr/share/3ginfo-lite/mccmnc.dat
+ COPS=Bite
+ '[' xBite '=' x ]
+ '[' -z  ]
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ awk '-F["]' '/^\+COPS: .,0/ {print $2}'
+ T=BITE_LV
+ '[' xBITE_LV '!=' x ]
+ COPS=BITE_LV
+ echo BITE_LV
+ sed ':s;s/\(\<\S*\>\)\(.*\)\<\1\>/\1\2/g;ts'
+ COPZ=BITE_LV
+ echo BITE_LV
+ awk '{for(i=1;i<=NF;i++){ $i=toupper(substr($i,1,1)) substr($i,2) }}1'
+ COPS=BITE_LV
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ awk '-F[, ]' '/^\+CPIN:/ {print $0;exit}'
+ xargs
+ T='+CPIN: READY'
+ '[' -n '+CPIN: READY' ]
+ '[' '+CPIN: READY' '=' '+CPIN: READY' ]
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ awk '-F[, ]' '/^\+CME ERROR:/ {print $0;exit}'
+ T=
+ '[' -n  ]
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ awk '-F[,]' '/^\+CREG/ {gsub(/[[:space:]"]+/,"");printf "T=\"%d\";LAC_HEX=\"%X\";CID_HEX=\"%X\";LAC_DEC=\"%d\";CID_DEC=\"%d\";MODE_NUM=\"%d\"", $2, "0x"$3, "0x"$4, "0x"$3, "0x"$4, $5}'
+ eval 'T="6";LAC_HEX="D5";CID_HEX="20EB35";LAC_DEC="213";CID_DEC="2157365";MODE_NUM="7"'
+ T=6
+ LAC_HEX=D5
+ CID_HEX=20EB35
+ LAC_DEC=213
+ CID_DEC=2157365
+ MODE_NUM=7
+ REG=6
+ '[' -z 7 ]
+ '[' x7 '=' x0 ]
+ MODE=LTE
+ sms_tool -d /dev/ttyACM2 at at+cereg
' OTX='at+cereg
' echo 'at+cereg
+ awk '-F[,]' '/^\+CEREG/ {printf "%s", toupper($3)}'
+ sed 's/[^A-F0-9]//g'
+ TAC=
+ '[' x '!=' x ]
+ TAC=-
+ TAC_HEX=-
+ uci -q get '3ginfo.@3ginfo[0].device'
+ DEVICE=/dev/ttyACM2
+ echo x/dev/ttyACM2+ grep -q 192.168.

+ awk '/Vendor=/{gsub(/.*Vendor=| ProdID=| Rev.*/,"");print}' /sys/kernel/debug/usb/devices
+ sort -u
+ _DEVS='1d6b0002
1d6b0003
8087095a'
+ '[' -e /usr/share/3ginfo-lite/3ginfo-addon/1d6b0002 ]
+ '[' -e /usr/share/3ginfo-lite/3ginfo-addon/1d6b0003 ]
+ '[' -e /usr/share/3ginfo-lite/3ginfo-addon/8087095a ]
+ cat /tmp/sysinfo/board_name
+ . /usr/share/3ginfo-lite/3ginfo-addon/8087095a
+ sms_tool -d /dev/ttyACM2 at 'at+xlec?;+rsrp;+rsrq;+xmci=1'
+ O='at+xlec?;+rsrp;+rsrq;+xmci=1
+XLEC: 0,3,5,5,3,BAND_LTE_1,3,20,0,0
'
+ echo 'at+xlec?;+rsrp;+rsrq;+xmci=1
+XLEC: 0,3,5,5,3,BAND_LTE_1,3,20,0,0
'
+ awk -F '[:_]' '/\+XLEC:/{gsub("\r",""); print $4}'
+ awk -F '[:,]' '{print $1}'
+ T=1
+ '[' -n 1 ]
+ band 1
+ echo 'B1  (2100 MHz)'
+ MODE='LTE B1  (2100 MHz)'
+ echo 'at+xlec?;+rsrp;+rsrq;+xmci=1
+XLEC: 0,3,5,5,3,BAND_LTE_1,3,20,0,0
'
+ awk -F '[:,]' '/\+RSRP:/{printf "%.0f\n", $4}'
+ RSRP=
+ echo 'at+xlec?;+rsrp;+rsrq;+xmci=1
+XLEC: 0,3,5,5,3,BAND_LTE_1,3,20,0,0
'
+ awk -F '[:,]' '/\+RSRQ:/{printf "%.0f\n", $4}'
+ RSRQ=
+ RSSI=-103
+ echo 'at+xlec?;+rsrp;+rsrq;+xmci=1
+XLEC: 0,3,5,5,3,BAND_LTE_1,3,20,0,0
'
+ awk -F '[:,]' '/\+XMCI: 4/{gsub(/"/,""); printf "%.0f\n", $13/4+5}'
+ SINR=
+ echo 'at+xlec?;+rsrp;+rsrq;+xmci=1
+XLEC: 0,3,5,5,3,BAND_LTE_1,3,20,0,0
'
+ awk -F '[:,]' '/\+XLEC:/{print $3}'
+ LTE_Cx=3
+ LTE_CA=2
3ginfo.sh: /usr/share/3ginfo-lite/3ginfo-addon/8087095a: line 242: syntax error: unterminated quoted string
```

4IceG, 2023-02-20:
Ok, I found two bugs. I corrected the file for replacement.
[808707f9.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/10783052/808707f9.zip)

newdanceradio, 2023-02-20:
Replaced the file, but, error is the same.

```
+ RES=/usr/share/3ginfo-lite
+ uci -q get '3ginfo.@3ginfo[0].device'
+ DEVICE=/dev/ttyACM2
+ '[' x/dev/ttyACM2 '=' x ]
+ echo /dev/ttyACM2
+ '[' x/dev/ttyACM2 '=' x ]
+ '[' x/dev/ttyACM2 '=' x ]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=wan
+ SUB=@
+ '[[' wan '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wan
+ '[' -z wan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wan
+ grep '"up": true'
+ NETUP='       "up": true,'
+ '[' -n '      "up": true,' ]
+ uci -q -P /var/state/ get network.wan.connect_time
+ CT=
+ '[' -z ]
+ ifstatus wan
+ awk '-F[:,]' '/uptime/ {print $2}'
+ xargs
+ CT=2139
+ '[' '!' -z 2139 ]
+ expr 2139 / 60 / 60 / 24
+ D=0
+ expr 2139 / 60 / 60 '%' 24
+ H=0
+ expr 2139 / 60 '%' 60
+ M=35
+ expr 2139 '%' 60
+ S=39
+ printf '%dd, %02d:%02d:%02d' 0 0 35 39
+ CONN_TIME='0d, 00:35:39'
+ ifstatus wan
+ awk '-F"' '/l3_device/ {print $4}'
+ IFACE=wan
+ '[' -n wan ]
+ ifconfig wan
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='1.1 MiB'
+ ifconfig wan
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='325.8 KiB'
+ sms_tool -D -d /dev/ttyACM2 at 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB35",7

'K
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 5,6

+CPIN: READY

[... 227 lines cut ...]

+XMCI: 5,000,000,"0xFFFE","0xFFFFFFFF","0x00F8","0x0000012C","0xFFFFFFFF","0xFFFFFFFF",17,0,255,"0x7FFFFFFF","0x00000000"
'
+ awk -F '[:,]' '/\+XMCI: 4/{gsub(/"/,""); printf "%.0f\n", $13/4+5}'
+ SINR=14
+ echo 'at+xlec?;+rsrp?;+rsrq?;+xmci=1
+XLEC: 0,3,5,5,3,BAND_LTE_1,0,0,0,0

+RSRP: 502,300,-103.00,502,1820,-84.00,502,6200,-90.00,248,300,-124.00

+RSRQ: 502,300,-7.-50,502,1820,-7.00,502,6200,-9.-50,248,300,-20.00

+XMCI: 4,247,05,"0x00D5","0x0020EB35","0x01F6","0x0000012C","0x0000477C","0xFFFFFFFF",37,26,34,"0x00000017","0x00000000"

+XMCI: 5,000,000,"0xFFFE","0xFFFFFFFF","0x01F6","0x0000071C","0xFFFFFFFF","0xFFFFFFFF",57,26,255,"0x7FFFFFFF","0x00000000"

+XMCI: 5,000,000,"0xFFFE","0xFFFFFFFF","0x01F6","0x00001838","0xFFFFFFFF","0xFFFFFFFF",51,25,255,"0x7FFFFFFF","0x00000000"

+XMCI: 5,000,000,"0xFFFE","0xFFFFFFFF","0x00F8","0x0000012C","0xFFFFFFFF","0xFFFFFFFF",17,0,255,"0x7FFFFFFF","0x00000000"
'
+ awk -F '[:,]' '/\+XLEC:/{print $3}'
+ LTE_Cx=3
+ LTE_CA=2
3ginfo.sh: /usr/share/3ginfo-lite/3ginfo-addon/8087095a: line 242: syntax error: unterminated quoted string
```

4IceG, 2023-02-20:
The script shows more and now there is another error, but unfortunately it's still an error and has a problem showing the data. I'll check it.

4IceG, 2023-02-20:
Ok, It should work now.
[808707f9.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/10783519/808707f9.zip)

newdanceradio, 2023-02-20:
Thank you @4IceG, almost done.
[image]

Also i don't have band B2
[image]
I have band B20 (800 Mhz)

[image]

newdanceradio, 2023-02-20:
Also here is visible at command
[image]
And there is no firmware version
also protocol is missing

4IceG, 2023-02-20:
You must have sent me inaccurate command outputs in the beginning, because I wrote data slicing scripts for them.

OK, show me the result of the commands, I'll fix it to this output.
```
sh -x /usr/share/3ginfo-lite/3ginfo.sh
cat /sys/kernel/debug/usb/devices
```

newdanceradio, 2023-02-20:
Here is 3ginfo.sh output: 

```
root@OpenWrt:/usr/share/3ginfo-lite# sh -x 3ginfo.sh
+ RES=/usr/share/3ginfo-lite
+ uci -q get '3ginfo.@3ginfo[0].device'
+ DEVICE=/dev/ttyACM2
+ '[' x/dev/ttyACM2 '=' x ]
+ echo /dev/ttyACM2
+ '[' x/dev/ttyACM2 '=' x ]
+ '[' x/dev/ttyACM2 '=' x ]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=ppp0
+ SUB=@
+ '[[' ppp0 '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=ppp0
+ '[' -z ppp0 ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus ppp0
+ grep '"up": true'
+ NETUP=
+ '[' -n  ]
+ sms_tool -D -d /dev/ttyACM2 at 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 15,5

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB17",7

'K
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 15,5

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB17",7

'K
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=15
+ '[' x15 '=' x ]
+ '[' 15 -ge 0 -a 15 -le 31 ]
+ CSQ_PER=48
+ echo 'AT+CSQ;+CPIN?;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CSQ: 15,5

+CPIN: READY

+COPS: 0,0,"BITE_LV",7

+COPS: 0,2,"24705",7

+CREG: 2,6,"00D5","0020EB17",7

'K
+ awk '-F["]' '/^\+COPS: .,2/ {print $2}'
+ COPS_NUM=24705
+ '[' x24705 '=' x ]
[... 508 lines cut ...]
"tac_d":"",
"cid_dec":"2157335",
"cid_hex":"20EB17",
"pci":"",
"earfcn":"",
"pband":"B3  (1800 MHz) @20 MHz",
"s1band":"B2  (1900 MHz) @10  MHz",
"s1pci":"",
"s1earfcn":"",
"s2band":"B1  (2100 MHz) @20  MHz",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-85",
"rsrq":"-7",
"rssi":"-83",
"sinr":"18"
}
+ exit 0
```

4IceG, 2023-02-20:
OK thanks. I have one more question. This modem now works on the ncm protocol?

I will write if I can fix something more, I have to look for some other list of bands, because what I have now indicates B2 not B20, strange.

obsy, 2023-02-20:
elif [ $1 -ge 6150 ] && [ $1 -le 6449 ]; then
    newband=2

should by 20.

4IceG, 2023-02-21:
@newdanceradio did you test the latest version, is everything ok now?

newdanceradio, 2023-02-21:
Oh, I'm sorry @4IceG, i didn't have enough time to test, was busy with other things. I'll test later today and let you know.
I also have a question, will 3ginfo-lite will work with modemmanager ?

4IceG, 2023-02-21:
Ok, no rush, you'll test it when you have time.

My packages are not compatible with MM, it makes it very difficult for other programs/packages to access the modem.

MM is good because it takes control over the operation of the modem, monitors the connection and allows you to read data from the modem. It would be best to create dedicated packages for LuCI to MM so that it does not clash with other packages.

closed 2023-03-12 by 4IceG

newdanceradio, 2023-05-23:
Hello dear @4IceG.
Sorry for big delay from my side about testing modem.

Now i have successfully working L860-GL modem with my OpenWRT router without modemmanager.

I have small questions.
Script is successfully reading data from modem, except Connection statistics and i don't know why it is showing blue rectangle with second information

[image]

Also, it isn't showing protocol
[image]

Here will be 

``
cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.10
S:  Manufacturer=Linux 5.10.176 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=095a Rev= 3.50
S:  Manufacturer=Intel Corp.
S:  Product=MODEM + 2 CDC-ACM + 3 CDC-NCM + SS
S:  SerialNumber=004999010640000:004999010640000:004999010640000:004999010640000
C:* #Ifs=13 Cfg#= 1 Atr=e0 MxPwr=100mA
A:  FirstIf#= 0 IfCount= 2 Cls=02(comm.) Sub=02 Prot=01
A:  FirstIf#= 2 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00
A:  FirstIf#= 4 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00
A:  FirstIf#= 6 IfCount= 2 Cls=02(comm.) Sub=0d Prot=00
A:  FirstIf#= 8 IfCount= 2 Cls=02(comm.) Sub=0d Prot=00
A:  FirstIf#=10 IfCount= 2 Cls=02(comm.) Sub=0d Prot=00
I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=01 Driver=cdc_acm
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 3 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 6 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0d Prot=00 Driver=cdc_ncm
E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:  If#= 7 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm
I:* If#= 7 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm
E:  Ad=88(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=08(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 8 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0d Prot=00 Driver=cdc_ncm
E:  Ad=89(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:  If#= 9 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm
I:* If#= 9 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm
E:  Ad=8a(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0a(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#=10 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0d Prot=00 Driver=cdc_ncm
E:  Ad=8b(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:  If#=11 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm
I:* If#=11 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=01 Driver=cdc_ncm
E:  Ad=8c(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0c(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#=12 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=01 Prot=00 Driver=(none)
E:  Ad=8d(I) Atr=03(Int.) MxPS=   4 Ivl=125us

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.10
S:  Manufacturer=Linux 5.10.176 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

4IceG, 2023-05-23:
Hi @newdanceradio , the package shows the user what needs to be corrected/changed in the configuration. If it can't show connection data, you need to change the name of the interface on which the modem works when connecting to the Internet.

As for the missing protocol, the modem is running as ncm?

newdanceradio, 2023-05-23:
> As for the missing protocol, the modem is running as ncm?

Yes, the modem is running as ncm

newdanceradio, 2023-05-23:
> you need to change the name of the interface on which the modem works when connecting to the Internet

Sorry, forgot about that, changed interface and now all is good
[image]

4IceG, 2023-05-23:
One more thing, you wrote above about rssi. Why do you think rssi is wrong?

4IceG, 2023-05-23:
The protocol should also be read correctly, in the evening I will check the script why has a problem and it does not show anything.

newdanceradio, 2023-05-23:
> One more thing, you wrote above about rssi. Why do you think rssi is wrong?

I have one more router Keenetic, when my L860-GL modem is connected to Keenetic router, it is showing stronger RSSI signal "-64" - "-59".
Also i have one more modem status script installed on my OpenWrt router and it is showing different result.

[image]

4IceG, 2023-05-23:
I understand, but in the example shown you have 80% of the signal, and with my package you have 38%, you should compare the indications in the same modem operating conditions.

If you have another script, you can show me how it reads this value, maybe my script is doing it wrong.

newdanceradio, 2023-05-23:
> you can show me how it reads this value

Sorry @4IceG, i'm not a big professional in these things, but i think it's here 
https://github.com/koshev-msk/modemfeed/blob/master/packages/net/modeminfo/root/usr/share/modeminfo/scripts/INTEL

```
CSQ=$(echo $RSRP $NP | awk '{printf "%.0f\n", (($1+10*log(12*$2)/log(10))+113)/2}')            
                [ "x$CSQ" = "x" ] && CSQ=-1                                                                    
                if [ $CSQ -ge 0 -a $CSQ -le 31 ]; then                                                         
                        CSQ_PER=$(($CSQ * 100/31))                                                   
                        CSQ_COL="red"                                                                
                        [ $CSQ -ge 10 ] && CSQ_COL="red"                                             
                        [ $CSQ -ge 15 ] && CSQ_COL="orange"                                          
                        [ $CSQ -ge 20 ] && CSQ_COL="green"                                           
                        CSQ_RSSI=$((2 * CSQ - 113))                                                
                        # Fibocom RSSI up -51dBm. Thanks to TSA74 at 4pda.to                       
                elif [ $CSQ -ge 32 -a $CSQ -le 50 ]; then                                          
                        CSQ_PER="100"                                                              
                        CSQ_COL="green"                                     
                        CSQ_RSSI=$((2 * CSQ - 113))                         
                else                                                        
                        CSQ_PER="0"                                         
                        CSQ_COL="black"                                     
                fi        
```

newdanceradio, 2023-05-23:
> you should compare the indications in the same modem operating conditions

Conditions are the same, because i have one external antenna to which modem L860-GL is connected every time, i just replug usb from one router to another.

4IceG, 2023-05-23:
The values ​​should be the same, my script:
RSSI=$((2 * CSQ - 113))
your code:
CSQ_RSSI=$((2 * CSQ - 113)

From what I see, the CSQ value is also calculated, and I don't do it additionally.

newdanceradio, 2023-05-23:
Hmm, very interesting, but why it can be, that it is showing different results ?

obsy, 2023-05-23:
Not the same. CSQ from koshev's repository was calculated from RSRP, not taken from AT+CSQ

4IceG, 2023-05-24:
Hi @newdanceradio , check if it already shows the protocol correctly. I didn't change anything in CSQ since it reads values normally.
[luci-app-3ginfo-lite_1.0.47-20230524_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/11551398/luci-app-3ginfo-lite_1.0.47-20230524_all.zip)

newdanceradio, 2023-05-25:
> Hi @newdanceradio , check if it already shows the protocol correctly. I didn't change anything in CSQ since it reads values normally. [luci-app-3ginfo-lite_1.0.47-20230524_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/11553669/luci-app-3ginfo-lite_1.0.47-20230524_all.zip)

Yes, now all is good.
Thank you @4IceG

newdanceradio, 2023-05-26:
I'm really sorry to bother you again, i found one more thing that isn't showing with L860 modem. 
This is PCI and EARFCN
It should be like this

[image]

But with L860 modem it looks like this

[image]

newdanceradio, 2023-05-26:
> I didn't change anything in CSQ since it reads values normally.

About RSSI.
Here is LTE RSRP, RSRQ, RSSI calculator.
I made some calculation, and there are discrepancies here.
[image]

If we calculate, that i have RSSI -89 and bandwidth of B3 i have 20MHz
then my RSRP signal should be -119, but in reality i have

4IceG, 2023-05-26:
Hi @newdanceradio , Ok I can add PCI and EARFCN if these are the indicated values, I will do it over the weekend when I find a quiet/free moment.

As for the value, I'm stupid, I don't know which value to take for granted and to relate to it.
I'm not sure but the AT+XCESQ? command it can also read RSRP or RSSI 🤔

newdanceradio, 2023-05-26:
> Ok I can add PCI and EARFCN if these are the indicated values, I will do it over the weekend when I find a quiet/free moment.

Yes, shure @4IceG no problem.
Yes, this values are indicated by command AT+RSRP?
Here is live example from modem L860
```
+RSRP: 502,1820,-89.00,502,6200,-90.00,502,300,-95.00,494,1820,-103.00
OK
```
Explaining about example above 
marked with bold is PCI 
**502**,1820,-89.00,**502**,6200,-90.00,**502**,300,-95.00,494,1820,-103.00
marked with bold is EARFCN
502,**1820**,-89.00,502,**6200**,-90.00,502,**300**,-95.00,494,1820,-103.00

newdanceradio, 2023-05-29:
Hi @4IceG ,
All is good now.
RSSI is calculated correctly!
Also PCI and EARFCN is showing now!
Brilliant !
I would like to tell thank you @4IceG  
[image]

4IceG, 2023-05-29:
Great 👍🏻 , I'm glad that everything works and the data shown is correct.

newdanceradio, 2024-06-19:
Hello @4IceG.
I have upgraded package and now something strange is going on
[image]
it doesn't show SINR anymore
and about bands also it showing strange things
[image]
It is showing twice band B1
but in reality output from modem is:
AT+XLEC?
+XLEC: 0,3,5,3,5,BAND_LTE_3,20,1,0,0

4IceG, 2024-06-19:
Hi,
@newdanceradio Unpack archive [fix_test.zip](https://github.com/user-attachments/files/15900366/fix_test.zip) and replace the files. File path \usr\share\3ginfo-lite\modem\usb.

Too many modems, I don't remember all the changes in scripts.

newdanceradio, 2024-06-19:
> Hi, @newdanceradio Unpack archive [fix_test.zip](https://github.com/user-attachments/files/15900366/fix_test.zip) and replace the files. File path \usr\share\3ginfo-lite\modem\usb.
> 
> Too many modems, I don't remember all the changes in scripts.

Thank you, BAND's looks fine now, also SINR is showing, but there is something with SINR calculation.
Sometime it is showing 
[image]

4IceG, 2024-06-19:
[fix2_test.zip](https://github.com/user-attachments/files/15901003/fix2_test.zip)
@newdanceradio Is it OK now?

newdanceradio, 2024-06-19:
Yeah, now all is good, seems everything is correct.

I would like to ask one more thing, in modemdefine Fibocom L860 is showing as
[image]

4IceG, 2024-06-19:
@newdanceradio In the `Manufacturer / Product` field you can enter the corrected name of modem (manually).

Data is taken from command result `cat /sys/kernel/debug/usb/devices`, sometimes they show stupid things there.

newdanceradio, 2024-06-19:
> `Manufacturer / Product` field you can enter the corrected name of modem (manually).

Heh, right, didn't noticed right away.
Entered manufacturer and product manually.
Thank you @4IceG as always.

4IceG, 2024-06-19:
> Heh, right, didn't noticed right away.
> Entered manufacturer and product manually.
> Thank you @4IceG as always.

Super! The OpenWrt community should help each other 💪🏻 😃.

PovilasID, 2025-01-02:
For me SIM seams to be connected but no data is transferred.  (▼ 0.0 B ▲ 0.0 B)  How do I debug? Can I use any AT commands that would check connection.