4IceG/luci-app-3ginfo-lite #41: config/3ginfo conflict

Jalakas, 2023-02-25:
I am having build conflict:
`Collected errors:
 * check_data_file_clashes: Package luci-app-3ginfo-lite wants to install file /home/user/openwrt/build_dir/target-mips_24kc_musl/root-ath79/etc/config/3ginfo
	But that file is already provided by package  * 3ginfo-text
 * opkg_install_cmd: Cannot install package luci-app-3ginfo-lite.`
when installing both packages:
CONFIG_PACKAGE_3ginfo=y (from https://github.com/obsy/packages)
CONFIG_PACKAGE_luci-app-3ginfo-lite=y

4IceG, 2023-02-25:
You have to decide whether you are using/compiling a package from Cezary or my version.

closed 2023-02-25 by 4IceG

Jalakas, 2023-02-25:
Well then, but:
https://github.com/4IceG/luci-app-3ginfo
and https://github.com/4IceG/luci-app-3ginfo-lite
give the same conflict.

4IceG, 2023-02-25:
You are using/compiling only the package luci-app-3ginfo-lite or just 3ginfo-text & luci-app-3ginfo.

Packages use one config file, so there will be a compilation conflict.