4IceG/luci-app-3ginfo-lite #43: add support for Quectel EC200T modem

jfrogg, 2023-04-22:
Please add support for Quectel EC200T modem, Vendor=2c7c ProdID=6026, using it in ECM mode.

Here is the output of some commands, please let me know if anything else is needed.

```
AT+QTEMP
+QTEMP: 38,255,255

AT+qnwinfo
+QNWINFO: "FDD LTE","12301","LTE BAND 3",1802

AT+qeng="servingcell"
+QENG: "servingcell","NOCONN","LTE","FDD",123,01,43300C,348,1802,3,5,5,5443,-87,-16,-69,5,34

```

Please note that SINR value requires no conversion.

4IceG, 2023-04-23:
Hi @jfrogg , unzip and copy this file to folder `/usr/share/3ginfo-lite/modem/`
[2c7c6026.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/11303922/2c7c6026.zip)

I haven't found a good description for this modem, so I'm not sure if all the data will be correct.

jfrogg, 2023-04-23:
This is a CAT4 modem that supports most of Quectel common commands.
At the moment the following information is missing:
TAC
Primary band | PCI & EARFCN
there is no RSRP,RSRQ,SINR,... data at all

What is shown:
Operator
SIM Info
Modem type
Revision / Firmware
Protocol
Temperature
MCC MNC
Cell ID
LAC
CSQ

I run a quick manual test and it looks like `sms_tool` is not working properly with multiple AT commands separated by `;`

jfrogg, 2023-04-23:
The problem is somewhere around `sms_tool`, it cannot handle multiple commands at once, probably due to limitations to its internal buffer size.
Here is the potentially weak place: `O=$(sms_tool -d $DEVICE at "at+qnwinfo;+qeng=\"servingcell\";+qspn")`
See the various combinations of the commands:
```
root@router:~# sms_tool -d /dev/ttyUSB1 at "at+qnwinfo;+qeng=\"servingcell\";+qspn"
at+qnwinfo;+qeng="servingcell";+qspn

root@router:~# sms_tool -d /dev/ttyUSB1 at "at+qnwinfo;+qeng=\"servingcell\""
at+qnwinfo;+qeng="servingcell"
```
There are no results in both cases. Now running them separately:
```
root@router:~# sms_tool -d /dev/ttyUSB1 at "at+qnwinfo;+qspn"
at+qnwinfo;+qspn
+QNWINFO: "FDD LTE","12301","LTE BAND 3",1802

+QSPN: "xxx xxx","xxx xxx","xxx xxx",0,"12301"

root@router:~# sms_tool -d /dev/ttyUSB1 at "at+qeng=\"servingcell\""
at+qeng="servingcell"
+QENG: "servingcell","NOCONN","LTE","FDD",123,01,12300A,456,1802,3,5,5,4321,-87,-15,-69,5,34
```

4IceG, 2023-04-23:
@jfrogg  since you know (better) what's wrong then correct yourself so it works. I added modem support and the data should be output.

In my opinion..
We communicate with the modem on the same port on which the connection takes place which blocks further reading of data. I don't know how many ports for communication this modem has, maybe changing to another would solve the problem.

We could try renaming the variables to separate the data more (silly but it helps some modems).

You didn't send me the result of the command, so I can only divine from a glass ball.

closed 2023-04-23 by 4IceG

jfrogg, 2023-04-23:
Corrected and it works. Using a separate variable helped to obtain the missing data.
Regarding the ports - this particular modem in its current configuration has 1 ECM interface and 3 /dev/ttyUSBx where 
ttyUSB0 used for DM /I guess they mean Download Mode/
ttyUSB1 used for AT command communication /this is what I'm using/
ttyUSB2 used for PPP connections or AT command communication

Please consider a few minor changes:
- currently both LAC and TAC are shown with the same value, but in LTE mode LAC should not be there at all;
- this is a CAT4 modem but I have 4 empty lines with "CA band (Sx)"

Please don't take any of my comments as negative, I'm trying to contribute in my way ;)
Thank you for all your efforts.
[2c7c6026.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/11304476/2c7c6026.zip)

4IceG, 2023-04-23:
> currently both LAC and TAC are shown with the same value, but in LTE mode LAC should not be there at all;    

This value may not be there, but it is not a rule. Let it stay that way because immediately some user will report an error because he does not have all the data.

> this is a CAT4 modem but I have 4 empty lines with "CA band (Sx)"    

Quectel EC200T modem is LTE only, 150Mbps downlink and 50Mbps uplink. Band aggregation starts with CAT6 modems, so seriously but I don't know what the package is supposed to show there?

I try to approach each user with understanding, but I also require respect for my time and work. What I make available for LuCI are my private packages, everything works for me and I don't have to add or correct anything because someone would like to have the same at home. Users have "only requirements" few will thank / give a star or support $. Everyone wants to be smarter, so the question is if you are so smart why didn't you make such a package/packages yourself? After all, it's all so simple.

jfrogg, 2023-04-23:
Sorry if I was not very specific.
For CAT4 modems the lines "CA band (Sx)" from my perspective should not be displayed at all.
TAC is specific to LTE and 5G/NR while LAC is an attribute of 3G/UMTS, so I propose to implement the simple check in the script before displaying the either line.
I see your point, so this is not a demand but a kind suggestion to make the package look better.

4IceG, 2023-04-24:
> For CAT4 modems the lines "CA band (Sx)" from my perspective should not be displayed at all.   

You are the first user who is bothered by the visibility of these fields. I could hide it, but it's not my priority at this point. I'm wasting my time looking up modem specs to adjust the package for their band aggregation.

> TAC is specific to LTE and 5G/NR while LAC is an attribute of 3G/UMTS

Interesting, but many modems with LTE provide this data. Again, should I not show them because that's the rule? again there will be users writing that the parameter is there, and I do not show it.

_P.S.
I don't force anyone to install my package/packages, if you don't like the package you can always install another, better one or write Your own._