4IceG/luci-app-3ginfo-lite #45: Support Fibocom FM150-AE-00(with pcie bus lanes)

ChaingTsung, 2023-08-02:
Does it support Fibocom FM150-AE-00? I

ChaingTsung, 2023-08-08:
[doc.zip share with sendanywhere](https://sendanywhe.re/LDQ4GNGU)

obsy, 2023-08-11:
Upload doc again, please.

4IceG, 2023-08-11:
Hi @ChaingTsung 
I tried to download before and nothing worked from what you shared.
Add the documentation normally here as a zip file attachment.

ChaingTsung, 2023-08-11:
https://1drv.ms/u/s!AlmKO5JLGJEOgdxrvkVeP0vGMFW1vQ?e=4Fbu14

ChaingTsung, 2023-08-11:
> 你好 @ChaingTsung 我之前尝试过下载，但分享的内容没有任何效果。您 通常将文档作为 zip 文件附件添加到此处。

I share with you via onedriver.https://1drv.ms/u/s!AlmKO5JLGJEOgdxrvkVeP0vGMFW1vQ?e=4Fbu14

ChaingTsung, 2023-08-11:
> Upload doc again, please.

Alreadyly uploaded.https://1drv.ms/u/s!AlmKO5JLGJEOgdxrvkVeP0vGMFW1vQ?e=4Fbu14

ChaingTsung, 2023-08-11:
Of course, source also uploaded into my  repository at https://github.com/ChaingTsung/Fibocom_FM150-AE-00_PCIE-DRIVER

ChaingTsung, 2023-08-11:
> 当然，源代码也上传到我的存储库中：https://github.com/ChaingTsung/Fibocom_FM150-AE-00_PCIE-DRIVER

this  repository does not include official doc, official included in https://1drv.ms/u/s!AlmKO5JLGJEOgdxrvkVeP0vGMFW1vQ?e=4Fbu14.

ChaingTsung, 2023-08-11:
OFFICIAL Dial tools has been uploaded into https://github.com/ChaingTsung/Fibocom_dail_software_Linux.git. I built this tool on openwrt. Although tool is sucessfully built, the tool is unavailable.(As long as you run the built dial tool named `fibocom_dail`{path : /usr/bin/fibo_dail), it says segments error.)

4IceG, 2023-08-11:
As for me, there is nothing here, a lot of pdfs of little use. No list of at commands with their explanation.

obsy, 2023-08-12:
> https://1drv.ms/u/s!AlmKO5JLGJEOgdxrvkVeP0vGMFW1vQ?e=4Fbu14

Thanks. It's quite useful, even though it's in Chinese :)

Show me:
```
cat /sys/kernel/debug/usb/devices
```
Install picocom (opkg update; opkg install picocom), try
```
picocom -b 115200 /dev/ttUSB0
picocom -b 115200 /dev/ttUSB1
picocom -b 115200 /dev/ttUSB2
picocom -b 115200 /dev/ttUSB3
etc
```
Type ATI, the modem should respond. If not, try the next interface (ttyUSB1, ttyUSB2 and so on). If modem respond, type and show output of
```
ATI
AT+CPIN?
AT+CSQ
AT+COPS=3,0;+COPS?
AT+COPS=3,2;+COPS?
AT+CREG=2;+CREG?"
AT+GTCAINFO?
AT+GTCCINFO?
AT+MTSM=1
AT+MTSM=6
AT+MTSM=7
```

ChaingTsung, 2023-08-12:
[FIBOCOM FG150 & FM150 Series AT Commands_V3.6.7.pdf](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327627/FIBOCOM.FG150.FM150.Series.AT.Commands_V3.6.7.pdf)
Here is the AT doc in English.

ChaingTsung, 2023-08-12:
[FIBOCOM FG150 & FM150 Series AT Commands_GNSS_V1.0.3.pdf](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327629/FIBOCOM.FG150.FM150.Series.AT.Commands_GNSS_V1.0.3.pdf)

ChaingTsung, 2023-08-12:
[FIBOCOM AT Commands_eSIM_V1.0.3.pdf](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327632/FIBOCOM.AT.Commands_eSIM_V1.0.3.pdf)

ChaingTsung, 2023-08-12:
I have to upload one by one due to the limit of file size .

ChaingTsung, 2023-08-12:
In fact, all doc included in "Zh-doc\3-SoftwareZh-doc\3-Software\通用"

ChaingTsung, 2023-08-12:
> > https://1drv.ms/u/s!AlmKO5JLGJEOgdxrvkVeP0vGMFW1vQ?e=4Fbu14
> 
> 谢谢。虽然是中文的，但还是很有用的：)
> 
> 给我看看：
> 
> ```
> cat /sys/kernel/debug/usb/devices
> ```
> 
> 安装 picocom（opkg 更新；opkg 安装 picocom），尝试
> 
> ```
> picocom -b 115200 /dev/ttUSB0
> picocom -b 115200 /dev/ttUSB1
> picocom -b 115200 /dev/ttUSB2
> picocom -b 115200 /dev/ttUSB3
> etc
> ```
> 
> 输入 ATI，调制解调器应该响应。如果没有，请尝试下一个接口（ttyUSB1、ttyUSB2 等）。如果调制解调器响应，请键入并显示输出
> 
> ```
> ATI
> AT+CPIN?
> AT+CSQ
> AT+COPS=3,0;+COPS?
> AT+COPS=3,2;+COPS?
> AT+CREG=2;+CREG?"
> AT+GTCAINFO?
> AT+GTCCINFO?
> AT+MTSM=1
> AT+MTSM=6
> AT+MTSM=7
> ```

Unfortunately, the FM150-AE(version : 00) only runs via pcie bus, not usb. "/dev/mhi_DUN" is for AT. And all device descriptor for this module(version: 00) is starting with "mhi_" in  '/dev'.

ChaingTsung, 2023-08-12:
> > > https://1drv.ms/u/s!AlmKO5JLGJEOgdxrvkVeP0vGMFW1vQ?e=4Fbu14
> > 
> > 
> > 谢谢。虽然是中文的，但还是很有用的：)
> > 给我看看：
> > ```
> > cat /sys/kernel/debug/usb/devices
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 安装 picocom（opkg 更新；opkg 安装 picocom），尝试
> > ```
> > picocom -b 115200 /dev/ttUSB0
> > picocom -b 115200 /dev/ttUSB1
> > picocom -b 115200 /dev/ttUSB2
> > picocom -b 115200 /dev/ttUSB3
> > etc
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 输入 ATI，奥斯应响应。如果没有，请尝试下一个接口（ttyUSB1、ttyUSB2 等）。如果奥斯响应，请键入并显示输出
> > ```
> > ATI
> > AT+CPIN?
> > AT+CSQ
> > AT+COPS=3,0;+COPS?
> > AT+COPS=3,2;+COPS?
> > AT+CREG=2;+CREG?"
> > AT+GTCAINFO?
> > AT+GTCCINFO?
> > AT+MTSM=1
> > AT+MTSM=6
> > AT+MTSM=7
> > ```
> 
> 不幸的是，FM150-AE（版本：00）仅通过 PCIe 总线运行，而不是 USB。“/dev/mhi_DUN” 用于 AT。该模块（版本：00）的所有设备描述符均以 “mhi_” 开头（在 “/dev” 目录中）。

Look at this picture.
[image]

ChaingTsung, 2023-08-12:
# MHI virtual channel function description

| Number |  mhi descriptor | function |
|:----- | :-----:|-----: |
| 1 | mhi_BHI | Port for downloading firehose file |
| 2 | mhi_DUN | Port for AT commands |
| 3 | mhi_LOOPBACK | Port for testing loopback |
| 4 | mhi_DIAG | Port for catching up QXDM log |
| 5 | mhi_QMI0 | Port for transporting QMI message |
| 6 |  mhi_QMI1 | Reserved |
| 7 |  mhi_GNSS | Port for getting & uploading GPS NEMA data |
| 8 |  mhi_EDL | Port for firmware  upgrade ( starting to enum when the module is in edl mode) |

ChaingTsung, 2023-08-12:
[FIBOCOM ECM Dial USB Driver Option_Linux.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327700/FIBOCOM.ECM.Dial.USB.Driver.Option_Linux.zip)
[FIBOCOM Dump Collect Tool_Linux & Android_V1.0.4.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327701/FIBOCOM.Dump.Collect.Tool_Linux.Android_V1.0.4.zip)
[FIBOCOM Download Tool_Linux & Android_V1.8.8.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327702/FIBOCOM.Download.Tool_Linux.Android_V1.8.8.zip)
[FIBOCOM Diaggrab Log Tool_Linux & Android_V1.0.9.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327703/FIBOCOM.Diaggra
[FIBOCOM Linux GobiNet Driver_V2.01.00.02.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327707/FIBOCOM.Linux.GobiNet.Driver_V2.01.00.02.zip)
b.Log.Tool_Linux.And
[FIBOCOM Hostproxy TooL_Linux & Android_V1.0.2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327706/FIBOCOM.Hostproxy.TooL_Linux.Android_V1.0.2.zip)
roid_
[FIBOCOM Foat Tool_Linux_V1.0.2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327705/FIBOCOM.Foat.Tool_Linux_V1.0.2.zip)
V1.0.9.zip)
[FIBOCOM ATsend_TooL_Linux_V1.0.1.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327704/FIBOCOM.ATsend_TooL_Linux_V1.0.1.zip)

ChaingTsung, 2023-08-12:
[FIBOCOM PCIE MHI Driver_Linux_V1.0.3.tar.gz](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327708/FIBOCOM.PCIE.MHI.Driver_Linux_V1.0.3.tar.gz)
[Fibocom_Linux_pcie_host_driver_1.0.5.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327709/Fibocom_Linux_p
[Fibocom_QMI_WWAN_Driver_V1.0.2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327710/Fibocom_QMI_WWAN_Driver_V1.0.2.zip)
cie_host_driver_1.0.5.zip)

ChaingTsung, 2023-08-12:
All driver source code for FM150-AE ( Both pcie and usb bus) uploaded.

obsy, 2023-08-12:
> # MHI virtual channel function description
> Number	mhi descriptor	function
> 1	mhi_BHI	Port for downloading firehose file
> 2	mhi_DUN	Port for AT commands
> 3	mhi_LOOPBACK	Port for testing loopback
> 4 [...]

So run these commands on /dev/mhi_DUN

And show ```cat /sys/bus/pci/devices/*/uevent```

ChaingTsung, 2023-08-13:
> > # MHI 虚拟通道功能说明
> > 编号 mhi 描述符功能
> > 1 mhi_BHI 用于下载 firehose 文件的端口
> > 2 mhi_DUN 用于 AT 命令的端口
> > 3 mhi_LOOPBACK 用于测试环回的端口
> > 4 mhi_DIAG 用于捕获 QXDM 日志的
> > 端口 5 mhi_QMI0 用于传输 QMI 消息的端口
> > 6 mhi_QMI1 保留
> > 7 mhi_GNSS 用于获取和上传 GPS NEMA 的端口 data 
> > 8 mhi_EDL 固件升级端口（模块处于 edl 模式时开始枚举）
> 
> 因此，在 /dev/mhi_DUN 上运行这些命令
> 
> 并展示`cat /sys/bus/pci/devices/*/uevent`

```shell
root@OpenWrt:~# echo "ati\r" > /dev/mhi_DUN
[  785.513393] [I][mhi0][__mhi_prepare_channel] Entered: preparing channel:32
[  785.530066] [I][mhi0][mhi_dump_tre] carl_ev evt_cmd_comp code=1, type=33
[  785.536764] [I][mhi0][__mhi_prepare_channel] Chan:32 successfully moved to start state
[  785.544664] [I][mhi0][__mhi_prepare_channel] Entered: preparing channel:33
[  785.558769] [I][mhi0][mhi_dump_tre] carl_ev evt_cmd_comp code=1, type=33
[  785.565457] [I][mhi0][__mhi_prepare_channel] Chan:33 successfully moved to start state
```

obsy, 2023-08-13:
Use picocom, not echo.

ChaingTsung, 2023-08-16:
> 使用 picocom，而不是 echo。

not supported. @4lceG the driver dones not support fm150-ae( via pcie bus).  The result of command `lspci -vvv` shoud return ` Kernel driver in use: fibo_mhi` not ` Kernel driver in use: mhi_q`.  The driver code I uploaded passes on Ubuntu 20.04 x64 with kernel 5.15.125, however I do not know how to edit or modify its `makefile` for OpenWRT.

4IceG, 2023-08-17:
> not supported. @4lceG the driver dones not support fm150-ae( via pcie bus).

If I understand correctly, mhi is not supported and needs to be manually compiled in OpenWrt.

I found an interesting pdf that describes it [> LINK <](https://www.nxp.com/docs/en/application-note/AN13455.pdf) unfortunately I don't know where the necessary sources are on github.

obsy, 2023-08-17:
Supported in 23.05 and snapshot.

ChaingTsung, 2023-08-17:
> > 不支持。@4lceG 驱动程序不支持 fm150-ae（通过 pcie 总线）。
> 
> 如果我理解正确的话，mhi 不支持，需要在 OpenWrt 中手动编译。
> 
> 我发现了一个有趣的 pdf 描述它[> 链接 <](https://www.nxp.com/docs/en/application-note/AN13455.pdf)不幸的是我不知道 github 上必要的来源在哪里。

[image]
In that pdf, the fibocom 5G module  named FM150-AE-01( via usb bus) is different from FM150-AE-00 (via pcie bus).

ChaingTsung, 2023-08-17:
As for gosuncn 5G module GM800, it connects to net. via usb if its version is `c1b` , only `c1a` is pcie bus.

ChaingTsung, 2023-08-17:
As for quectel rm 50x series, most of them can be run via usb or pcie, just use AT commands to change.Most of quectel 5g module supports usb and pcie if it' consists of `Qualcomm sdx55 or sdx65`.

ChaingTsung, 2023-08-17:
fibocom fm150-ae or 160 series : 
00, 10, 20 is pcie bus
01, 11, 21 is usb bus.

ChaingTsung, 2023-08-17:
> Supported in 23.05 and snapshot.

If the kernel is 6.x, the module can not work. It seems a lot of APIs to 'net' has been changed. And default kernel pcie mhi module does not make this 5G module work. So do the dial tools.

obsy, 2023-08-17:
Please, this is NOT openwrt support. Just show us output of these AT command, And that's it - from the point of view of this project, these implementation details are irrelevant.

ChaingTsung, 2023-08-17:
> Please, this is NOT openwrt support. Just show us output of these AT command, And that's it - from the point of view of this project, these implementation details are irrelevant.

No message returns after input these at command. The driver does not work! So no reply for at command.

4IceG, 2023-08-17:
> In that pdf, the fibocom 5G module named FM150-AE-01( via usb bus) is different from FM150-AE-00 (via pcie bus).

But they also showed modem support on pcie..
[image]

ChaingTsung, 2023-08-17:
Info shown in your pic. reveals this is `00` version(type).
[image]

ChaingTsung, 2023-08-17:
I guess what the doc described is two drivers.

4IceG, 2023-08-17:
> Info shown in your pic. reveals this is `00` version(type).

It's not about the exact version of the modem, but about the procedure. After all, we want to talk to this modem somehow.

obsy, 2023-08-17:
> > Please, this is NOT openwrt support. Just show us output of these AT command, And that's it - from the point of view of this project, these implementation details are irrelevant.
> 
> No message returns after input these at command. The driver does not work! So no reply for at command.

Show us this "no message".

ChaingTsung, 2023-08-18:
> > > 请注意，这不是 openwrt 支持。只需向我们展示这些 AT 命令的输出，就这样 - 从这个项目的角度来看，这些实现细节是无关紧要的。
> > 
> > 
> > 在命令中输入这些后不会返回任何消息。司机不工作！所以 at 命令没有回复。
> 
> 向我们展示这个 “无消息”。

# Okay, now I rebuilt the ofiicial, clean firmware used @4IceG drivers.Info in here.

> dmesg
```
root@OpenWrt:/# dmesg |grep mhi
[   22.725767] mhi_init Quectel_Linux_PCIE_MHI_Driver_V1.3.4
[   22.731485] mhi_q 0000:01:00.0: no of_node; not parsing pinctrl DT
[   22.737707] mhi_q 0000:01:00.0: assign IRQ: got 0
[   22.742431] mhi_pci_probe pci_dev->name = 0000:01:00.0, domain=0, bus=1, slot=0, vendor=17CB, device=0306
[   22.752195] mhi_q 0000:01:00.0: BAR 0: assigned [mem 0x20000000-0x20000fff 64bit]
[   22.759703] mhi_q 0000:01:00.0: enabling device (0000 -> 0002)
[   22.765548] mhi_q 0000:01:00.0: enabling bus mastering
[   22.813231] [I][mhi0][mhi_init_pci_dev] msi_required = 5, msi_allocated = 5, msi_irq = 149
[   22.821491] [I][mhi0][mhi_power_up] dev_state:READY
[   22.826353] [I][mhi0][mhi_async_power_up] Requested to power on
[   22.832309] [I][mhi0][mhi_alloc_coherent] size = 114688, dma_handle = 49860000
[   22.839512] [I][mhi0][mhi_init_dev_ctxt] mhi_ctxt->ctrl_seg = 0000000040e71c04
[   22.847016] [I][mhi0][mhi_async_power_up] dev_state:READY ee:AMSS
[   22.853184] [I][mhi0][mhi_async_power_up] Power on setup success
[   22.859214] [I][mhi0][mhi_pci_show_link] LnkCap:     Speed 16GT/s, Width x2
[   22.865815] [I][mhi0][mhi_pci_show_link] LnkSta:     Speed 5GT/s, Width x1
[   22.872335] [I][mhi0][mhi_pci_probe] Return successful
[   22.877815] [I][mhi0][mhi_pm_st_worker] Transition to state:READY
[   22.883936] [I][mhi0][mhi_pm_st_worker] INVALID_EE -> AMSS
[   22.884265] [I][mhi0][mhi_intvec_threaded_handlr] device ee:AMSS dev_state:READY, pm_state:POR
[   22.889415] [I][mhi0][mhi_ready_state_transition] Waiting to enter READY state
[   22.898025] [I][mhi0][mhi_intvec_threaded_handlr] device ee:AMSS dev_state:READY, INVALID_EE
[   22.913716] [I][mhi0][mhi_ready_state_transition] Device in READY State
[   22.920322] [I][mhi0][mhi_tryset_pm_state] Transition to pm state from:POR to:POR
[   22.933344] [I][mhi0][mhi_init_mmio] Initializing MMIO
[   22.938481] [I][mhi0][mhi_init_mmio] CHDBOFF:0x300
[   22.943278] [I][mhi0][mhi_init_mmio] ERDBOFF:0x700
[   22.948071] [I][mhi0][mhi_init_mmio] Programming all MMIO values.
[   23.019657] [I][mhi0][mhi_dump_tre] carl_ev evt_state_change mhistate=2
[   23.026289] [I][mhi0][mhi_process_ctrl_ev_ring] MHI state change event to state:M0
[   23.033845] [I][mhi0][mhi_pm_m0_transition] Entered With State:READY PM_STATE:POR
[   23.041312] [I][mhi0][mhi_tryset_pm_state] Transition to pm state from:POR to:M0
[   23.048830] [I][mhi0][mhi_dump_tre] carl_ev evt_ee_state execenv=2
[   23.055000] [I][mhi0][mhi_process_ctrl_ev_ring] MHI EE received event:AMSS
[   23.062209] [I][mhi0][mhi_pm_st_worker] Transition to state:MISSION MODE
[   23.068903] [I][mhi0][mhi_pm_st_worker] INVALID_EE -> AMSS
[   23.074381] [I][mhi0][mhi_pm_mission_mode_transition] Processing Mission Mode Transition
[   23.089922] [I][mhi0][mhi_init_timesync] TIME OFFS:0xa20
[   23.102172] [I][mhi0][mhi_init_timesync] TIMESYNC_DB OFFS:0xf1a4b3de
[   23.108534] [I][mhi0][mhi_init_timesync] Could not find timesync event ring
[   23.115512] [I][mhi0][mhi_pm_mission_mode_transition] Adding new devices
[   23.122480] mhi_uci_q 0306_00.01.00_LOOPBACK: no of_node; not parsing pinctrl DT
[   23.135033] mhi_uci_q 0306_00.01.00_DIAG: no of_node; not parsing pinctrl DT
[   23.147647] mhi_uci_q 0306_00.01.00_QMI0: no of_node; not parsing pinctrl DT
[   23.168139] [I][mhi0][mhi_dtr_probe] Enter for DTR control channel
[   23.174336] [I][mhi0][__mhi_prepare_channel] Entered: preparing channel:18
[   23.185832] [I][mhi0][mhi_dump_tre] carl_ev evt_cmd_comp code=1, type=33
[   23.194150] [I][mhi0][__mhi_prepare_channel] Chan:18 successfully moved to start state
[   23.202100] [I][mhi0][__mhi_prepare_channel] Entered: preparing channel:19
[   23.213368] [I][mhi0][mhi_dump_tre] carl_ev evt_cmd_comp code=1, type=33
[   23.220112] [I][mhi0][__mhi_prepare_channel] Chan:19 successfully moved to start state
[   23.228102] [I][mhi0][mhi_dtr_probe] Exit with ret:0
[   23.238878] mhi_uci_q 0306_00.01.00_DUN: no of_node; not parsing pinctrl DT
[   23.246134] mhi_netdev 0306_00.01.00_IP_HW0: no of_node; not parsing pinctrl DT
[   23.258831] [I][mhi_netdev_enable_iface] Prepare the channels for transfer
[   23.265728] [I][mhi0][__mhi_prepare_channel] Entered: preparing channel:100
[   23.282797] [I][mhi0][mhi_dump_tre] carl_ev evt_cmd_comp code=1, type=33
[   23.289538] [I][mhi0][__mhi_prepare_channel] Chan:100 successfully moved to start state
[... 116 lines cut ...]
ts are   : 1
es[  419.250663] [I][mhi0][__mhi_prepare_channel] Entered: preparing channel:33
cape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is    [  419.266751] [I][mhi0][mhi_dump_tre] carl_ev evt_cmd_comp code=1, type=33
  : no
nolock i[  419.273913] [I][mhi0][__mhi_prepare_channel] Chan:33 successfully moved to start state
s      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        :
omap is        :
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

FATAL: failed to add port: Filedes is not a tty
root@OpenWrt:/#  picocom -b 115200/dev/mhi_DUN
No port given
Run with '--help'.
```

ChaingTsung, 2023-08-18:
Look at these pic. Canot dail as the tool is not right.

[image]

```
root@OpenWrt:/# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP group default qlen 1000
    link/ether b6:d7:3b:7d:29:aa brd ff:ff:ff:ff:ff:ff
    inet6 fe80::b4d7:3bff:fe7d:29aa/64 scope link
       valid_lft forever preferred_lft forever
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-wan state DOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:ab brd ff:ff:ff:ff:ff:ff
4: wan@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-wan state LOWERLAYERDOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:ab brd ff:ff:ff:ff:ff:ff permaddr b6:d7:3b:7d:29:aa
5: lan1@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:aa brd ff:ff:ff:ff:ff:ff
6: lan2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:aa brd ff:ff:ff:ff:ff:ff
7: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:aa brd ff:ff:ff:ff:ff:ff
8: lan4@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:aa brd ff:ff:ff:ff:ff:ff
9: sfp2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:aa brd ff:ff:ff:ff:ff:ff
10: tunl0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/ipip 0.0.0.0 brd 0.0.0.0
11: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
12: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 12:20:4f:4b:59:4c brd ff:ff:ff:ff:ff:ff
13: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
14: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
15: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
16: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
17: wwan0: <NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-wan5G state UNKNOWN group default qlen 1000
    link/ether e6:b1:20:0f:4f:e8 brd ff:ff:ff:ff:ff:ff
18: rmnet_mhi0: <NO-CARRIER,NOARP,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/none
19: rmnet_mhi0.1: <NO-CARRIER,NOARP,UP> mtu 1500 qdisc fq_codel master br-wan5G state DOWN group default qlen 1000
    link/ether 02:50:f4:00:00:01 brd ff:ff:ff:ff:ff:ff
22: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether b6:d7:3b:7d:29:aa brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.1/24 brd 10.0.0.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd5c:7183:f409::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::b4d7:3bff:fe7d:29aa/64 scope link
       valid_lft forever preferred_lft forever
23: br-wan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether b6:d7:3b:7d:29:ab brd ff:ff:ff:ff:ff:ff
    inet6 fe80::b4d7:3bff:fe7d:29ab/64 scope link
       valid_lft forever preferred_lft forever
24: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 82:0c:43:26:60:00 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::800c:43ff:fe26:6000/64 scope link
       valid_lft forever preferred_lft forever
28: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 00:0c:43:26:60:00 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::20c:43ff:fe26:6000/64 scope link
       valid_lft forever preferred_lft forever
32: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default
    link/ether 02:42:40:40:94:be brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
36: br-wan5G: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 02:50:f4:00:00:01 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::50:f4ff:fe00:1/64 scope link
       valid_lft forever preferred_lft forever
```

ChaingTsung, 2023-08-18:
In `lspci -vvv`, `mhi_q` is quectels'.

ChaingTsung, 2023-08-18:
although it was successfully identified( quectel and fibocom 150 use sdx55), it can not connect to net. ` rmnet_mhi0`  and ` rmnet_mhi0.1` have no ip, no route.

ChaingTsung, 2023-08-18:
Seems that some differnces among  them.

ChaingTsung, 2023-08-18:
> [广和通 PCIE MHI Driver_Linux_V1.0.3.tar.gz](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327708/FIBOCOM.PCIE.MHI.Driver_Linux_V1.0.3.tar.gz) [Fibocom_Linux_pcie_host_driver_1.0.5.zip]( [https://github.com/4IceG/luci-app-3ginfo-lite/files/12327709/Fibocom_Linux_p ](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327709/Fibocom_Linux_p) [Fibocom_QMI_WWAN_Driver_V1.0.2.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12327710/Fibocom_QMI_WWAN_Driver_V1.0.2.zip) cie_host_driver_1.0.5. 压缩）

successfully made driver with my source code and connected to network with `fibocom-dial ` tool.
```log
root@OpenWrt:/# fibocom-dial -s 5gscuiot -d /dev/mhi_QMI0 &
root@OpenWrt:/# [08-13_11:45:10:218] Fibocom-dial_Linux_Tool_V2.0.10
[08-13_11:45:10:218] fibocom-dial profile[1] = 5gscuiot///0, pincode = (null)
[08-13_11:45:10:218] socket[3] successfuly!
[08-13_11:45:10:218] Waiting client to connect.....
[08-13_11:45:10:219] qmichannel(/dev/mhi_QMI0) usbnet_adapter(pcie_mhi0)
[08-13_11:45:10:219] pcie mode
[08-13_11:45:10:219] ioctl(0x89f3, qmap_settings) failed: Not supported, rc=-1
[08-13_11:45:10:219] qmap_mode = 1, muxid = 0x81, qmap_netcard = pcie_mhi0
[08-13_11:45:10:219] ioctl(0x89f3, qmap_settings) failed: Not supported, rc=-1
[08-13_11:45:10:219] qmap_mode = 1, muxid = 0x81, qmap_netcard = pcie_mhi0
[08-13_11:45:10:219] qmap_mode=1
[08-13_11:45:10:267] cdc_wdm_fd = 9
[08-13_11:45:10:267] write triger_event: 4098 to qmidevice_control_fd
[08-13_11:45:11:268] QmiThreadSendQMITimeout pthread_cond_timeout_np=110, errno: 22 (Invalid argument)
[08-13_11:45:12:277] QmiWwanGetClientID: QMIType = 1 clientid 14
[08-13_11:45:12:278] Get clientWDS = 14
[08-13_11:45:12:279] QmiWwanGetClientID: QMIType = 2 clientid 1
[08-13_11:45:12:279] Get clientDMS = 1
[08-13_11:45:12:281] QmiWwanGetClientID: QMIType = 3 clientid 2
[08-13_11:45:12:281] Get clientNAS = 2
[08-13_11:45:12:283] QmiWwanGetClientID: QMIType = 11 clientid 2
[08-13_11:45:12:283] Get clientUIM = 2
[08-13_11:45:12:285] QmiWwanGetClientID: QMIType = 26 clientid 1
[08-13_11:45:12:285] Get clientWDA = 1
[08-13_11:45:12:287] requestBaseBandVersion 89602.1000.00.04.07.33
[08-13_11:45:12:289] qmap_settings.rx_urb_size = 32768
[08-13_11:45:12:311] dev: /dev/mhi_DUN
sendbuffer:AT+GTDUALSIMEN?

[08-13_11:45:15:311] read 16 AT+GTDUALSIMEN?
[08-13_11:45:15:311] read 0/2/3

[08-13_11:45:15:311] sim_select = 0

[08-13_11:45:15:315] curr_ints_sim1
[08-13_11:45:15:315] curr_ints->CardState is 1
[08-13_11:45:15:315] curr_ints->NumApp is 1
[08-13_11:45:15:315] AppType = 2

[08-13_11:45:15:315] requestGetSIMStatus SIMStatus: SIM_READY
[08-13_11:45:15:317] requestGetICCID DeviceICCID: 89860621280023767567
[08-13_11:45:15:318] requestGetIMSI DeviceIMSI: 460062798273216
[08-13_11:45:15:318] requestSetProfile[1] 5gscuiot///0
[08-13_11:45:15:341] requestGetProfile[1] 5gscuiot///0
[08-13_11:45:15:344] requestRegistrationState2 MCC: 460, MNC: 1, PS: Attached, DataCap: NR5G
[08-13_11:45:15:344] write signo: 12 to signal_control_fd
[08-13_11:45:15:344] epoll fd = 6, events = 0x0001
[08-13_11:45:15:344] get signo: 12
[08-13_11:45:15:350] requestQueryDataCall IPv4ConnectionStatus: DISCONNECTED
[08-13_11:45:15:350] usbnet_link_change :link:0
[08-13_11:45:15:350] enter udhcpc_stop_pcie
[08-13_11:45:15:350] ifconfig pcie_mhi0 down
[08-13_11:45:15:361] ifconfig pcie_mhi0 0.0.0.0
ifconfig: SIOCSIFFLAGS: Address not available
[08-13_11:45:15:363] write signo: 10 to signal_control_fd
[08-13_11:45:15:363] epoll fd = 7, events = 0x0000
[08-13_11:45:15:363] epoll fd = 6, events = 0x0001
[08-13_11:45:15:364] get signo: 10
[08-13_11:45:15:364] usbnet_link_change :link:0
[08-13_11:45:15:366] requestRegistrationState2 MCC: 460, MNC: 1, PS: Attached, DataCap: NR5G
[08-13_11:45:15:934] requestSetupDataCall WdsConnectionIPv4Handle: 0x390193a0
[08-13_11:45:15:934] write triger_event: 4101 to qmidevice_contro[  296.947158] invalid ether addr
l_fd
[08-13_11:45:15:936] epoll[  296.953900] [I][mhi_netdev_open] Opened net dev interface
 fd = 7, events = 0x0000
[08-13_11:45:15:936] epoll fd = 6, events = 0x0000
[08-13_11:45:15:936] epoll fd = 7, events = 0x0001
[08-13_11:45:15:937] write signo: 12 to signal_control_fd
[08-13_11:45:15:937] epoll fd = 6, events = 0x0001
[08-13_11:45:15:937] get signo: 12
[08-13_11:45:15:941] usbnet_link_change :link:1
[08-13_11:45:15:943] enter udhcpc_start_pcie
[08-13_11:45:15:944] echo 1 > /sys/module/fibo_mhi/parameters/macaddr_check
[08-13_11:45:15:950] ifconfig pcie_mhi0 up
[08-13_11:45:15:959] ifconfig pcie_mhi0 172.16.0.122 netmask 255.255.255.252
[08-13_11:45:15:963] route add default gw 172.16.0.121 dev pcie_mhi0
[08-13_11:45:15:966] epoll fd = 7, events = 0x0000
[08-13_11:45:30:981] write signo: 12 to signal_control_fd
[08-13_11:45:30:981] epoll fd = 6, events = 0x0001
[08-13_11:45:30:981] get signo: 12
[08-13_11:45:30:988] usbnet_link_change :link:1
[08-13_11:45:30:988] epoll fd = 7, events = 0x0000

```

4IceG, 2023-08-31:
Show result of the command:
`sh -x /usr/share/3ginfo-lite/3ginfo.sh`

Some command is returning the wrong value and messing up the json layout.