4IceG/luci-app-3ginfo-lite #47: 3ginfo.sh working from shell but not populating luci

brian4286, 2023-09-05:
Running 1.0.54-20230903 I purged all and forced a reinstall via opkg.

Seems like the output from the shell script is working but I am not seeing it render in LUCI. I am testing in OpenWrt 23.05.0-rc3 r23389-5deed175a5 so this could be related, however it was working through rc2.

```
root@OpenWrt:~# sh /usr/share/3ginfo-lite/3ginfo.sh
{
"connt":"-",
"conntx":"-",
"connrx":"-",
"modem":"Quectel RM520N-GL",
"mtemp":"33 &deg;C",
"firmware":"RM520NGLAAR03A01M4G",
"cport":"/dev/ttyUSB3",
"protocol":"mbim",
"csq":"",
"signal":"",
"operator_name":"T-Mobile",
"operator_mcc":"310",
"operator_mnc":"260",
"mode":"- n41 (2500 MHz) + n41 (2500 MHz)",
"registration":"0",
"simslot":"1",
"imei":"123456789012345",
"imsi":"123456789012345",
"iccid":"12345678901234567890",
"lac_dec":"0",
"lac_hex":"0",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"0",
"cid_hex":"0",
"pci":"",
"earfcn":"521310",
"pband":"n41 (2500 MHz) @? MHz",
"s1band":"n41 (2500 MHz) @20 MHz",
"s1pci":"271",
"s1earfcn":"501870",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":""
}
```

Happy to dig deeper, wanted to validate I am not alone before I do.

4IceG, 2023-09-05:
Show result of the command:
sh -x /usr/share/3ginfo-lite/3ginfo.sh

brian4286, 2023-09-05:
@4IceG sim is working fine, the modem is online and the AT commands are working as expected. I am sorry I did not follow up sooner but I ran debug earlier the script, it is working. I was trying to find why the variables were not being displayed. I can try to modify the script and echo the variables to figure out which one errors.

I will try to get to that later and debug more from my side as well, apologies dumping it and not providing this output.

```
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB3
+ '[' -z /dev/ttyUSB3 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB3 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=br-wan
+ SUB=@
+ '[[' br-wan '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=br-wan
+ '[' -z br-wan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus br-wan
+ grep '"up": true'
+ NETUP=
+ '[' -n  ]
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=99
+ '[' x99 '=' x ]
+ '[' 99 -ge 0 -a 99 -le 31 ]
+ CSQ=
+ CSQ_PER=
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ awk '-F["]' '/^\+COPS: .,2/ {print $2}'
+ COPS_NUM=310260
[... 554 lines cut ...]
"tac_d":"",
"cid_dec":"0",
"cid_hex":"0",
"pci":"",
"earfcn":"521310",
"pband":"n41 (2500 MHz) @? MHz",
"s1band":"n41 (2500 MHz) @20 MHz",
"s1pci":"271",
"s1earfcn":"501870",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":""
}
+ exit 0
````

4IceG, 2023-09-05:
@obsy Cezary chyba znow będą skrypty do korekty.. ciagle coś wychodzi :)

brian4286, 2023-09-05:
For example, where the variable for $CSQ is created, I immediately echo'ed out the variable and it works. So somewhere below that the variable gets cleared.

```
root@OpenWrt:~# sh  /usr/share/3ginfo-lite/3ginfo.sh
>>>>>>>>>>> CSQ = 99 <<<<<<<<<<<<
{
"connt":"-",
"conntx":"-",
"connrx":"-",
"modem":"Quectel RM520N-GL",
"mtemp":"34 &deg;C",
"firmware":"RM520NGLAAR03A01M4G",
"cport":"/dev/ttyUSB3",
"protocol":"mbim",
"csq":"",
"signal":"",
"operator_name":"T-Mobile",
"operator_mcc":"310",
"operator_mnc":"260",
"mode":"- n41 (2500 MHz) + n41 (2500 MHz)",
"registration":"0",
"simslot":"1",
"imei":";;redacted;;",
"imsi":";;redacted;;",
"iccid":";;redacted;;",
"lac_dec":"0",
"lac_hex":"0",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"0",
"cid_hex":"0",
"pci":"",
"earfcn":"521310",
"pband":"n41 (2500 MHz) @? MHz",
"s1band":"n41 (2500 MHz) @20 MHz",
"s1pci":"271",
"s1earfcn":"501870",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":""
}
```

obsy, 2023-09-05:
To jest tryb połączenia którego u nas nie ma i nie parsujemy: 11, czyli wg dokumentacji "NR connected to 5GCN". Wg danych wygląda jak czyste 5G (SA). Zdecydowanie do dodania do skryptów, bo teraz po prostu tego nie analizujemy.

obsy, 2023-09-05:
@brian4286: this is a new mode that requires our script changes to be supported. Right now the scripts don't parse it like they should.

brian4286, 2023-09-05:
5GSA FTW? 😄

obsy, 2023-09-05:
@4IceG: tylko że wg tych danych +CREG: 2,0 czyli wg dokumentacji on nie jest zarejestrowany w sieci i nie szuka sieci. Czyli wg ideologi dane nie powinny być parsowane bo aktywnego połączenia nie ma. Jeżeli on ma połączenie jak napisał to znów coś quectel kłamie i nie mamy statusu rejestracji karty sim w sieci.

4IceG, 2023-09-05:
@obsy Pewnie zaraz wyjdzie że jest jakies polecenie +CREG5G:, ktore dane pokazuje.. i tyle po Naszej logice. Bądź tu mądry.. tylko na koncu czego się trzymać i czym kierowac.

obsy, 2023-09-05:
@brian4286: Do you have a normal, working internet connection? Can you show the result again:
```
ping -c3 google.com
sh -x /usr/share/3ginfo-lite/3ginfo.sh
```
after ifdown wan; ifup wan (or change "wan" if you have a section with a name other than wan).

obsy, 2023-09-05:
> @obsy Pewnie zaraz wyjdzie że jest jakies polecenie +CREG5G:, ktore dane pokazuje.. i tyle po Naszej logice. Bądź tu mądry.. tylko na koncu czego się trzymać i czym kierowac.

AT+C5GREG jest :)

brian4286, 2023-09-05:
```
root@OpenWrt:/usr/share/3ginfo-lite# sh -x 3ginfo.sh 
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB3
+ '[' -z /dev/ttyUSB3 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB3 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=br-wan
+ SUB=@
+ '[[' br-wan '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=br-wan
+ '[' -z br-wan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus br-wan
+ grep '"up": true'
+ NETUP=
+ '[' -n  ]
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=99
+ '[' x99 '=' x ]
+ '[' 99 -ge 0 -a 99 -le 31 ]
+ CSQ=
+ CSQ_PER=
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ awk '-F["]' '/^\+COPS: .,2/ {print $2}'
+ COPS_NUM=310260
+ '[' x310260 '=' x ]
+ COPS_MCC=310
+ COPS_MNC=260
[... 591 lines cut ...]
Tue Sep  5 15:55:57 EDT 2023
PING 1.0.0.1 (1.0.0.1): 56 data bytes
64 bytes from 1.0.0.1: seq=0 ttl=53 time=31.213 ms
64 bytes from 1.0.0.1: seq=1 ttl=53 time=31.406 ms
64 bytes from 1.0.0.1: seq=2 ttl=53 time=25.652 ms

--- 1.0.0.1 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 25.652/29.423/31.406 ms
root@OpenWrt:/usr/share/3ginfo-lite# date; ping -c3 9.9.9.9
Tue Sep  5 15:56:04 EDT 2023
PING 9.9.9.9 (9.9.9.9): 56 data bytes
64 bytes from 9.9.9.9: seq=0 ttl=56 time=27.691 ms
64 bytes from 9.9.9.9: seq=1 ttl=56 time=27.348 ms
64 bytes from 9.9.9.9: seq=2 ttl=56 time=40.015 ms

--- 9.9.9.9 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 27.348/31.684/40.015 ms
```

> after ifdown wan; ifup wan (or change "wan" if you have a section with a name other than wan).
Can you rephrase this part. Are you asking me to run the command ifdown my primary WAN connection, then bring it back online? 

I don't call it wan, it is called wwan on my box if that helps.

obsy, 2023-09-05:
And show

sms_tool -d /dev/ttyUSB3 at "AT+C5GREG?"

brian4286, 2023-09-05:
Here you go...

``` 
root@OpenWrt:/usr/share/3ginfo-lite# sms_tool -d /dev/ttyUSB3 at "AT+C5GREG?"
AT+C5GREG?
+C5GREG: 0,1
```

obsy, 2023-09-05:
Better:

sms_tool -d /dev/ttyUSB3 at "AT+C5GREG=2;+C5GREG?"

??

brian4286, 2023-09-05:
```
root@OpenWrt:/usr/share/3ginfo-lite# sms_tool -d /dev/ttyUSB3 at "AT+C5GREG=2;+C5GREG?"
AT+C5GREG=2;+C5GREG?
+C5GREG: 2,1,"59E700","14758F12F",11,1,"01"
```

obsy, 2023-09-05:
Thanks. Ok, we should be able to do something with that data.

brian4286, 2023-09-05:
Just @ me and I will be happy to pull down the change and test before you package it up.

obsy, 2023-09-06:
First try:
```
{
  "csq": "",
  "signal": "",
  "operator_name": "T-Mobile",
  "operator_mcc": "310",
  "operator_mnc": "260",
  "mode": "5G n41 (2500 MHz) / n41 (2500 MHz)",
  "registration": "1",
  "lac_dec": "0",
  "lac_hex": "0",
  "cid_dec": "5491978543",
  "cid_hex": "14758F12F",
  "addon": [
    {
      "idx": 10,
      "key": "Temperature",
      "value": "34 &deg;C"
    },
    {
      "idx": 23,
      "key": "TAC",
      "value": "5891840 (59E700)"
    },
    {
      "idx": 32,
      "key": "Bandwidth",
      "value": "80 MHz"
    },
    {
      "idx": 36,
      "key": "RSRP",
      "value": "-89 dBm"
    },
    {
      "idx": 37,
      "key": "RSRQ",
      "value": "-11 dB"
    },
    {
      "idx": 38,
      "key": "SINR",
      "value": "19 dB"
    },
    {
      "idx": 30,
      "key": "Primary band",
      "value": "n41 (2500 MHz)"
    },
    {
      "idx": 34,
      "key": "EARFCN",
      "value": "521310"
    },
    {
      "idx": 33,
      "key": "PCI",
      "value": "271"
    },
    {
      "idx": 54,
      "key": "(S1) EARFCN",
      "value": "501870"
    },
    {
      "idx": 53,
      "key": "(S1) PCI",
      "value": "271"
    },
    {
      "idx": 56,
      "key": "(S1) RSRP",
      "value": "0 dBm"
    },
    {
      "idx": 50,
      "key": "(S1) band",
      "value": "n41 (2500 MHz) @20 MHz"
    }
  ]
}
```

@4IceG : o ile pamiętasz nie dawno walczyliśmy z parametrami dla SCC, bo się okazało że dla naszego NR5G-NSA inaczej trzeba to wyliczać (brak było pola "state"). No to tu dla utrudnienia nie ma pola "state" dla PCC więc wylicza się to inaczej, ale jest "state" dla SCC i bandu NR5G, więc znów wylicza się inaczej. Robi się cała masa if'ów.

4IceG, 2023-09-06:
@obsy U Ciebie Cezary wyglada fajnie, mam nadzieje że jakoś odnajde się w tych skryptach i uda się to przenieść dla LuCI.

obsy, 2023-09-08:
@brian4286 : work in progress

4IceG, 2023-09-10:
Hi @brian4286 ,
Check the new version of the package.

brian4286, 2023-09-11:
@4IceG I am on a different tower right now so can't be certain if it is parsing correctly. I will update this issue when I am back at that tower over the next week or two.