4IceG/luci-app-3ginfo-lite #48: Not parsing status screen if any default LTE bands are removed

brian4286, 2023-09-11:
Using modemband with a RM520n-gl if I remove any single/combination of the LTE bands 3ginfo-light no longer parses the status screen. This seem to only occur as long as the modem is connected to pure 4G LTE bands, once it connects to 5GSA it parses correctly again.

```
SyntaxError JSON Parse error: Unterminated string

Error: JSON Parse error: Unterminated string
```

Here is the debug output from the 3ginfo.sh. I want to draw your attention to "**addon: not found**". I checked and the line numbers don't exists in /usr/share/3ginfo-lite/modem/2c7c0801. I noticed this was new to 1.0.55-20230910.

```
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB3
+ '[' -z /dev/ttyUSB3 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB3 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=wwan0
+ SUB=@
+ '[[' wwan0 '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wwan0
+ '[' -z wwan0 ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wwan0
+ grep '"up": true'
+ NETUP=
+ '[' -n  ]
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11

+CREG: 2,0

'K
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=99
+ '[' x99 '=' x ]
+ '[' 99 -ge 0 -a 99 -le 31 ]
+ CSQ=
+ CSQ_PER=
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"T-Mobile",11

+COPS: 0,2,"310260",11
[... 689 lines cut ...]
"tac_d":"",
"cid_dec":"4241194012",
"cid_hex":"939204254",
",ci":"24
"earfcn":"126490",
"pband":"n71 (600 MHz)",
"s1band":"n41 (2500 MHz) @10 MHz",
"s1pci":"625",
"s1earfcn":"520110",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-104",
"rsrq":"-13",
"rssi":"-104",
"sinr":"11"
}
+ exit 0
```

4IceG, 2023-09-11:
I love analyzing scripts to look for errors. I'll take a look at it this week.

4IceG, 2023-09-22:
Hi @brian4286 , 
Check if this version works properly.
[1.0.56-20230922.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12703038/1.0.56-20230922.zip)