4IceG/luci-app-3ginfo-lite #51: Telit LN940 incorrect data

jfrogg, 2023-09-25:
Using Telit LN940 (1bc7:1900), but the same is applicable to 1bc7:1901, 03f0:0857, 03f0:0a57.
The app shows this:
"Technology    B2 (1900 MHz)"
but it should be like this:
"Technology    LTE B3 (1800 MHz)"

"LTE " is missing and "B2" comes from a typo here:
https://github.com/4IceG/luci-app-3ginfo-lite/blob/main/luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/modem/1bc71900#L23
and also in 1bc71901, 03f00857, 03f00a57

Thank you

jfrogg, 2023-09-26:
SIM info and band fixed, "LTE | " is still missing in Technology.
[1bc71900.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12731125/1bc71900.zip)

4IceG, 2023-09-27:
hi @jfrogg ,
Replace this file and see if "LTE/LTE-A" will be visible.

[1bc71900_LTEFix.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12735120/1bc71900_LTEFix.zip)

jfrogg, 2023-09-27:
Thank you, but it looks a bit weird now:
"LTE-A | -A | + B1 (2100 MHz)" appears initially and in a moment it gets replaced with "B3 (1800 MHz)".

4IceG, 2023-09-27:
Ok, so I added the MODE="LTE-A |" line unnecessarily, the problem was only with LTE itself. Delete this line from the file and it should be fine.

jfrogg, 2023-09-27:
Now it is either "B3 (1800 MHz) + B1 (2100 MHz)" or "B3 (1800 MHz)" without any indication of LTE or LTE-A.

4IceG, 2023-09-27:
Ok, show me the output of the AT^CA_INFO? command. I will correct the scripts in the evening.

jfrogg, 2023-09-27:
Here it is:
```
# sms_tool -d /dev/ttyUSB2 at "AT^CA_INFO?"
AT^CA_INFO?
PCC info: Band is LTE_B3, Band_width is 15.0 MHz

# sms_tool -d /dev/ttyUSB2 at "AT^CA_INFO?"
AT^CA_INFO?
PCC info: Band is LTE_B3, Band_width is 15.0 MHz
SCC1 info: Band is LTE_B1, Band_width is 10.0 MHz

```

Today I had more time than yesterday, so below is fixed and slightly optimised version.
[1bc71900.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12738161/1bc71900.zip)

4IceG, 2023-09-27:
I quickly checked @jfrogg your script and I would replace this line 21
`MODE="$MODE | $(band $PC "LTE ")"`
for this version
`MODE="LTE | $(band $PC)"`

jfrogg, 2023-09-27:
That works, thank you. I tried to stay close to the original )
[1bc71900.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/12739702/1bc71900.zip)

closed 2023-09-30 by 4IceG