4IceG/luci-app-3ginfo-lite #55: Add support for Foxconn T99W175 also known as Dell DW5930e

4IceG, 2023-12-28:
Would you like to add support for Foxconn T99W175 also known as Dell DW5930e? 
Device connected to board with PCI MHI and AT port available at /dev/wwan0at0, MBIM at /dev/wwan0mbim0 and QCDM at /dev/wwan0qcdm0

AT Commands available here:
https://www.codico.com/media/productattach/m/v/mv31_w_atc_v00058_2111538_1.pdf

by now only this info are shown
[image]

_Originally posted by @riccardoferti in https://github.com/4IceG/luci-app-3ginfo-lite/issues/28#issuecomment-1871015490_

4IceG, 2023-12-28:
Hi @riccardoferti,

I moved your post to a new thread. We can try to add support for this modem.

First, show the output of the commands:
```
cat /sys/kernel/debug/usb/devices
sms_tool -d /dev/wwan0at0 at "at^ca_info?"
sms_tool -d /dev/wwan0at0 at "at^debug?"
```

riccardoferti, 2023-12-28:
> cat /sys/kernel/debug/usb/devices

No USB devices available, but:

cat /sys/kernel/debug/mhi/mhi0/devices

`mhi0: type: Controller dev_wake: 0
LOOPBACK: type: Transfer dev_wake: 0 channels: 0(UL)/1(DL)
DIAG: type: Transfer dev_wake: 0 channels: 4(UL)/5(DL)
MBIM: type: Transfer dev_wake: 0 channels: 12(UL)/13(DL)
DUN: type: Transfer dev_wake: 0 channels: 32(UL)/33(DL)
IP_HW0_MBIM: type: Transfer dev_wake: 0 channels: 100(UL)/101(DL)`

sms_tool -d /dev/wwan0at0 at "at^ca_info?"

`LTE serving information :
PCC info: Band is LTE_B3, Band_width is 20.0 MHz
SCC1 info: Band is LTE_B20, Band_width is 10.0 MHz
SCC2 info: Band is LTE_B7, Band_width is 15.0 MHz
PCC info: Band is NR5G_N78, Band_width is 80.0 MHz`

sms_tool -d /dev/wwan0at0 at "at^debug?"

`RAT:LTE+NR
mcc:222,mnc:10
lte_cell_id:19403297
lte_tac:43083
lte_tx_pwr:NA
lte_ant_rsrp:rx_diversity:3 (-94.1dBm,-103.5dBm,NA,NA)
pcell: lte_band:3 lte_band_width:20.0MHz
channel:1850 pci:389
lte_rsrp:-93.0dBm,rsrq:-12.7dB
lte_rssi:-55.9dBm,lte_snr:1.0dB
scell: lte_band:20 lte_band_width:10.0MHz
channel:6400 pci:230
lte_rsrp:-84.2dBm,rsrq:-11.9dB
lte_rssi:-64.6dBm,lte_snr:NA
scell: lte_band:7 lte_band_width:15.0MHz
channel:3025 pci:53
lte_rsrp:-97.1dBm,rsrq:-11.6dB
lte_rssi:-77.5dBm,lte_snr:NA
scell: lte_band:1 lte_band_width:15.0MHz
channel:525 pci:168
lte_rsrp:-93.2dBm,rsrq:-12.0dB
lte_rssi:-73.4dBm,lte_snr:NA

nr_band:n78
nr_band_width:80.0MHz
nr_channel:643296
nr_pci:736
nr_rsrp:-98dBm rx_diversity: 15 (-110.1dBm,-98.0dBm,-103.9dBm,-103.8dBm)
nr_rsrq:-11dB
nr_snr:21.0dB`

4IceG, 2023-12-29:
@obsy Cezary masz moze pomysl jak zidentyfikowac w systemie takie modemy, aby je potem przypisac do skryptow? 

Same wyniki tez sa ciekawe, dwa pasma glowne.. 5G komplikuje znow skrypty.

obsy, 2023-12-29:
Kod nie jest przystosowany do innych urządzeń niż usb. Albo robisz nowe obejście tak jak zrobiłeś dla np. hilinka albo trzeba sporo zmodyfikować główny kod żeby wykrywał też urządzenia pci. 
Przydał by się bezpośredni dostęp do takiego modemu.

obsy, 2023-12-29:
@riccardoferti  

Could you show also 
```
ls -al /dev/* 
```

riccardoferti, 2023-12-29:
> @riccardoferti
> 
> Could you show also
> 
> ```
> ls -al /dev/* 
> ```

```
crw-------    1 root     root        5,   1 Dec 29 00:42 /dev/console
crw-------    1 root     root       10, 125 Dec 29 00:42 /dev/cpu_dma_latency
crw-------    1 root     root       29,   0 Dec 29 00:42 /dev/fb0
lrwxrwxrwx    1 root     root            13 Dec 29 00:42 /dev/fd -> /proc/self/fd
crw-rw-rw-    1 root     root        1,   7 Dec 29 00:42 /dev/full
crw-------    1 root     root       10, 228 Dec 29 00:42 /dev/hpet
crw-------    1 root     root       10, 183 Dec 29 00:42 /dev/hwrng
crw-------    1 root     root       89,   0 Dec 29 00:42 /dev/i2c-0
crw-------    1 root     root       89,   1 Dec 29 00:42 /dev/i2c-1
crw-------    1 root     root       89,   2 Dec 29 00:42 /dev/i2c-2
crw-------    1 root     root       89,   3 Dec 29 00:42 /dev/i2c-3
crw-------    1 root     root       89,   4 Dec 29 00:42 /dev/i2c-4
crw-------    1 root     root       89,   5 Dec 29 00:42 /dev/i2c-5
crw-------    1 root     root        1,  11 Dec 29 00:42 /dev/kmsg
srw-rw-rw-    1 root     root             0 Dec 29 00:42 /dev/log
crw-------    1 root     root       10, 237 Dec 29 00:42 /dev/loop-control
brw-------    1 root     root        7,   0 Dec 29 00:42 /dev/loop0
brw-------    1 root     root        7,   1 Dec 29 00:42 /dev/loop1
brw-------    1 root     root        7,   2 Dec 29 00:42 /dev/loop2
brw-------    1 root     root        7,   3 Dec 29 00:42 /dev/loop3
brw-------    1 root     root        7,   4 Dec 29 00:42 /dev/loop4
brw-------    1 root     root        7,   5 Dec 29 00:42 /dev/loop5
brw-------    1 root     root        7,   6 Dec 29 00:42 /dev/loop6
brw-------    1 root     root        7,   7 Dec 29 00:42 /dev/loop7
crw-rw-rw-    1 root     root        1,   3 Dec 29 00:42 /dev/null
crw-------    1 root     root       10, 144 Dec 29 00:42 /dev/nvram
crw-------    1 root     root        1,   4 Dec 29 00:42 /dev/port
crw-------    1 root     root      108,   0 Dec 29 00:42 /dev/ppp
crw-rw-rw-    1 root     root        5,   2 Dec 29 09:34 /dev/ptmx
crw-------    1 root     root      249,   0 Dec 29 00:42 /dev/ptp0
crw-rw-rw-    1 root     root        1,   8 Dec 29 00:42 /dev/random
crw-------    1 root     root      251,   0 Dec 29 00:42 /dev/rtc0
brw-------    1 root     root        8,   0 Dec 29 00:42 /dev/sda
brw-------    1 root     root        8,   1 Dec 29 00:42 /dev/sda1
brw-------    1 root     root      259,   0 Dec 29 00:42 /dev/sda128
brw-------    1 root     root        8,   2 Dec 29 00:42 /dev/sda2
brw-------    1 root     root        8,  16 Dec 29 00:42 /dev/sdb
brw-------    1 root     root        8,  17 Dec 29 00:42 /dev/sdb1
brw-------    1 root     root        8,  18 Dec 29 00:42 /dev/sdb2
brw-------    1 root     root        8,  19 Dec 29 00:42 /dev/sdb3
brw-------    1 root     root        8,  20 Dec 29 00:42 /dev/sdb4
brw-------    1 root     root        8,  21 Dec 29 00:42 /dev/sdb5
lrwxrwxrwx    1 root     root             8 Dec 29 00:42 /dev/shm -> /tmp/shm
lrwxrwxrwx    1 root     root            15 Dec 29 00:42 /dev/stderr -> /proc/self/fd/2
lrwxrwxrwx    1 root     root            15 Dec 29 00:42 /dev/stdin -> /proc/self/fd/0
lrwxrwxrwx    1 root     root            15 Dec 29 00:42 /dev/stdout -> /proc/self/fd/1
crw-rw-rw-    1 root     root        5,   0 Dec 29 00:42 /dev/tty
crw-rw----    1 root     dialout     4,   0 Dec 29 00:42 /dev/tty0
crw-rw----    1 root     dialout     4,   1 Dec 29 00:43 /dev/tty1
crw-rw----    1 root     dialout     4,  10 Dec 29 00:42 /dev/tty10
crw-rw----    1 root     dialout     4,  11 Dec 29 00:42 /dev/tty11
crw-rw----    1 root     dialout     4,  12 Dec 29 00:42 /dev/tty12
crw-rw----    1 root     dialout     4,  13 Dec 29 00:42 /dev/tty13
crw-rw----    1 root     dialout     4,  14 Dec 29 00:42 /dev/tty14
crw-rw----    1 root     dialout     4,  15 Dec 29 00:42 /dev/tty15
crw-rw----    1 root     dialout     4,  16 Dec 29 00:42 /dev/tty16
crw-rw----    1 root     dialout     4,  17 Dec 29 00:42 /dev/tty17
crw-rw----    1 root     dialout     4,  18 Dec 29 00:42 /dev/tty18
crw-rw----    1 root     dialout     4,  19 Dec 29 00:42 /dev/tty19
crw-rw----    1 root     dialout     4,   2 Dec 29 00:42 /dev/tty2
crw-rw----    1 root     dialout     4,  20 Dec 29 00:42 /dev/tty20
[... 81 lines cut ...]
drwxr-xr-x    3 root     root            60 Dec 29 00:42 .
drwxr-xr-x    6 root     root          2920 Dec 29 00:42 ..
drwxr-xr-x    5 root     root           100 Dec 29 00:42 usb

/dev/cpu:
drwxr-xr-x    6 root     root           120 Dec 29 00:42 .
drwxr-xr-x    6 root     root          2920 Dec 29 00:42 ..
drwxr-xr-x    2 root     root            80 Dec 29 00:42 0
drwxr-xr-x    2 root     root            80 Dec 29 00:42 1
drwxr-xr-x    2 root     root            80 Dec 29 00:42 2
drwxr-xr-x    2 root     root            80 Dec 29 00:42 3

/dev/dri:
drwxr-xr-x    2 root     root            80 Dec 29 00:42 .
drwxr-xr-x    6 root     root          2920 Dec 29 00:42 ..
crw-------    1 root     root      226,   0 Dec 29 00:42 card0
crw-------    1 root     root      226, 128 Dec 29 00:42 renderD128

/dev/pts:
drwxr-xr-x    2 root     root             0 Dec 29 00:42 .
drwxr-xr-x    6 root     root          2920 Dec 29 00:42 ..
crw--w--w-    1 root     root      136,   0 Dec 29 09:34 0
c---------    1 root     root        5,   2 Dec 29 00:42 ptmx

```

obsy, 2023-12-29:
And show us:
```
AT+CPIN?
AT+CSQ
AT+COPS=3,0;+COPS?
AT+COPS=3,2;+COPS?
AT+CREG=2;+CREG?
```

riccardoferti, 2023-12-29:
> sms_tool -d /dev/wwan0at0 at "AT+CPIN?"
> sms_tool -d /dev/wwan0at0 at "AT+CSQ"
> sms_tool -d /dev/wwan0at0 at "AT+COPS=3,0;+COPS?"
> sms_tool -d /dev/wwan0at0 at "AT+COPS=3,2;+COPS?"
> sms_tool -d /dev/wwan0at0 at "AT+CREG=2;+CREG?"
```

sms_tool -d /dev/wwan0at0 at "AT+CPIN?"

+CPIN: READY

root@OpenWrt:~# sms_tool -d /dev/wwan0at0 at "AT+CSQ"

+CSQ: 29,99

root@OpenWrt:~# sms_tool -d /dev/wwan0at0 at "AT+COPS=3,0;+COPS?"

+COPS: 0,0,"vodafone IT",13

root@OpenWrt:~# sms_tool -d /dev/wwan0at0 at "AT+COPS=3,2;+COPS?"

+COPS: 0,2,"22210",13

root@OpenWrt:~# sms_tool -d /dev/wwan0at0 at "AT+CREG=2;+CREG?"

+CREG: 2,1,"A84B","1281221",13

```

riccardoferti, 2023-12-29:
> Hmm, very sad that /dev/mhi_DIAG does not exist like in Quectel.

It should be /dev/wwan0qcdm0 but don't know if It works

riccardoferti, 2023-12-29:
what i see in device:

root@OpenWrt:/sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/mhi0# ls
mhi0_DIAG         mhi0_IP_HW0_MBIM  mhi0_MBIM         power             subsystem         wakeup
mhi0_DUN          mhi0_LOOPBACK     oem_pk_hash       serial_number     uevent            wwan

obsy, 2023-12-29:
```
sms_tool -d /dev/wwan0at0 at "AT+CGMI"
sms_tool -d /dev/wwan0at0 at "AT+CGMM"
sms_tool -d /dev/wwan0at0 at "AT+CGMR"
sms_tool -d /dev/wwan0at0 at "AT+CGSN"
sms_tool -d /dev/wwan0at0 at "AT+ICCID"
```

riccardoferti, 2023-12-29:
> sms_tool -d /dev/wwan0at0 at "AT+CGMI"
> sms_tool -d /dev/wwan0at0 at "AT+CGMM"
> sms_tool -d /dev/wwan0at0 at "AT+CGMR"
> sms_tool -d /dev/wwan0at0 at "AT+CGSN"
> sms_tool -d /dev/wwan0at0 at "AT+ICCID"

`AT+CGMI DELL`
`AT+CGMM T99W175`
`AT+CGMR T99W175.F0.1.0.0.9.VF.008  1  [Jan 20 2023 12:00:00]`
`AT+CGSN 3522801100xxxxx`
`AT+ICCID ICCID: 8939104520038678xxxx`

obsy, 2023-12-30:
I made the parser and it seems to work. But I'm still wondering how to create a framework to support PCI devices.

riccardoferti, 2023-12-30:
> I made the parser and it seems to work. But I'm still wondering how to create a framework to support PCI devices.

It's not simple to use AT port as device?

obsy, 2023-12-30:
Nope. It's a matter of discovering what is available and what script to use to process it. The script itself is simple.

riccardoferti, 2024-01-02:
> Nope. It's a matter of discovering what is available and what script to use to process it. The script itself is simple.

On USB 2.0 debug mode, it shows AT console in /dev/ttyUSB3

obsy, 2024-01-02:
Show us 
```
cat /sys/kernel/debug/usb/devices
```
in this "usb 2.0 debug mode".

riccardoferti, 2024-01-02:
> Show us
> 
> ```
> cat /sys/kernel/debug/usb/devices
> ```
> 
> in this "usb 2.0 debug mode".

```
T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=05c6 ProdID=90d5 Rev= 4.14
S:  Manufacturer=Microsoft
S:  Product=Generic Mobile Broadband Adapter
S:  SerialNumber=90b86942
C:* #Ifs= 6 Cfg#= 1 Atr=a0 MxPwr=500mA
A:  FirstIf#= 2 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=30 Driver=option
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=42 Prot=01 Driver=option
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=32ms
I:  If#= 3 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
I:* If#= 3 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
E:  Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=option
E:  Ad=84(I) Atr=03(Int.) MxPS=  64 Ivl=32ms
I:* If#= 5 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=86(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
```

obtained assigning 
`echo "05c6 90d5 ff" > /sys/bus/usb-serial/drivers/option1/new_id`

jfrogg, 2024-01-03:
Let me share some thoughts on _discovery_ process.
The user will enter or select the serial port in the configuration. No change here.
Knowing serial device like "/dev/wwan0at0" allows the parent device be discovered the same way as with USB, but in this case the result will be a PCI path/device. The code should be similar to what is currently used in https://github.com/4IceG/luci-app-3ginfo-lite/blob/main/luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/3ginfo.sh#L151

Unfortunately, I do not have a suitable device to experiment with at the moment, so this is an example based on PCI Wi-Fi card:

```
cat /sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/vendor
0x14c3
cat /sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/device
0x7916
```
It should be possible to find a path similar to "1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0" for "/dev/wwan0at0" from SysFS.

"vendor + product" combination is exactly what is currently used to select modem-specific scripts.

Hope it helps )

riccardoferti, 2024-01-03:
> Let me share some thoughts on _discovery_ process.
> The user will enter or select the serial port in the configuration. No change here.
> Knowing serial device like "/dev/wwan0at0" allows the parent [...]

Clear, It Is as you said

obsy, 2024-01-04:
@riccardoferti

Show me
```
find /sys/bus/pci
find /sys -name wwan0at0
```
also install pciutils and show lspci -nn

riccardoferti, 2024-01-06:
> lspci -nn

```
find /sys/bus/pci
/sys/bus/pci
/sys/bus/pci/uevent
/sys/bus/pci/slots
/sys/bus/pci/drivers_autoprobe
/sys/bus/pci/devices
/sys/bus/pci/devices/0000:00:1f.2
/sys/bus/pci/devices/0000:00:1c.0
/sys/bus/pci/devices/0000:00:1f.0
/sys/bus/pci/devices/0000:02:00.0
/sys/bus/pci/devices/0000:00:04.0
/sys/bus/pci/devices/0000:00:1c.5
/sys/bus/pci/devices/0000:00:16.0
/sys/bus/pci/devices/0000:01:00.0
/sys/bus/pci/devices/0000:00:1b.0
/sys/bus/pci/devices/0000:00:19.0
/sys/bus/pci/devices/0000:00:1f.3
/sys/bus/pci/devices/0000:00:00.0
/sys/bus/pci/devices/0000:00:03.0
/sys/bus/pci/devices/0000:00:1d.0
/sys/bus/pci/devices/0000:00:02.0
/sys/bus/pci/devices/0000:00:14.0
/sys/bus/pci/rescan
/sys/bus/pci/resource_alignment
/sys/bus/pci/drivers_probe
/sys/bus/pci/drivers
/sys/bus/pci/drivers/pata_amd
/sys/bus/pci/drivers/pata_amd/uevent
/sys/bus/pci/drivers/pata_amd/bind
/sys/bus/pci/drivers/pata_amd/new_id
/sys/bus/pci/drivers/pata_amd/remove_id
/sys/bus/pci/drivers/pata_amd/unbind
/sys/bus/pci/drivers/ata_piix
/sys/bus/pci/drivers/ata_piix/uevent
/sys/bus/pci/drivers/ata_piix/bind
/sys/bus/pci/drivers/ata_piix/new_id
/sys/bus/pci/drivers/ata_piix/remove_id
/sys/bus/pci/drivers/ata_piix/unbind
/sys/bus/pci/drivers/ehci-pci
/sys/bus/pci/drivers/ehci-pci/uevent
/sys/bus/pci/drivers/ehci-pci/bind
/sys/bus/pci/drivers/ehci-pci/new_id
/sys/bus/pci/drivers/ehci-pci/remove_id
/sys/bus/pci/drivers/ehci-pci/unbind
/sys/bus/pci/drivers/ehci-pci/0000:00:1d.0
/sys/bus/pci/drivers/mptsas
/sys/bus/pci/drivers/mptsas/uevent
/sys/bus/pci/drivers/mptsas/bind
/sys/bus/pci/drivers/mptsas/new_id
/sys/bus/pci/drivers/mptsas/remove_id
/sys/bus/pci/drivers/mptsas/unbind
/sys/bus/pci/drivers/xhci_hcd
/sys/bus/pci/drivers/xhci_hcd/uevent
/sys/bus/pci/drivers/xhci_hcd/bind
/sys/bus/pci/drivers/xhci_hcd/new_id
/sys/bus/pci/drivers/xhci_hcd/remove_id
/sys/bus/pci/drivers/xhci_hcd/unbind
/sys/bus/pci/drivers/xhci_hcd/0000:00:14.0
/sys/bus/pci/drivers/pata_oldpiix
/sys/bus/pci/drivers/pata_oldpiix/uevent
/sys/bus/pci/drivers/pata_oldpiix/bind
/sys/bus/pci/drivers/pata_oldpiix/new_id
/sys/bus/pci/drivers/pata_oldpiix/remove_id
/sys/bus/pci/drivers/pata_oldpiix/unbind
/sys/bus/pci/drivers/intel_pch_thermal
/sys/bus/pci/drivers/intel_pch_thermal/uevent
/sys/bus/pci/drivers/intel_pch_thermal/bind
[... 360 lines cut ...]
```
 find /sys -name wwan0at0
/sys/class/wwan/wwan0at0
/sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/mhi0/wwan/wwan0/wwan0at0
```

```
lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Broadwell-U Host Bridge -OPI [8086:1604] (rev 09)
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 5500 [8086:1616] (rev 09)
00:03.0 Audio device [0403]: Intel Corporation Broadwell-U Audio Controller [8086:160c] (rev 09)
00:04.0 Signal processing controller [1180]: Intel Corporation Broadwell-U Processor Thermal Subsystem [8086:1603] (rev 09)
00:14.0 USB controller [0c03]: Intel Corporation Wildcat Point-LP USB xHCI Controller [8086:9cb1] (rev 03)
00:16.0 Communication controller [0780]: Intel Corporation Wildcat Point-LP MEI Controller #1 [8086:9cba] (rev 03)
00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection (3) I218-LM [8086:15a2] (rev 03)
00:1b.0 Audio device [0403]: Intel Corporation Wildcat Point-LP High Definition Audio Controller [8086:9ca0] (rev 03)
00:1c.0 PCI bridge [0604]: Intel Corporation Wildcat Point-LP PCI Express Root Port #1 [8086:9c90] (rev e3)
00:1c.5 PCI bridge [0604]: Intel Corporation Wildcat Point-LP PCI Express Root Port #6 [8086:9c9a] (rev e3)
00:1d.0 USB controller [0c03]: Intel Corporation Wildcat Point-LP USB EHCI Controller [8086:9ca6] (rev 03)
00:1f.0 ISA bridge [0601]: Intel Corporation Wildcat Point-LP LPC Controller [8086:9cc3] (rev 03)
00:1f.2 SATA controller [0106]: Intel Corporation Wildcat Point-LP SATA Controller [AHCI Mode] [8086:9c83] (rev 03)
00:1f.3 SMBus [0c05]: Intel Corporation Wildcat Point-LP SMBus Controller [8086:9ca2] (rev 03)
01:00.0 SD Host controller [0805]: O2 Micro, Inc. SD/MMC Card Reader Controller [1217:8520] (rev 01)
02:00.0 Wireless controller [0d40]: Foxconn International, Inc. Device [105b:e0b0]
```

jfrogg, 2024-01-06:
@riccardoferti please run when you have a chance:
`readlink -f /sys/class/wwan/wwan0at0/device`

BTW, could you please share the details about the module you use - t99w175.?? version and whether it is Dell/HP/Lenovo branded.

obsy, 2024-01-06:
cat /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/vendor
cat /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/device

?

riccardoferti, 2024-01-06:
> readlink -f /sys/class/wwan/wwan0at0/device

 readlink -f /sys/class/wwan/wwan0at0/device
/sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/mhi0/wwan/wwan0

The device Is a Dell Prototype version, bought on AliExpress.
You can change the device vendor switching at^customer=x
1 Dell
0 Foxconn
4 Lenovo
6 HP (maybe)

Mine Is 0 actually, with latest Lenovo firmware installed. It's the latest in Building time.

riccardoferti, 2024-01-06:
> cat /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/vendor

0x105b

> cat /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/device

0xe0b0

jfrogg, 2024-01-06:
@riccardoferti I was mainly interested what is printed/engraved on the modem itself. Thank you.

riccardoferti, 2024-01-06:
> @riccardoferti I was mainly interested what is printed/engraved on the modem itself. Thank you.

Who wants can DM me on Telegram @nerdherd96

obsy, 2024-01-08:
@4IceG : zaimplementuj to u siebie: https://github.com/obsy/packages/commit/e1daf558d51b275efe732ff5cdf6de4c062a7e6d

Jeżeli będzie wszystko ok to trzeba rozważyć czy nie przenieść danych do oddzielnych podkatalogów pci/ i usb/

4IceG, 2024-01-08:
> @4IceG : zaimplementuj to u siebie: [obsy/packages@e1daf55](https://github.com/obsy/packages/commit/e1daf558d51b275efe732ff5cdf6de4c062a7e6d)
> 
> Jeżeli będzie wszystko ok to trzeba rozważyć czy nie przenieść danych do oddzielnych podkatalogów pci/ i usb/

Ok, thx @obsy Cezary.
Jeżeli kod mnie nie zaskoczy to postaram się możliwie szybko go zaimplementować u siebie.

Co do modyfikacji podkatalogów myślę, że jest to wskazana, przyszłościowa koncepcja.

obsy, 2024-01-08:
Teraz do testów dodałem po prostu prefix "pci". Ale docelowo trzeba będzie to przerzuć, tym bardziej że modemy pci chyba raczej będą stanowić przyszłość.

4IceG, 2024-01-08:
@riccardoferti install package and check if it shows data ok. 

I don't know yet what to do with the SINR parameter, we read SNR instead. (for test SINR=SNR).

[luci-app-3ginfo-lite_1.0.62-20240108_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/13864594/luci-app-3ginfo-lite_1.0.62-20240108_all.zip)

riccardoferti, 2024-01-08:
> @riccardoferti install package and check if it shows data ok.
>
> I don't know yet what to do with the SINR parameter, we read SNR instead. (for test SINR=SNR). [...]

working also on pci mode?

4IceG, 2024-01-08:
>  @riccardoferti install package and check if it shows data ok.
> working also on pci mode?

Yes, it should work.

riccardoferti, 2024-01-08:
> > @riccardoferti install package and check if it shows data ok.
> > working also on pci mode?
> 
> Yes, it should work.

a little bit buggy but works

[image]

[image]

obsy, 2024-01-08:
Where?

4IceG, 2024-01-08:
@riccardoferti Copy extracted file to > /usr/share/3ginfo-lite/modem

I just fix scripts to make it faster.
[pci105be0b0.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/13865492/pci105be0b0.zip)

It should look better. I don't know yet what protocol the modem works on.

riccardoferti, 2024-01-08:
> @riccardoferti Copy extracted file to > /usr/share/3ginfo-lite/modem
> 
> I just fix scripts to make it faster.
> [pci105be0b0.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/13865492/pci105be0b0.zip)
> 
> It should look better. I don't know yet what protocol the modem works on.
> 

Should be MBIM

riccardoferti, 2024-01-08:
> Where?

It is slow to read and often it doesn't find modem

riccardoferti, 2024-01-08:
[image]

All seems good maybe need only to check this:

<html>
<body>

5G NSA \| B3 (1800 MHz) + B28 (700 MHz) / B7 (2600 MHz) / B1 (2100 MHz) / n78 (3500 MHz)
--

<br class="Apple-interchange-newline">
</body>
</html>

I advise to set only + or / symbol

4IceG, 2024-01-08:
> > Where?
> 
> It is slow to read and often it doesn't find modem
> 

Maybe other package is using modem at the same time?

4IceG, 2024-01-08:
> [image]
> 
> All seems good maybe need only to check this:
> 
> <html>
> <body>
> 
> 5G NSA \| B3 (1800 MHz) + B28 (700 MHz) / B7 (2600 MHz) / B1 (2100 MHz) / n78 (3500 MHz)
> --
> 
> 
> <br class="Apple-interchange-newline">
> </body>
> </html>
> 
> I advise to set only + or / symbol
> 
> 

I'll check it tomorrow.

riccardoferti, 2024-01-08:
> > > Where?
> > 
> > It is slow to read and often it doesn't find modem
> > 
> 
> Maybe other package is using modem at the same time?
> 
> 

Should be Modem Manager

4IceG, 2024-01-09:
> > Maybe other package is using modem at the same time?
> 
> Should be Modem Manager

So we have problem because MM and 3ginfo-lite cannot work together.

riccardoferti, 2024-01-10:
> > > Maybe other package is using modem at the same time?
> > 
> > Should be Modem Manager
> 
> So we have problem because MM and 3ginfo-lite cannot work together.

Oh ok, but T99w175 can't work without MM

frauhottelmann, 2024-03-14:
> > > > Maybe other package is using modem at the same time?
> > > 
> > > 
> > > Should be Modem Manager
> > 
> > 
> > So we have problem because MM and 3ginfo-lite cannot work together.
> 
> Oh ok, but T99w175 can't work without MM

In what device are you running this card with OpenWrt?

sp2003, 2024-04-19:
Can the package also be installed for USB2 version QMI protocol?

```
# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.01
S:  Manufacturer=Linux 6.1.82 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1e2d ProdID=00b7 Rev= 4.14
S:  Manufacturer=Cinterion
S:  Product=Cinterion PID 0x00B7 USB Mobile Broadband
S:  SerialNumber=7a7435d9
C:* #Ifs= 4 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan
E:  Ad=81(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
E:  Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=83(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=85(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=30 Driver=option
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.01
S:  Manufacturer=Linux 6.1.82 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms
```

obsy, 2024-04-20:
It does not matter. Package requires a serial interface to talk to the modem.

sp2003, 2024-04-20:
I installed the package **luci-app-3ginfo-lite_1.0.62-20240108_all.ipk**, I replaced **pci105be0b0** with **pci105be0b0_last_fix**, then copied and renamed **1e2d00b7** in the folder **/usr/share/3ginfo-lite/modem/usb**, and this is the result. Some information on bands and ISPs is missing, and it is a bit slow in retrieving information. Is it possible to modify it to improve it?
[image]

4IceG, 2024-04-20:
Hi @sp2003 ,
Replace file as in this thread https://github.com/4IceG/luci-app-3ginfo-lite/issues/71#issuecomment-2066809912 to fix the missing operator error.

The command result is needed to diagnose remaining errors.
sh -x /usr/share/3ginfo-lite/3ginfo.sh

sp2003, 2024-04-20:
it's long report.
I'll attach a file to you
[3ginfo.txt](https://github.com/4IceG/luci-app-3ginfo-lite/files/15047119/3ginfo.txt)

4IceG, 2024-04-20:
> it's long report.
> I'll attach a file to you
> [3ginfo.txt](https://github.com/4IceG/luci-app-3ginfo-lite/files/15047119/3ginfo.txt)
> 

@sp2003 install the latest available version of package and then replace the files and check how it works.

sp2003, 2024-04-20:
done, but we made the situation worse
[image]

4IceG, 2024-04-20:
@sp2003 Install this version (contains current fixes) [luci-app-3ginfo-lite_1.0.66-20240420_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/15047593/luci-app-3ginfo-lite_1.0.66-20240420_all.zip).

If there are errors, show the command results:
`sh -x /usr/share/3ginfo-lite/3ginfo.sh`

sp2003, 2024-04-20:
it' perfect!

Thank you!!!
[image]

4IceG, 2024-04-20:
@sp2003 Show me the result of the command:
```
sms_tool -d /dev/ttyUSB0 at "AT^TEMP"
sms_tool -d /dev/ttyUSB0 at "AT^TEMP?"
```

sp2003, 2024-04-20:
of corse!

```
root@X-WRT:~# sms_tool -d /dev/ttyUSB0 at "AT^TEMP"
AT^TEMP
root@X-WRT:~# sms_tool -d /dev/ttyUSB0 at "AT^TEMP?"
AT^TEMP?
PA: 28C
Skin Sensor: 27C
TSENS: 32C
```
[3ginfo.txt](https://github.com/4IceG/luci-app-3ginfo-lite/files/15048207/3ginfo.txt)

sp2003, 2024-04-20:
For information, this modem can change communication protocol from QMI to MBIM with an AT command.
In this case, the modem becomes

```
  | T: Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 2 Spd=480 MxCh= 0 |  
  | D: Ver= 2.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs= 1 |  
  | P: Vendor=1e2d ProdID=00b3 Rev= 4.14 |  
  | S: Manufacturer=Cinterion |  
  | S: Product=Cinterion PID 0x00B3 USB Mobile Broadband |  
  | S: SerialNumber=f47fbe2c |  
  | C:* #Ifs= 6 Cfg#= 1 Atr=80 MxPwr=500mA |  
  | A: FirstIf#= 0 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00 |  
  | I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim |  
  | E: Ad=81(I) Atr=03(Int.) MxPS= 64 Ivl=32ms |  
  | I: If#= 1 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim |  
  | I:* If#= 1 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim |  
  | E: Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms |  
  | E: Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms |  
  | I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option |  
  | E: Ad=83(I) Atr=03(Int.) MxPS= 10 Ivl=32ms |  
  | E: Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms |  
  | E: Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms |  
  | I:* If#= 3 Alt= 0 #EPs= 1 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none) |  
  | E: Ad=84(I) Atr=03(Int.) MxPS= 64 Ivl=32ms |  
  | I:* If#= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option |  
  | E: Ad=86(I) Atr=03(Int.) MxPS= 10 Ivl=32ms |  
  | E: Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms |  
  | E: Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms |  
  | I:* If#= 5 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=30 Driver=option |  
  | E: Ad=87(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms |  
  | E: Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
```

so if you want you can also add this mode.

4IceG, 2024-04-20:
@sp2003 Uninstall package and install this version [luci-app-3ginfo-lite_1.0.66-20240420_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/15048369/luci-app-3ginfo-lite_1.0.66-20240420_all.zip).

P.S.
I also added mbim support.

sp2003, 2024-04-20:
in MBIM on ROOter VM installation it does not work
[image]

[9 more replies]