4IceG/luci-app-3ginfo-lite #56: Allow displaying both LTE and 5G in technology(mode)?

bnjdg, 2024-01-01:
I am using RG500Q-AE on a Telstra/Arcadyan AW1000.

The 3ginfo is ok but is it possible to show both LTE and 5G connected in the 3ginfo results? 

`root@OpenWrt:/usr/share/3ginfo-lite# ./3ginfo.sh json
uci: Entry not found
sh: invalid number '0x
-10'
{
"connt":"1d, 05:07:06",
"conntx":"158.1 MiB",
"connrx":"11.7 GiB",
"modem":"Quectel RG500Q-EA",
"mtemp":"50 &deg;C",
"firmware":"RG500QEAAAR11A06M4G",
"cport":"/dev/ttyUSB2",
"protocol":"mbim",
"csq":"27",
"signal":"87",
"operator_name":"SUN",
"operator_mcc":"515",
"operator_mnc":"03",
"mode":"LTE | B1 (2100 MHz)",
"registration":"1",
"simslot":"1",
"imei":"869119XXXXXXXX",
"imsi":"51503XXXXXXXXXXX",
"iccid":"89630XXXXXXXXXXXXXX",
"lac_dec":"19107",
"lac_hex":"4AA3",
"tac_dec":"19107",
"tac_hex":"4AA3",
"tac_h":"",
"tac_d":"",
"cid_dec":"53844326",
"cid_hex":"3359966",
"pci":"1",
"earfcn":"150",
"pband":"B1 (2100 MHz) @20 MHz",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-89",
"rsrq":"-10",
"rssi":"-58",
"sinr":"14.0"
}
`

`at+qnwinfo;+qcainfo;+qeng="servingcell";+qspn
+QNWINFO: "FDD LTE","51503","LTE BAND 1",150`
**`+QNWINFO: "TDD NR5G","51503","NR5G BAND 41",528030`**
`+QCAINFO: "PCC",150,100,"LTE BAND 1",1,1,-90,-10,-57,13
+QENG: "servingcell","NOCONN"
+QENG: "LTE","FDD",515,03,3359966,1,150,1,5,5,4AA3,-90,-11,-60,16,13,120,-`
**`+QENG:"NR5G-NSA",515,03,832,-78,19,-11,528030,41,8,1`**
`+QSPN: "SUN","SUN","SUN",0,"51503"
`

[image]

4IceG, 2024-01-01:
Hi @bnjdg ,
3ginfo should and does show correctly. To me it looks like a bug in the modem firmware.

You are logged into the 5G network, but the `at+qcainfo` command will not show this fact. If possible, check a different version of the modem firmware.

bnjdg, 2024-01-01:
> Hi @bnjdg , 3ginfo should and does show correctly. To me it looks like a bug in the modem firmware.
> 
> You are logged into the 5G network, but the `at+qcainfo` command will not show this fact. If possible, check a different version of the modem firmware.

That is already the latest firmware available for the modem, I’ll see if I can ask for lower or older versions of the firmware from quectel support.

thanks.

4IceG, 2024-01-01:
> That is already the latest firmware available for the modem

The latest fw is not always the best, they often fix something causing errors somewhere else.

bnjdg, 2024-01-01:
I’ll close this for now since it does seem a firmware issue. Probably will be fixed once I get other firmwares for my device. Thanks.

XT-Martinez, 2024-01-01:
I have a different formware from @bnjdg and I also cannot see 5G information
[image]

4IceG, 2024-01-01:
Hi @XT-Martinez ,
Install the latest version of the package.

Then show result of the command:
`sh -x /usr/share/3ginfo-lite/3ginfo.sh`

bnjdg, 2024-01-06:
I dont know if it was your recent changes or the newer firmware.

But it now properly displays 5G-NSA! thank you very much. 
[image]

XT-Martinez, 2024-01-06:
This also worked for me as well after upgrading to the same firmware. Thanks for this awesome package!

4IceG, 2024-01-06:
Hi @bnjdg , @XT-Martinez 

I'm glad that the package works. 
I also recommend testing other packages:
https://github.com/4IceG/luci-app-modemband
https://github.com/4IceG/luci-app-sms-tool-js
https://github.com/4IceG/luci-app-lite-watchdog

closed 2024-01-06 by 4IceG