4IceG/luci-app-3ginfo-lite #57: JSON error on information page

robreyes, 2024-01-09:
Hi, I have been trying to have the package work on my Quectel RG500Q-EA, I can see that RG502Q is supported and your other packages seem to work smoothly with the modem, but whenever I visit the information page I am greeted with this error 
[image]

3ginfo.sh also seems to work fine
`+ cat
{
"connt":"0d, 03:32:31",
"conntx":"156.4 KiB",
"connrx":"89.4 KiB",
"modem":"Quectel RG500Q-EA",
"mtemp":"50 &deg;C",
"firmware":"RG500QEAAAR13A01M4G",
"cport":"/dev/ttyUSB2",
"protocol":"mbim",
"csq":"",
"signal":"3",
"operator_name":"515 66",
"operator_mcc":"515",
"operator_mnc":"66",
"mode":"5G-SA | n78 (3500 MHz)",
"registration":"1",
"simslot":"1",
"imei":"xxxxxxxxxxxxxxxxxxx",
"imsi":"xxxxxxxxxxxxxxxxxxx",
"iccid":"xxxxxxxxxxxxxxxxxxxx",
"lac_dec":"0",
"lac_hex":"0",
"tac_dec":"xxxxxxxxxx",
"tac_hex":"xxxxxxxxxxx",
"tac_h":"",
"tac_d":"",
"cid_dec":"xxxxxxxxxxx",
"cid_hex":"xxxxxxxxxxx",
",ci":"538
"earfcn":"620640",
"pband":"n78 (3500 MHz) @100 MHz",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-111",
"rsrq":"-12",
"rssi":"-111",
"sinr":"14"
}
+ exit 0`

The page seems to work fine on 4G only settings, but whenever I am on 5G it doesn't work for me.

4IceG, 2024-01-09:
Hi @robreyes ,
You have error because 3ginfo-lite does not support results for 5G SA (so far we only had results for 5G NSA).

Show result of the command `sh -x /usr/share/3ginfo-lite/3ginfo.sh`

robreyes, 2024-01-09:
Here 

```
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
uci: Entry not found
+ DEVICE=/dev/ttyUSB2
+ '[' -z /dev/ttyUSB2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"515 66 DITO",11

+COPS: 0,2,"51566",11

+CREG: 2,0

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=wanb
+ SUB=@
+ '[[' wanb '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wanb
+ '[' -z wanb ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wanb
+ grep '"up": true'
+ NETUP='       "up": true,'
+ '[' -n '      "up": true,' ]
+ uci -q -P /var/state/ get network.wanb.connect_time
+ CT=
+ '[' -z ]
+ ifstatus wanb
+ xargs
+ awk '-F[:,]' '/uptime/ {print $2}'
+ CT=14821
+ '[' '!' -z 14821 ]
+ expr 14821 / 60 / 60 / 24
+ D=0
+ expr 14821 / 60 / 60 '%' 24
+ H=4
+ expr 14821 / 60 '%' 60
+ M=7
+ expr 14821 '%' 60
+ S=1
+ printf '%dd, %02d:%02d:%02d' 0 4 7 1
+ CONN_TIME='0d, 04:07:01'
+ ifstatus wanb+ awk '-F"' '/l3_device/ {print $4}'

+ IFACE=wwan0
+ '[' -n wwan0 ]
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='105.9 KiB'
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='172.9 KiB'
+ echo '
+CPIN: READY

+CSQ: 99,99

[... 643 lines cut ...]
"tac_d":"",
"cid_dec":"xxxxxxxx",
"cid_hex":"xxxxxxxxx",
",ci":"538
"earfcn":"620640",
"pband":"n78 (3500 MHz) @100 MHz",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-112",
"rsrq":"-12",
"rssi":"-112",
"sinr":"13"
}
+ exit 0
```

4IceG, 2024-01-09:
@robreyes Copy extracted file [2c7c0800.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/13877377/2c7c0800.zip) to > /usr/share/3ginfo-lite/modem and check if it shows data.

robreyes, 2024-01-09:
Alrighty, will try then. Thanks :)

robreyes, 2024-01-10:
Hi again, still getting the same result after uploading the file.

4IceG, 2024-01-10:
Hi @robreyes , 
Did you clear the browser cache after copying extracted file? 

If there is still an error, show the command result again:
`sh -x /usr/share/3ginfo-lite/3ginfo.sh`

robreyes, 2024-01-10:
Hi @4IceG 

Yep, tried on incognito and other devices as well. Will try again later after work and send the command results when I'm back home. Thanks

robreyes, 2024-01-10:
Hi @4IceG 

I noticed that the error is now on different lines: 

[image]

here's the output of 3ginfo.sh

```
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB2
+ '[' -z /dev/ttyUSB2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?
+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"515 66 DITO",11

+COPS: 0,2,"51566",11

+CREG: 2,0

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=wanb
+ SUB=@
+ '[[' wanb '==' '*@*' ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wanb
+ '[' -z wanb ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wanb
+ grep '"up": true'
+ NETUP='       "up": true,'
+ '[' -n '      "up": true,' ]
+ uci -q -P /var/state/ get network.wanb.connect_time
+ CT=
+ '[' -z ]
+ ifstatus wanb
+ awk '-F[:,]' '/uptime/ {print $2}'
+ xargs
+ CT=46409
+ '[' '!' -z 46409 ]
+ expr 46409 / 60 / 60 / 24
+ D=0
+ expr 46409 / 60 / 60 '%' 24
+ H=12
+ expr 46409 / 60 '%' 60
+ M=53
+ expr 46409 '%' 60
+ S=29
+ printf '%dd, %02d:%02d:%02d' 0 12 53 29
+ CONN_TIME='0d, 12:53:29'
+ ifstatus wanb
+ awk '-F"' '/l3_device/ {print $4}'
+ IFACE=wwan0
+ '[' -n wwan0 ]
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='369.7 KiB'
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='376.6 KiB'
[... 648 lines cut ...]
"tac_d":"",
"cid_dec":"4295434240",
"cid_hex":"100072000",
",ci":"538
"earfcn":"620640",
"pband":"n78 (3500 MHz) @100 MHz",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-109",
"rsrq":"-12",
"rssi":"-71",
"sinr":"14"
}
+ exit 0
```

4IceG, 2024-01-10:
@robreyes I check your data by testing it myself and I don't have any errors. I have no idea where the problem is.
I don't know, maybe you have some unusual packages configuration.

I compiled a new package, but I don't know if it will eliminate error 🤔 .
[luci-app-3ginfo-lite_1.0.62-20240110_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/13892927/luci-app-3ginfo-lite_1.0.62-20240110_all.zip)

robreyes, 2024-01-11:
Hi @4IceG 

Sadly, even with the ipk package it still shows the error. I'll deep further regarding the issue, and see where it is. 

Still thanks for looking into it. Cheers!

robreyes, 2024-01-11:
Oh yeah, just to add, I have tested the package with the vanilla openwrt with just the 3ginfo, sms, and modem band, the other packages seems to work fine.

4IceG, 2024-01-11:
@robreyes If my package doesn't work, you can try another one, e.g. [luci-app-modeminfo](https://github.com/koshev-msk/modemfeed/tree/master/luci/applications/luci-app-modeminfo).

robreyes, 2024-01-12:
Hi @4IceG 

Reopening this ticket since you might find this interesting, the JSON error was caused by a line break in the output of 3ginfo.sh when opening the dashboard:

[image]

Removing the PCI values worked:

[image]

4IceG, 2024-01-12:
@robreyes I know where the script returned error, but I couldn't repeat it myself. This is what your data looks like running on my router.

[image]

robreyes, 2024-01-19:
Hi @4IceG 

Sorry for the late response, was busy with work. I have tested it out and worked great! PCI and signal properties are now displayed as well :)

[image]

closed 2024-01-19 by robreyes