4IceG/luci-app-3ginfo-lite #58: No Protocol Output on MF289F

SuperKali, 2024-01-25:
Hello,

Thank you so much for your magnificent package. I wanted to report an issue: essentially, my ZTE MF289F, in the modem information, displays "-" for the protocol instead of QMI, and the LAC is 0 instead.

[image]
[image]

```json
{
"connt":"0d, 00:16:22",
"conntx":"13.3 MiB",
"connrx":"23.9 MiB",
"modem":"ZTE MF289F",
"mtemp":"",
"firmware":"BD_VDFDEMF289F1V1.0.0B05",
"cport":"/dev/ttyUSB1",
"protocol":"",
"csq":"22",
"signal":"70",
"operator_name":"CoopVoce",
"operator_mcc":"222",
"operator_mnc":"01",
"location":"Italy",
"mode":"LTE-A | B3 (1800 MHz) + B32 (1500 MHz) / B7 (2600 MHz) / B20 (800 MHz)",
"registration":"1",
"simslot":"",
"imei":"REMOVED",
"imsi":"REMOVED",
"iccid":"REMOVED",
"lac_dec":"0",
"lac_hex":"0",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"34C0",
"tac_d":"13504",
"cid_dec":"53294889",
"cid_hex":"32D3729",
"pci":"410",
"earfcn":"1350",
"pband":"B3 (1800 MHz) @20.0 MHz",
"s1band":"B32 (1500 MHz) @20.0 MHz",
"s1pci":"140",
"s1earfcn":"10020",
"s2band":"B7 (2600 MHz) @15.0 MHz",
"s2pci":"211",
"s2earfcn":"3175",
"s3band":"B20 (800 MHz) @10.0 MHz",
"s3pci":"418",
"s3earfcn":"6300",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-105",
"rsrq":"-15",
"rssi":"-69",
"sinr":"4.2"
}

```

Is it possible to fix this?

Thank you.

4IceG, 2024-01-25:
Hi @SuperKali ,
I do not guarantee that all data will be presented. Modem does not always return data and not all at commands are known (of course, it is better to show "-" than "0").

Ok, let's try to improve this a bit. Show command output:
```
sms_tool -d /dev/ttyUSB1 at "at+zcellinfo?"
cat /sys/kernel/debug/usb/devices
```

SuperKali, 2024-01-25:
Hello @4IceG,

Here is the output of the commands:
```bash
root@MF289:~# sms_tool -d /dev/ttyUSB1 at "at+zcellinfo?"
at+zcellinfo?
+ZCELLINFO: 53294889, 410, LTE B3, 1350

root@MF289F:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.137 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.0.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.15
S:  Manufacturer=Linux 5.15.137 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.0.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=19d2 ProdID=1432 Rev=58.11
S:  Manufacturer=ZTE,Incorporated
S:  Product=ZTE Mobile Broadband
S:  SerialNumber=P685M135MZTED000000
C:* #Ifs= 5 Cfg#= 1 Atr=80 MxPwr=896mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=option
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=option
E:  Ad=83(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=option
E:  Ad=85(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan
E:  Ad=86(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
E:  Ad=8e(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=0f(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=42 Prot=01 Driver=(none)
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=87(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.137 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 0
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.15
S:  Manufacturer=Linux 5.15.137 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=(none)
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

```

4IceG, 2024-01-26:
Hi,
@SuperKali Check if this version looks better :)

[luci-app-3ginfo-lite_1.0.63-20240126_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/14067855/luci-app-3ginfo-lite_1.0.63-20240126_all.zip)

SuperKali, 2024-01-26:
Hi @4IceG,

Now it's:

[image]

4IceG, 2024-01-26:
Yes, I can hide it if there is no data and if it looks bad now.

SuperKali, 2024-01-26:
This is amazing! I will wait a your version for testing it

4IceG, 2024-01-27:
Hi,
@SuperKali Check this version :)

[luci-app-3ginfo-lite_1.0.63-20240127_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/14072513/luci-app-3ginfo-lite_1.0.63-20240127_all.zip)

SuperKali, 2024-01-27:
Hi @4IceG,

Now is good:

[image]

closed 2024-01-27 by 4IceG