4IceG/luci-app-3ginfo-lite #61: add support to SimCom SIM7906 lte module

aivavia, 2024-03-02:
Hello there, mr Rafal. I beg you add full support of sim7906 lte module, because the signal meters contains only CSQ.
here are the response of AT commands, that shows full statistic of LTE signal quality:

```
AT+CPSI?
+CPSI: LTE,Online,250-02,0x0EF8,22060,58,EUTRAN-BAND3,1725,3,3,-108,-889,-614,19

OK
AT+CMGRMI=4
+CMGRMI: Main_Info,4,1,1,0,2,52709,0,2,11

+CMGRMI: Serving_Cell,1725,250,02,3832,2,22060,3,3,3,19,58,-107,-890,-625,0

+CMGRMI: CA_Scell,0,0,0,0,0,0,0,0,0,0,0,0

+CMGRMI: LTE_Intra,1,1725,58,1

+CMGRMI: LTE_Intra_Cell1,58,-107,-890,-625,0

+CMGRMI: LTE_Inter,0,Freq1,0,0,0,0,0,Freq2,0,0,0,0,0

+CMGRMI: GSM_Info,0,Freq_Group1,0,0,0,0,0,Freq_Group2,0,0,0,0,0

+CMGRMI: WCDMA_Info,0,Freq1,0,0,0,0,0,Freq2,0,0,0,0,0

+CMGRMI: CDMA1x_Info,0,Freq1,0,0,0,Freq2,0,0,0

+CMGRMI: CDMAprpd_Info,0,Freq1,0,0,0,Freq2,0,0,0

+CMGRMI: CDrx_Cfg,1,2,80,8,40,21,1,20,1

+CMGRMI: Cqi_Cfg,1,1,4,0,1,1,14,58,0,0,1,161,0,0,0,0

+CMGRMI: Ant_Cfg,4,0x000000000000003F,0,0

+CMGRMI: Idle_Drx_Cfg,640,2,135

OK
```

4IceG, 2024-03-02:
Hi @aivavia 

Check if this version [luci-app-3ginfo-lite_1.0.65-20240302_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/14469198/luci-app-3ginfo-lite_1.0.65-20240302_all.zip) of the package shows more data.

aivavia, 2024-03-03:
Sorry to say, but nope. Nothing of needed info is displayed. Only CSQ. No RSRP, RSSI, SINR meters at all :(
[image]

4IceG, 2024-03-03:
OK, I added all modem variants to the package. Now we need to see why the script is not reading the data correctly.

@aivavia Show result of the command `sh -x /usr/share/3ginfo-lite/3ginfo.sh`

aivavia, 2024-03-04:
Here you are, please.
[3ginfo.txt](https://github.com/4IceG/luci-app-3ginfo-lite/files/14477625/3ginfo.txt)

4IceG, 2024-03-04:
Ok thanks,

@aivavia Check if this version [luci-app-3ginfo-lite_1.0.65-20240304_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/14480079/luci-app-3ginfo-lite_1.0.65-20240304_all.zip) of package shows more data.

aivavia, 2024-03-05:
oh no, smth is totally broken now 
[image]

4IceG, 2024-03-05:
Ok, we are looking again.

@aivavia Show result of the command `sh -x /usr/share/3ginfo-lite/3ginfo.sh`

aivavia, 2024-03-05:
[3ginfo_new.txt](https://github.com/4IceG/luci-app-3ginfo-lite/files/14490646/3ginfo_new.txt)
here you are

aivavia, 2024-03-06:
You are the BEST! all meters are working now flawessly!
[image]

closed 2024-03-08 by aivavia