4IceG/luci-app-3ginfo-lite #65: Add RSRP\RSRQ\SINR of SCC cell for Quectel Module

stich86, 2024-04-07:
Hi @4IceG,

do you think it will be possible to add RSRP\RSRQ\SINR also for secondary cell?

the `AT+QCAINFO` reports it for SCC bands:

```
+QCAINFO: "PCC",1850,100,"LTE BAND 3",1,130,-91,-11,-62,13
+QCAINFO: "SCC",3025,75,"LTE BAND 7",1,16,-104,-8,-87,0,0,-,-
+QCAINFO: "SCC",525,75,"LTE BAND 1",1,225,-93,-8,-77,15,0,-,-
+QCAINFO: "SCC",643296,10,"NR5G BAND 78",622
```

It will be very useful to create something like i've ported to ZTE CPE:

[image]

So you can also decide which bands are better or not.

Let me know :)

4IceG, 2024-04-08:
Hi @stich86 ,
I have a problem with this 🤔. I am already receiving information from users that 3ginfo-lite shows something that their LTE modems do not have. Another thing is that new modems return much more data and now I get questions why the package shows so little...

I don't want to expand the package too much because it will lose its simplicity. I have to wonder if it's possible to keep the appearance for the primary band and somehow extend the data to the aggregation bands.

stich86, 2024-04-08:
> Hi @stich86 , I have a problem with this 🤔. I am already receiving information from users that 3ginfo-lite shows something that their LTE modems do not have. Another thing is that new modems return [...]

i'll try to add it in the SCC string, just to have a reference

4IceG, 2024-04-08:
Now I focused on the transfer package, I had no plans to modify 3ginfo-lite, but I will think about it.

I don't want to create a package that will scroll like a long pdf, and I've seen such modifications (it makes me sad). It's a waste of my work when someone then does something like that, like in a competition, who can read more from the modem and put it on one page.

closed 2024-04-09 by stich86