4IceG/luci-app-3ginfo-lite #66: Issues with Argon Theme

stich86, 2024-04-08:
Hi @4IceG,

i've instellad on my BPI-R3-Mini the Argon Theme, but 3ginfo-lite has some graphical issue like "SIM card" icon not alligned and the text "More information about the 3ginfo on the [eko.one.pl forum](https://eko.one.pl/?p=openwrt-3ginfo)." that goes out the div when page is resized, here is some example?

how to play with the DIV to align "SIM card" icon and avoid overlap of info text?

[image]

Enlarge window:

[image]

Resize window (small horizontal lenght):

[image]

thanks for any tips :)

4IceG, 2024-04-08:
Hi @stich86 , now I know why I don't use third-party themes 😄.

From what I can see, the material theme (Argon works on this theme) has a problem with the ifacebox object, I don't know if it is not described there or if it is there but defined differently (I did not investigate). On the luci-theme-openwrt-2020 theme it looks acceptable.

stich86, 2024-04-08:
I understand.. 
I'm tweaking a little bit playing with padding, I can get good alignment 

Just figure out where to add these mods. In which JS are this page generated? 3gdetail.js?

Thx

4IceG, 2024-04-08:
I know where I set it [Link](https://github.com/4IceG/luci-app-3ginfo-lite/blob/46864030221069da10759f074bc2131754bc4991/luci-app-3ginfo-lite/htdocs/luci-static/resources/view/modem/3gdetail.js#L925), the question is whether setting it permanently will not ruin the appearance of other themes.

In this case, it would be best to add a css file with the missing description for ifacebox [Link](https://github.com/openwrt/luci/blob/b5763397c4addf349bf8cd665ca8897f583b56b1/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css#L357). I'm just not sure that it will work with every theme.

Have you tried adding the missing css sections to the Argon theme?