4IceG/luci-app-3ginfo-lite #68: An error when opening information

prusa-dev, 2024-04-12:
An error when opening information.

I found that mccmnc.dat line ending is windows rather then unix.

[image]

### Console log:
[image]

### JSON:
```json
{
"connt":"0d, 00:25:25",
"conntx":"248.9 KiB",
"connrx":"1.2 MiB",
"modem":"",
"mtemp":"",
"firmware":"",
"cport":"/dev/ttyUSB4",
"protocol":"",
"csq":"10",
"signal":"32",
"operator_name":"MTS
",
"operator_mcc":"",
"operator_mnc":"",
"location":"-",
"mode":"LTE",
"registration":"6",
"simslot":"",
"imei":"",
"imsi":"",
"iccid":"",
"lac_dec":"12345",
"lac_hex":"1234",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"1234567",
"cid_hex":"123456",
"pci":"",
"earfcn":"",
"pband":"",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":""
}
```

4IceG, 2024-04-13:
Hi, 
This modem is weird. It doesn't show much. You can try replacing the 3ginfo.sh file `/usr/share/3ginfo-lite`.

[3ginfo.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/14968128/3ginfo.zip)

prusa-dev, 2024-04-14:
Hi
This is FM350 modem.
[I'm trying to update 3ginfo for fm350](https://github.com/prusa-dev/luci-app-3ginfo-lite/commit/3de4471437b90791b05cffeb2bc8c0ee980d3a98). Last release 1.0.65 has broken operator name.

here is logs:
[yours_3ginfo_sh.log](https://github.com/4IceG/luci-app-3ginfo-lite/files/14970743/yours_3ginfo_sh.log)
[my_modified_3ginfo_sh.log](https://github.com/4IceG/luci-app-3ginfo-lite/files/14970744/my_modified_3ginfo_sh.log)

4IceG, 2024-04-14:
Hi,
@prusa-dev Check these at commands to see if you get a modem response:
```
AT+COPS=3,2; +COPS?
AT+CEREG=2; +CEREG?
AT+CGREG=2; +CGREG?; +CSQ
AT+GTCCINFO?; +GTCAINFO?
AT+GTSENRDTEMP=1
AT+GTPKGVER?; +CGMI?; +CGMM?; +CGMR?; +CIMI?; +ICCID; +CGSN?
```

prusa-dev, 2024-04-15:
This is not problem of FM350, 
I think problem here https://github.com/4IceG/luci-app-3ginfo-lite/blob/main/luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/3ginfo.sh#L306

Maybe there should be $COPS_NUM variable instead of $COPS

prusa-dev, 2024-04-15:
> Hi, This modem is weird. It doesn't show much. You can try replacing the 3ginfo.sh file `/usr/share/3ginfo-lite`.
> 
> [3ginfo.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/14968128/3ginfo.zip)

@4IceG 
Adding xargs solve problem with line ending

4IceG, 2024-04-15:
@prusa-dev Is the operator name shown after recent script corrections?

I don't know if I found it right, but I think these at commands `+GTCURCAR?` or `AT+COPN` or `AT+COPN=?` it also allows to show the operator.

or read from a mccmnc.dat file.

prusa-dev, 2024-04-16:
No operator name is not shown.

Here is my COPS response:
```
sms_tool -d /dev/ttyUSB4 at 'AT+COPS=3,0;+COPS?;+COPS=3,2;+COPS?'

+COPS:0,0,"MTS RUS",7

+COPS:0,2,"25001",7

```

Please look at the 3ginfo.sh log:
```
+ awk '-F["]' '/^\+COPS:\s?.,2/ {print $2}'
+ COPS_NUM=25001
+ '[' -n 25001 ]
+ COPS_MCC=250
+ COPS_MNC=01
+ '[' -z  ]
+ awk '-F["]' '/^\+COPS:\s?.,0/ {print $2}'
+ echo '
+CPIN: READY

+CSQ: 9, 99

+COPS:0,0,"MTS RUS",7

+COPS:0,2,"25001",7

+CREG: 2,6,"1234","000123456",7

'K
+ COPS='MTS RUS'
+ '[' -z 'MTS RUS' ]
+ sed ':s;s/\(\<\S*\>\)\(.*\)\<\1\>/\1\2/g;ts'
+ echo MTS RUS
+ COPZ='MTS RUS'
+ awk '{for(i=1;i<=NF;i++){ $i=toupper(substr($i,1,1)) substr($i,2) }}1'
+ echo MTS RUS
+ COPS='MTS RUS'
+ + sms_tooltr -d '\r'
 -d /dev/ttyUSB4 at 'AT+COPS=3,2;+COPS?'
+ cut -d '"' -f2
+ sed -n 2p
+ isp=25001
+ isp_num='250 01'
+ isp_numws=25001
+ '[[' 'MTS RUS' '=' '250 01' '||' 'MTS RUS' '=' 25001 ]]
+ '[[' -n 'MTS RUS' ]]
+ awk '-F[;]' /^MTS 'RUS;/ {print $3}' /usr/share/3ginfo-lite/mccmnc.dat
awk: cmd. line:1: Unexpected end of string
+ COPS=
+ awk '-F[;]' '/^;/ {print $2}' /usr/share/3ginfo-lite/mccmnc.dat
+ LOC=
+ LOCATIONFILE=/tmp/location
+ '[' -e /tmp/location ]
+ touch /tmp/location
+ cat /tmp/location
+ LOC=Russia
+ '[' -n Russia ]
+ cat /tmp/location
+ LOC=Russia
```
As you can see `COPS` variable is `MTS RUS`, and later script compare `isp` with `COPS` that contain non numeric operator name.
I don't understand, why there `isp`  `isp_num` `isp_numws` code block?

4IceG, 2024-04-16:
> As you can see `COPS` variable is `MTS RUS`, and later script compare `isp` with `COPS` that contain non numeric operator name.
> I don't understand, why there `isp` `isp_num` `isp_numws` code block?

This means that part of the script with `isp` `isp_num` `isp_numws` needs to be corrected. This part of the code was supposed to read the operator name and its location in case COPS appeared as the MCCMNC value. This needs to be changed so that only numerical values are checked.

prusa-dev, 2024-04-18:
Hi @4IceG 
The error is gone, and the operator's name is displayed correctly.

[image]

Thank you!

4IceG, 2024-04-18:
Hi @prusa-dev , I'll take a look at the script tonight. I'll write later if it's ok.

closed 2024-04-21 by 4IceG