4IceG/luci-app-3ginfo-lite #71: No Operator Name in LT4220 (QMI)

bulzed, 2024-04-19:
There is no operator name for my modem HP LT4220 (QMI mode). I already ran 3ginfo.sh manually still no result in "operator_name" line, it's only prints "operator_mcc", "operator_mnc", "location" etc. So there is no operator name in luci neither, eventhough it shows "Location" instead of my provider under signal bar in luci.
Kindly fix it, please . . .

4IceG, 2024-04-19:
Hi @bulzed ,
Replace the 3ginfo.sh file (\usr\share\3ginfo-lite) and check if the error disappears.
[3ginfo.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/15042493/3ginfo.zip)

bulzed, 2024-04-20:
thanks a lot, dev...
let me try . . .

bulzed, 2024-04-20:
thanks dev, now everything works normally . . . 
🥂

closed 2024-04-20 by bulzed

Inlifeuser, 2024-04-21:
I have the same problem with Huawei E3372 in NCM mode, and replacing 3ginfo.sh helped me too.