4IceG/luci-app-3ginfo-lite #75: Support for DLINK DWR-921 C1

thomasst1012, 2024-05-06:
Hello, 

can you add support for https://openwrt.org/toh/d-link/dwr-921 Rev C1? 
That would be very nice :)

Thank you,
Greetings,
Thomas

P.S. I've already donated something.

4IceG, 2024-05-06:
Hi,
Thanks for your support 👍🏻, a nice gesture for a good start the week :).

We can try to add modem support, the question is whether the modem will be willing to cooperate.

Show command result (let's see what kind of modem it is)
`cat sys/kernel/debug/usb/devices`

Have you tried to communicate with modem using the at or qmi commands?

Greetings,
Rafał

4IceG, 2024-05-06:
> does that help you?

The more we know, the greater chance of adding support. Currently 3ginfo-lite doesn't read data from qmi, so I would have to add this. 

Question is if the modem has any ttyUSB port for communication. @thomasst1012 Could you show also

`ls -al /dev/*`

thomasst1012, 2024-05-06:
@4IceG - Here is the Output:

root@atnr-dlink:~# ls -al /dev/*
crw-------    1 root     root      180, 176 May  5 15:37 /dev/cdc-wdm0
crw-------    1 root     root        5,   1 Jan  1  1970 /dev/console
lrwxrwxrwx    1 root     root            13 Jan  1  1970 /dev/fd -> /proc/self/fd
crw-rw-rw-    1 root     root        1,   7 Jan  1  1970 /dev/full
crw-------    1 root     root      254,   0 Jan  1  1970 /dev/gpiochip0
crw-------    1 root     root      254,   1 Jan  1  1970 /dev/gpiochip1
crw-------    1 root     root      254,   2 Jan  1  1970 /dev/gpiochip2
crw-------    1 root     root      254,   3 Jan  1  1970 /dev/gpiochip3
crw-------    1 root     root       10, 183 Jan  1  1970 /dev/hwrng
crw-------    1 root     root        1,  11 Jan  1  1970 /dev/kmsg
srw-rw-rw-    1 root     root             0 May  5 15:37 /dev/log
crw-------    1 root     root       90,   0 Jan  1  1970 /dev/mtd0
crw-------    1 root     root       90,   1 Jan  1  1970 /dev/mtd0ro
crw-------    1 root     root       90,   2 Jan  1  1970 /dev/mtd1
crw-------    1 root     root       90,   3 Jan  1  1970 /dev/mtd1ro
crw-------    1 root     root       90,   4 Jan  1  1970 /dev/mtd2
crw-------    1 root     root       90,   5 Jan  1  1970 /dev/mtd2ro
crw-------    1 root     root       90,   6 Jan  1  1970 /dev/mtd3
crw-------    1 root     root       90,   7 Jan  1  1970 /dev/mtd3ro
crw-------    1 root     root       90,   8 Jan  1  1970 /dev/mtd4
crw-------    1 root     root       90,   9 Jan  1  1970 /dev/mtd4ro
crw-------    1 root     root       90,  10 Jan  1  1970 /dev/mtd5
crw-------    1 root     root       90,  11 Jan  1  1970 /dev/mtd5ro
brw-------    1 root     root       31,   0 Jan  1  1970 /dev/mtdblock0
brw-------    1 root     root       31,   1 Jan  1  1970 /dev/mtdblock1
brw-------    1 root     root       31,   2 Jan  1  1970 /dev/mtdblock2
brw-------    1 root     root       31,   3 Jan  1  1970 /dev/mtdblock3
brw-------    1 root     root       31,   4 Jan  1  1970 /dev/mtdblock4
brw-------    1 root     root       31,   5 Jan  1  1970 /dev/mtdblock5
crw-rw-rw-    1 root     root        1,   3 Jan  1  1970 /dev/null
crw-------    1 root     root        1,   4 Jan  1  1970 /dev/port
crw-------    1 root     root      108,   0 May  5 15:37 /dev/ppp
crw-rw-rw-    1 root     root        5,   2 May  6 09:55 /dev/ptmx
crw-rw-rw-    1 root     root        1,   8 Jan  1  1970 /dev/random
lrwxrwxrwx    1 root     root             8 Jan  1  1970 /dev/shm -> /tmp/shm
lrwxrwxrwx    1 root     root            15 Jan  1  1970 /dev/stderr -> /proc/self/fd/2
lrwxrwxrwx    1 root     root            15 Jan  1  1970 /dev/stdin -> /proc/self/fd/0
lrwxrwxrwx    1 root     root            15 Jan  1  1970 /dev/stdout -> /proc/self/fd/1
crw-rw-rw-    1 root     root        5,   0 Jan  1  1970 /dev/tty
crw-rw----    1 root     dialout     4,  64 Jan  1  1970 /dev/ttyS0
crw-rw----    1 root     dialout     4,  65 Jan  1  1970 /dev/ttyS1
crw-rw----    1 root     dialout     4,  74 Jan  1  1970 /dev/ttyS10
crw-rw----    1 root     dialout     4,  75 Jan  1  1970 /dev/ttyS11
crw-rw----    1 root     dialout     4,  76 Jan  1  1970 /dev/ttyS12
crw-rw----    1 root     dialout     4,  77 Jan  1  1970 /dev/ttyS13
crw-rw----    1 root     dialout     4,  78 Jan  1  1970 /dev/ttyS14
crw-rw----    1 root     dialout     4,  79 Jan  1  1970 /dev/ttyS15
crw-rw----    1 root     dialout     4,  66 Jan  1  1970 /dev/ttyS2
crw-rw----    1 root     dialout     4,  67 Jan  1  1970 /dev/ttyS3
crw-rw----    1 root     dialout     4,  68 Jan  1  1970 /dev/ttyS4
crw-rw----    1 root     dialout     4,  69 Jan  1  1970 /dev/ttyS5
crw-rw----    1 root     dialout     4,  70 Jan  1  1970 /dev/ttyS6
crw-rw----    1 root     dialout     4,  71 Jan  1  1970 /dev/ttyS7
crw-rw----    1 root     dialout     4,  72 Jan  1  1970 /dev/ttyS8
crw-rw----    1 root     dialout     4,  73 Jan  1  1970 /dev/ttyS9
crw-rw----    1 root     dialout   188,   0 May  5 19:34 /dev/ttyUSB0
crw-rw----    1 root     dialout   188,   1 May  6 07:53 /dev/ttyUSB1
crw-rw----    1 root     dialout   188,   2 May  5 19:21 /dev/ttyUSB2
crw-rw----    1 root     dialout   188,   3 May  5 19:21 /dev/ttyUSB3
crw-rw-rw-    1 root     root        1,   9 Jan  1  1970 /dev/urandom
crw-------    1 root     root       10, 130 Jan  1  1970 /dev/watchdog
crw-------    1 root     root      253,   0 Jan  1  1970 /dev/watchdog0
crw-rw-rw-    1 root     root        1,   5 Jan  1  1970 /dev/zero

/dev/bus:
drwxr-xr-x    3 root     root            60 Jan  1  1970 .
drwxr-xr-x    4 root     root          1340 May  5 15:37 ..
drwxr-xr-x    4 root     root            80 Jan  1  1970 usb

/dev/pts:
drwxr-xr-x    2 root     root             0 Jan  1  1970 .
drwxr-xr-x    4 root     root          1340 May  5 15:37 ..
crw--w--w-    1 root     root      136,   0 May  6 09:55 0
c---------    1 root     root        5,   2 Jan  1  1970 ptmx

4IceG, 2024-05-06:
Let's check if any port responds to at command (install sms-tool if you don't have it and send commands individually)
```
sms_tool -d /dev/ttyUSB0 at "ATI"
sms_tool -d /dev/ttyUSB1 at "ATI"
sms_tool -d /dev/ttyUSB2 at "ATI"
sms_tool -d /dev/ttyUSB3 at "ATI"
```

thomasst1012, 2024-05-06:
root@atnr-dlink:~# sms_tool -d /dev/ttyUSB0 at "ATI"
No response from modem.
root@atnr-dlink:~# sms_tool -d /dev/ttyUSB1 at "ATI"
ATI
Manufacturer: BroadMobi
Model: BM806C
Revision: M1.2.0_E1.0.1_A1.1.8
IMEI: 358289080491316
+GCAP: +CGSM

root@atnr-dlink:~# sms_tool -d /dev/ttyUSB2 at "ATI"
ATI
Manufacturer: BroadMobi
Model: BM806C
Revision: M1.2.0_E1.0.1_A1.1.8
IMEI: 358289080491316
+GCAP: +CGSM

root@atnr-dlink:~# sms_tool -d /dev/ttyUSB3 at "ATI"
No response from modem.

4IceG, 2024-05-06:
Install the luci-app-3ginfo-lite package, we will see how much data it shows. 

Super modem communicates normally via /dev/ttyUSB2 or /dev/ttyUSB1.

thomasst1012, 2024-05-06:
[image]
[image]

4IceG, 2024-05-06:
I thought it would be worse. Now all that remains is to fill in the missing data.

@thomasst1012 Could you show also

```
sms_tool -d /dev/ttyUSB1 at "AT+BMTCELLINFO"
sms_tool -d /dev/ttyUSB1 at "AT+CGMI"
sms_tool -d /dev/ttyUSB1 at "AT+CGMM]"
sms_tool -d /dev/ttyUSB1 at "AT+CGMR"
```

4IceG, 2024-05-06:
OK thanks. @thomasst1012 In the evening I will fix missing data and send to you new version for testing.

thomasst1012, 2024-05-06:
Thank You @4IceG

4IceG, 2024-05-06:
Hi Thomas,
@thomasst1012 Install this version [luci-app-3ginfo-lite_1.0.70-20240506_all.zip](https://github.com/4IceG/luci-app-3ginfo-lite/files/15226681/luci-app-3ginfo-lite_1.0.70-20240506_all.zip) and check how much data it shows.

Greetings,
Rafał

thomasst1012, 2024-05-07:
Hei Rafał, 

how can i install a zip file in openwrt?

Thank you, 
Thomas

4IceG, 2024-05-07:
Hi, 
Installation package is inside, you need to unpack the archive.

thomasst1012, 2024-05-07:
Hei, 

looks better than before :) Data Transfer was visibile before but not now. Take a look:
[image]
[image]
[image]

4IceG, 2024-05-07:
OK, I changed these statistics. 
Thomas clear your browser cache.

thomasst1012, 2024-05-07:
After clearing the browser cache its now OK
[image]

4IceG, 2024-05-07:
So everything works. I also encourage to test my other packages :).

thomasst1012, 2024-05-07:
Thank you  very much Rafał

I have installed also the SMS tool. Anything more which i can use?

closed 2024-05-09 by 4IceG