4IceG/luci-app-3ginfo-lite #77: Undefined connection statistics in LuCI

Inlifeuser, 2024-05-11:
This only appears in last version 1.0.70 (after upgrade from 1.0.69). In previous version 1.0.69 (after downgrade) statistics are displayed correctly.
[image]

I also noticed that the format for displaying these statistics in JSON has changed, and it seems that changes to the web interface were missed.

1.0.70:
[image]

1.0.69:
[image]

4IceG, 2024-05-11:
Hi @Inlifeuser ,
Clear your browser cache.

Inlifeuser, 2024-05-11:
Before this I also tried to reboot OpenWrt, and it did not help. After clearing the browser cache, statistics are now displayed correctly. Thanks!

closed 2024-05-11 by Inlifeuser