4IceG/luci-app-3ginfo-lite #78: Cannot read properties of undefined (reading 'length')

QooGeek, 2024-05-12:
```
Model | Bananapi BPi-R3 Mini
Architecture | ARMv8 Processor rev 4
Target Platform | mediatek/filogic
Firmware Version | OpenWrt SNAPSHOT r26289-61e8728d86 / LuCI Master 24.120.25050~8f2dfa7
Kernel Version | 6.6.30
Local Time | 2024-05-12 16:37:54
Uptime | 2h 13m 21s
Load Average | 0.00, 0.00, 0.00
```

TypeError
Cannot read properties of undefined (reading 'length')

4IceG, 2024-05-14:
Hi @QooGeek ,
Show command result:
```
sh -x /usr/share/3ginfo-lite/3ginfo.sh
```

QooGeek, 2024-05-14:
[image]

4IceG, 2024-05-14:
@QooGeek What modem are you using? Scripts don't know it.

QooGeek, 2024-05-15:
my  5g module  fm350gl

4IceG, 2024-05-15:
@QooGeek Is modem set to USB (only this mode is added), without Modem Manager? what does the cat /sys/kernel/debug/usb/devices command show?

QooGeek, 2024-05-15:
```
root@R3mini-Router:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.06
S:  Manufacturer=Linux 6.6.30 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.06
S:  Manufacturer=Linux 6.6.30 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=11200000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms
root@R3mini-Router:~#
```

4IceG, 2024-05-15:
If this is the full result of the command, then as you can see, modem is not available as a USB device. You need to change the configuration to make modem visible to 3ginfo-lite.

prusa-dev, 2024-05-15:
@QooGeek 
Hi

What package are you using to connect the modem via USB?
Try this one - https://github.com/koshev-msk/modemfeed/tree/master/packages/net/fm350-modem

QooGeek, 2024-05-15:
@4IceG @prusa-dev    Thank you very much for your help. My fm350 is broken. I replaced it and it works.

closed 2024-05-16 by QooGeek