4IceG/luci-app-3ginfo-lite #80: the package don't work properly on web pages

yyllxxhh, 2024-05-23:
I followed the instructions in the README.md to compile an IPK installer, and I installed it on my system using the OPKG command. However, there's no MODEM option in the status bar of the web interface, even though the SIM card is working fine.

Details:
- OpenWrt Version: LEAN LEDE
- Linux Kernel Version: 4.9.84

Any ideas on what might be going wrong? Thanks!

4IceG, 2024-05-23:
Hi @yyllxxhh , if there is no modem menu, you may need:
- click somewhere else in the menu 
- logging out of the router 
- clearing the browser cache

obsy, 2024-05-24:
- use current version of luci...

yyllxxhh, 2024-05-31:
after updating the luci version, the modem menu is now working properly.

closed 2024-05-31 by yyllxxhh