4IceG/luci-app-3ginfo-lite #82: Fibocom L850-GL modem support

atdx, 2024-05-31:
Hi,
Can you add support for L850-GL modem? This is a lower version of L860-GL, but L850-GL supports mbim protocol.

[2cb70007.txt](https://github.com/user-attachments/files/15514269/2cb70007.txt)

This is a firmware file made by someone on discord, I don't know if this file is the result of renaming L860-GL or he edited it himself, but it works in old 3ginfo-lite, but in new 3ginfo-lite it doesn't show band number.

Can you improve it and add it to official support?

[image]
[image]

atdx, 2024-05-31:
This diagnostic resulut

cat /sys/kernel/debug/usb/devices
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=-2097, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.15
S:  Manufacturer=Linux 5.15.150 dwc_otg_hcd
S:  Product=DWC OTG Controller
S:  SerialNumber=3f980000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 5
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=0424 ProdID=9514 Rev= 2.00
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  2mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=01 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms
I:* If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0424 ProdID=ec00 Rev= 2.00
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  2mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=ff Driver=smsc95xx
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=83(I) Atr=03(Int.) MxPS=  16 Ivl=1ms

T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  4 Spd=480  MxCh= 0
D:  Ver= 2.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=2cb7 ProdID=0007 Rev= 3.33
S:  Manufacturer=Fibocom Wireless Inc.
S:  Product=L850-GL
S:  SerialNumber=004999010640000
C:* #Ifs= 8 Cfg#= 1 Atr=e0 MxPwr=100mA
A:  FirstIf#= 0 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00
A:  FirstIf#= 2 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00
A:  FirstIf#= 4 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00
A:  FirstIf#= 6 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00
I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:  If#= 1 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
I:* If#= 1 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 3 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 6 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 7 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=88(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=08(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms

atdx, 2024-05-31:
ls /dev

bus
cdc-wdm0
console
cpu_dma_latency
fb0
fd
full
gpiochip0
gpiochip1
gpiochip2
gpiomem
hwrng
input
kmsg
log
loop-control
loop0
loop1
loop2
loop3
loop4
loop5
loop6
loop7
mmcblk0
mmcblk0p1
mmcblk0p2
net
null
port
ppp
ptmx
pts
ram0
ram1
ram10
ram11
ram12
ram13
ram14
ram15
ram2
ram3
ram4
ram5
ram6
ram7
ram8
ram9
random
shm
snd
stderr
stdin
stdout
tty
tty0
tty1
tty10
tty11
tty12
tty13
tty14
tty15
tty16
tty17
tty18
tty19
tty2
tty20
tty21
tty22
tty23
tty24
tty25
tty26
tty27
tty28
tty29
tty3
tty30
tty31
tty32
tty33
tty34
tty35
tty36
tty37
tty38
tty39
tty4
tty40
tty41
tty42
tty43
tty44
tty45
tty46
tty47
tty48
tty49
tty5
tty50
tty51
tty52
tty53
tty54
tty55
tty56
tty57
tty58
tty59
tty6
tty60
tty61
tty62
tty63
tty7
tty8
tty9
ttyACM0
ttyACM1
ttyACM2
ttyAMA0
ttyS0
urandom
vc-mem
vchiq
vcio
vcs
vcs1
vcsa
vcsa1
vcsu
vcsu1
watchdog
watchdog0
zero
zram0

atdx, 2024-05-31:
sh -x /usr/share/3ginfo-lite/3ginfo.sh

{
"conn_time":"0d, 01:11:19",
"conn_time_sec":"4279",
"conn_time_since":"20240531202339",
"rx":"198.7 MiB",
"tx":"83.7 MiB",
"modem":"+CGMI: Fibocom L850 LTE Module",
"mtemp":"",
"firmware":"18500.5001.00.05.27.16_5001.05.001.033",
"cport":"/dev/ttyACM2",
"protocol":"",
"csq":"33",
"signal":"100",
"operator_name":"XL Axiata",
"operator_mcc":"510",
"operator_mnc":"11",
"location":"Indonesia",
"mode":"LTE-A | +",
"registration":"6",
"simslot":"",
"imei":"015184004775859",
"imsi":"510113883742876",
"iccid":"8962116638837428763F",
"lac_dec":"32008",
"lac_hex":"7D08",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"81935366",
"cid_hex":"4E23C06",
"pci":" 113",
"earfcn":"1325",
"pband":" @20 MHz",
"s1band":" @5  MHz",
"s1pci":"113",
"s1earfcn":"3774",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-78",
"rsrq":"-13",
"rssi":"-47",
"sinr":"6"
}
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
uci: Entry not found
+ DEVICE=/dev/ttyACM2
+ '[' -z /dev/ttyACM2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyACM2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='
+CPIN: READY

+CSQ: 18,4

+COPS: 0,0,"XL Axiata",7

+COPS: 0,2,"51011",7

[... 678 lines cut ...]
E:  Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 6 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 7 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=88(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=08(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms'
+ awk '-FVendor=8087 ProdID=07f9' '{print $2}'
+ echo T: 'Bus=01' 'Lev=00' 'Prnt=00' 'Port=00' 'Cnt=00' 'Dev#=' 1 'Spd=480' 'MxCh=' 1 B: 'Alloc=' 0/800 us '(' '0%),' '#Int=-2124,' '#Iso=' 0 D: 'Ver=' 2.00 'Cls=09(hub' ')' 'Sub=00' 'Prot=01' 'MxPS=64' '#Cfgs=' 1 P: 'Vendor=1d6b' 'ProdID=0002' 'Rev=' 5.15 S: 'Manufacturer=Linux' 5.15.150 dwc_otg_hcd S: 'Product=DWC' OTG Controller S: 'SerialNumber=3f980000.usb' 'C:*' '#Ifs=' 1 'Cfg#=' 1 'Atr=e0' 'MxPwr=' 0mA 'I:*' 'If#=' 0 'Alt=' 0 '#EPs=' 1 'Cls=09(hub' ')' 'Sub=00' 'Prot=00' 'Driver=hub' E: 'Ad=81(I)' 'Atr=03(Int.)' 'MxPS=' 4 'Ivl=256ms' T: 'Bus=01' 'Lev=01' 'Prnt=01' 'Port=00' 'Cnt=01' 'Dev#=' 2 'Spd=480' 'MxCh=' 5 D: 'Ver=' 2.00 'Cls=09(hub' ')' 'Sub=00' 'Prot=02' 'MxPS=64' '#Cfgs=' 1 P: 'Vendor=0424' 'ProdID=9514' 'Rev=' 2.00 'C:*' '#Ifs=' 1 'Cfg#=' 1 'Atr=e0' 'MxPwr=' 2mA I: 'If#=' 0 'Alt=' 0 '#EPs=' 1 'Cls=09(hub' ')' 'Sub=00' 'Prot=01' 'Driver=hub' E: 'Ad=81(I)' 'Atr=03(Int.)' 'MxPS=' 1 'Ivl=256ms' 'I:*' 'If#=' 0 'Alt=' 1 '#EPs=' 1 'Cls=09(hub' ')' 'Sub=00' 'Prot=02' 'Driver=hub' E: 'Ad=81(I)' 'Atr=03(Int.)' 'MxPS=' 1 'Ivl=256ms' T: 'Bus=01' 'Lev=02' 'Prnt=02' 'Port=00' 'Cnt=01' 'Dev#=' 3 'Spd=480' 'MxCh=' 0 D: 'Ver=' 2.00 'Cls=ff(vend.)' 'Sub=00' 'Prot=01' 'MxPS=64' '#Cfgs=' 1 P: 'Vendor=0424' 'ProdID=ec00' 'Rev=' 2.00 'C:*' '#Ifs=' 1 'Cfg#=' 1 'Atr=e0' 'MxPwr=' 2mA 'I:*' 'If#=' 0 'Alt=' 0 '#EPs=' 3 'Cls=ff(vend.)' 'Sub=00' 'Prot=ff' 'Driver=smsc95xx' E: 'Ad=81(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=02(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=83(I)' 'Atr=03(Int.)' 'MxPS=' 16 'Ivl=1ms' T: 'Bus=01' 'Lev=02' 'Prnt=02' 'Port=01' 'Cnt=02' 'Dev#=' 4 'Spd=480' 'MxCh=' 0 D: 'Ver=' 2.10 'Cls=ef(misc' ')' 'Sub=02' 'Prot=01' 'MxPS=64' '#Cfgs=' 1 P: 'Vendor=2cb7' 'ProdID=0007' 'Rev=' 3.33 S: 'Manufacturer=Fibocom' Wireless Inc. S: 'Product=L850-GL' S: 'SerialNumber=004999010640000' 'C:*' '#Ifs=' 8 'Cfg#=' 1 'Atr=e0' 'MxPwr=100mA' A: 'FirstIf#=' 0 'IfCount=' 2 'Cls=02(comm.)' 'Sub=0e' 'Prot=00' A: 'FirstIf#=' 2 'IfCount=' 2 'Cls=02(comm.)' 'Sub=02' 'Prot=00' A: 'FirstIf#=' 4 'IfCount=' 2 'Cls=02(comm.)' 'Sub=02' 'Prot=00' A: 'FirstIf#=' 6 'IfCount=' 2 'Cls=02(comm.)' 'Sub=02' 'Prot=00' 'I:*' 'If#=' 0 'Alt=' 0 '#EPs=' 1 'Cls=02(comm.)' 'Sub=0e' 'Prot=00' 'Driver=cdc_mbim' E: 'Ad=81(I)' 'Atr=03(Int.)' 'MxPS=' 64 'Ivl=1ms' I: 'If#=' 1 'Alt=' 0 '#EPs=' 0 'Cls=0a(data' ')' 'Sub=00' 'Prot=02' 'Driver=cdc_mbim' 'I:*' 'If#=' 1 'Alt=' 1 '#EPs=' 2 'Cls=0a(data' ')' 'Sub=00' 'Prot=02' 'Driver=cdc_mbim' E: 'Ad=82(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=02(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 2 'Alt=' 0 '#EPs=' 1 'Cls=02(comm.)' 'Sub=02' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=83(I)' 'Atr=03(Int.)' 'MxPS=' 64 'Ivl=1ms' 'I:*' 'If#=' 3 'Alt=' 0 '#EPs=' 2 'Cls=0a(data' ')' 'Sub=00' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=84(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=04(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 4 'Alt=' 0 '#EPs=' 1 'Cls=02(comm.)' 'Sub=02' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=85(I)' 'Atr=03(Int.)' 'MxPS=' 64 'Ivl=1ms' 'I:*' 'If#=' 5 'Alt=' 0 '#EPs=' 2 'Cls=0a(data' ')' 'Sub=00' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=86(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=06(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 6 'Alt=' 0 '#EPs=' 1 'Cls=02(comm.)' 'Sub=02' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=87(I)' 'Atr=03(Int.)' 'MxPS=' 64 'Ivl=1ms' 'I:*' 'If#=' 7 'Alt=' 0 '#EPs=' 2 'Cls=0a(data' ')' 'Sub=00' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=88(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=08(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms'
+ PVCAX=
+ echo 
+ cut -c-2453
+ PVCA=
+ awk '-FVendor=8087 ProdID=095a' '{print $2}'
[... cut ...] d=86(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=06(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 6 'Alt=' 0 '#EPs=' 1 'Cls=02(comm.)' 'Sub=02' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=87(I)' 'Atr=03(Int.)' 'MxPS=' 64 'Ivl=1ms' 'I:*' 'If#=' 7 'Alt=' 0 '#EPs=' 2 'Cls=0a(data' ')' 'Sub=00' 'Prot=00' 'Driver=cdc_acm' E: 'Ad=88(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=08(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms'
+ PVCBX=
+ echo 
+ cut -c-2453
+ PVCB=
+ '[' -n  ]
+ '[' -n  ]
+ cat
+ exit 0

4IceG, 2024-06-01:
Hi,
@atdx Unpack archive [2cb70007.zip](https://github.com/user-attachments/files/15520472/2cb70007.zip) and replace the file. File path `\usr\share\3ginfo-lite\modem\usb`.

Check if everything is OK now.

atdx, 2024-06-01:
Hi, 
@4IceG  not working

[image]

{
"conn_time":"0d, 00:06:44",
"conn_time_sec":"404",
"conn_time_since":"20240601134603",
"rx":"456.9 KiB",
"tx":"402.5 KiB",
"modem":"Fibocom L850",
"mtemp":"",
"firmware":"18500.5001.00.05.27.16_5001.05.001.033",
"cport":"/dev/ttyACM2",
"protocol":"mbim",
"csq":"17",
"signal":"100",
"operator_name":"XL Axiata",
"operator_mcc":"510",
"operator_mnc":"11",
"location":"Indonesia",
"mode":"LTE_A B3
 + B8 (900 MHz) / B3 (1800 MHz) / B3 (1800 MHz) / B8 (900 MHz)",
"registration":"6",
"simslot":"",
"imei":"015184004775859",
"imsi":"510113883742876",
"iccid":"8962116638837428763F",
"lac_dec":"32008",
"lac_hex":"7D08",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"32008",
"cid_dec":"81935364",
"cid_hex":"4E23C04",
"pci":"111",
"earfcn":"1325",
"pband":"B3
 @20 MHz",
"s1band":"B8 (900 MHz) @5 MHz",
"s1pci":"111",
"s1earfcn":"3774",
"s2band":"B3 (1800 MHz)",
"s2pci":"113",
"s2earfcn":"1325",
"s3band":"B3 (1800 MHz)",
"s3pci":"112",
"s3earfcn":"1325",
"s4band":"B8 (900 MHz)",
"s4pci":"113",
"s4earfcn":"3774",
"rsrp":"-80",
"rsrq":"-13",
"rssi":"-51",
"sinr":""
}
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
uci: Entry not found
+ DEVICE=/dev/ttyACM2
+ '[' -z /dev/ttyACM2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyACM2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='
+CPIN: READY

+CSQ: 17,5
[... 813 lines cut ...]
I:* If#= 2 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 3 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 6 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm
E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 7 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm
E:  Ad=88(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=08(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms'
+ awk '-FVendor=2cb7 ProdID=0007' '{print $2}'
+ echo 'T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1' 'B:  Alloc=  0/800 us ( 0%), #Int=-437, #Iso=  0' 'D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1' 'P:  Vendor=1d6b ProdID=0002 Rev= 5.15' 'S:  Manufacturer=Linux 5.15.150 dwc_otg_hcd' 'S:  Product=DWC OTG Controller' 'S:  SerialNumber=3f980000.usb' 'C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA' 'I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub' 'E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms' 'T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 5' 'D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1' 'P:  Vendor=0424 ProdID=9514 Rev= 2.00' 'C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  2mA' 'I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=01 Driver=hub' 'E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms' 'I:* If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub' 'E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms' 'T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480  MxCh= 0' 'D:  Ver= 2.00 Cls=ff(vend.) Sub=00 Prot=01 MxPS=64 #Cfgs=  1' 'P:  Vendor=0424 ProdID=ec00 Rev= 2.00' 'C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  2mA' 'I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=ff Driver=smsc95xx' 'E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'E:  Ad=83(I) Atr=03(Int.) MxPS=  16 Ivl=1ms' 'T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  4 Spd=480  MxCh= 0' 'D:  Ver= 2.10 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1' 'P:  Vendor=2cb7 ProdID=0007 Rev= 3.33' 'S:  Manufacturer=Fibocom Wireless Inc.' 'S:  Product=L850-GL' 'S:  SerialNumber=004999010640000' 'C:* #Ifs= 8 Cfg#= 1 Atr=e0 MxPwr=100mA' 'A:  FirstIf#= 0 IfCount= 2 Cls=02(comm.) Sub=0e Prot=00' 'A:  FirstIf#= 2 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00' 'A:  FirstIf#= 4 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00' 'A:  FirstIf#= 6 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00' 'I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=0e Prot=00 Driver=cdc_mbim' 'E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=1ms' 'I:  If#= 1 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim' 'I:* If#= 1 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=02 Driver=cdc_mbim' 'E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'I:* If#= 2 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm' 'E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=1ms' 'I:* If#= 3 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm' 'E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'I:* If#= 4 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm' 'E:  Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=1ms' 'I:* If#= 5 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm' 'E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'E:  Ad=06(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'I:* If#= 6 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm' 'E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=1ms' 'I:* If#= 7 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm' 'E:  Ad=88(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms' 'E:  Ad=08(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms'
[... cut ...] Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=1ms I:* If#= 5 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E:  Ad=06(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 6 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=cdc_acm E:  Ad=87(I) Atr=03(Int.) MxPS=  64 Ivl=1ms I:* If#= 7 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_acm E:  Ad=88(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E:  Ad=08(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms' ]
+ PROTO=mbim
+ cat
+ exit 0

4IceG, 2024-06-01:
Ok, @atdx thanks for test and info. I'll fix it later.

4IceG, 2024-06-01:
@atdx Unpack archive [2cb70007.zip](https://github.com/user-attachments/files/15521333/2cb70007.zip) and replace the file. File path `\usr\share\3ginfo-lite\modem\usb`.

atdx, 2024-06-01:
> @atdx Unpack archive [2cb70007.zip](https://github.com/user-attachments/files/15521333/2cb70007.zip) and replace the file. File path `\usr\share\3ginfo-lite\modem\usb`.

@4IceG  Thanks friend, the file works fine, the only remaining problem is probably the text "Only SMS".

[image]

4IceG, 2024-06-01:
@atdx TAC (hex) still requires a correction, it is not calculated.

What you indicated is result of the command:
+**CREG: 2,6**,"7D08","04E23C04",7

https://onomondo.com/blog/at-command-cgreg/

I don't know if it's a modem firmware bug or if the manufacturer interprets this data differently.

atdx, 2024-06-01:
@4IceG I didn't notice it, and I don't know what it does.

the command displays
"No response from modem".

[image]

4IceG, 2024-06-01:
@atdx You need to enter the command in such a way AT+CREG=2;+CREG? to get result I showed.

atdx, 2024-06-01:
@4IceG oh I see, so it's possible that the modem is the problem, I don't really mind it, this is enough.

[image]

atdx, 2024-06-01:
@4IceG This is the result in the old 3ginfo-lite, with the file I shared in the first post

[image]

root@At01Wrt:~# sh -x /usr/share/3ginfo-lite/3ginfo.sh
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyACM2
+ '[' -z /dev/ttyACM2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyACM2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='
+CPIN: READY

+CSQ: 16,4

+COPS: 0,0,"XL Axiata",7

+COPS: 0,2,"51011",7

+CREG: 2,6,"7D08","04E23C06",7

'K
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SECT=@wan
+ SUB=@
+ '[[' @wan '==' '*@*' ]]
+ sed s/@//
+ echo @wan
+ SEC=wan
+ '[' -z wan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ grep '"up": true'
+ ifstatus wan
+ NETUP='       "up": true,'
+ '[' -n '      "up": true,' ]
+ uci -q -P /var/state/ get network.wan.connect_time
+ CT=
+ '[' -z ]
+ ifstatus wan
+ awk '-F[:,]' '/uptime/ {print $2}'
+ xargs
+ CT=489
+ '[' '!' -z 489 ]
+ expr 489 / 60 / 60 / 24
+ D=0
+ expr 489 / 60 / 60 '%' 24
+ H=0
+ expr 489 / 60 '%' 60
+ M=8
+ expr 489 '%' 60
+ S=9
+ printf '%dd, %02d:%02d:%02d' 0 0 8 9
+ CONN_TIME='0d, 00:08:09'
+ ifstatus wan
+ awk '-F"' '/l3_device/ {print $4}'
+ IFACE=wwan0
+ '[' -n wwan0 ]
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='4.8 MiB'
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='897.9 KiB'
+ echo '
+CPIN: READY

[... 646 lines cut ...]
"tac_h":"",
"tac_d":"",
"cid_dec":"81935366",
"cid_hex":"4E23C06",
"pci":" 113",
"earfcn":"1325",
"pband":"B3 (1800 MHz) @20 MHz",
"s1band":"B8 (900 MHz) @5  MHz",
"s1pci":"113",
"s1earfcn":"3774",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-82",
"rsrq":"-10",
"rssi":"-51",
"sinr":"8"
}
+ exit 0

4IceG, 2024-06-02:
Hi,
@atdx The read data is identical, before and now. The only difference is in the 3ginfo-lite package.
Older version had a simplified description of data, the current version has a more detailed description, so there is a message corresponding to the read data.

We can "cheat in scripts" and force another option.

At the moment it looks strange and illogical, I assume that everything works 100% OK and modem still shows network registration in a limited way.

4IceG, 2024-06-02:
@atdx This is script after changes, it should be OK.
Standard procedure, unpack archive [2cb70007.zip](https://github.com/user-attachments/files/15526878/2cb70007.zip) and replace the file. File path `\usr\share\3ginfo-lite\modem\usb`.

atdx, 2024-06-02:
This is perfect and worthy to be made official, thank you @4IceG  for working hard on this modem 😊👍

closed 2024-06-23 by 4IceG