4IceG/luci-app-3ginfo-lite #83: T99W175 with 6.0.0.6 FW 3ginfo issue

aivavia, 2024-06-09:
Hello mr. Rafal, I beg you to fix the meters.
[image]
The case is in FW 6.0.0.6, I think your 3ginfo can't parse meters correctly.
`root@OpenWrt:~# sms_tool -d /dev/ttyUSB2 at 'ati'
ati
Manufacturer: Qualcomm
Model: T99W175
Revision: T99W175.F0.6.0.0.6.GC.004  1  [Feb 03 2023 10:00:00]
SVN: 01
IMEI: 12345678912345
+GCAP: +CGSM
MPN: 01

root@OpenWrt:~# sms_tool -d /dev/ttyUSB2 at 'at^debug?'
at^debug?
RAT:LTE
mcc:250,mnc:02
lte_cell_id:22060
lte_tac:3832
lte_tx_pwr:-2.0dBm
lte_ant_rsrp:rx_diversity:1 (-103.2dBm,NA,NA,NA)
pcell: lte_band:3 lte_band_width:15.0MHz
channel:1725 pci:58
lte_rsrp:-103.2dBm,rsrq:-10.9dB
lte_rssi:-73.4dBm,lte_snr:13.6dB
`
the AT output and full 3ginfo optput
[T99W175_3ginfo.txt](https://github.com/user-attachments/files/15751414/T99W175_3ginfo.txt)

4IceG, 2024-06-09:
Hi @aivavia 
It seems that modem on this fw version has a changed ID, so scripts cannot read the data. I need to add a modem (with this ID).

aivavia, 2024-06-09:
> It seems that modem on this fw version has a changed ID
No matter what customer is chosen,  VID and PID are mostly 05c6:90d5 or 05c6:9025
Changing customer changes the "Manufacturer" string in ATI and nothing more.
if it will help you.

4IceG, 2024-06-09:
@aivavia Unpack archive [T99W175.zip](https://github.com/user-attachments/files/15751691/T99W175.zip) and copy 2 files to `\usr\share\3ginfo-lite\modem\usb` plus clear your browser cache.

aivavia, 2024-06-09:
Excellent! Thank u very much

closed 2024-06-10 by aivavia