4IceG/luci-app-3ginfo-lite #84: Problem with adding repository.

AnnaShmykova, 2024-06-15:
Hello. 

I can't get the packages from your repo:

~# wget https://github.com/4IceG/Modem-extras/raw/main/myrepo/IceG-repo.pub -O /tmp/IceG-repo.pub
Downloading 'https://github.com/4IceG/Modem-extras/raw/main/myrepo/IceG-repo.pub'
Connecting to 140.82.121.4:443
Redirected to /4IceG/Modem-extras/main/myrepo/IceG-repo.pub on raw.githubusercontent.com
SSL error: SSL - Bad input parameters to function
Connection error: Connection failed

If I change adress to ~# wget https://github.com/4IceG/Modem-extras/**tree**/main/myrepo/IceG-repo.pub -O /tmp/IceG-repo.pub
Downloading 'https://github.com/4IceG/Modem-extras/tree/main/myrepo/IceG-repo.pub'
Connecting to 140.82.121.4:443
Redirected to /4IceG/Modem-extras/blob/main/myrepo/IceG-repo.pub on github.com
Writing to '/tmp/IceG-repo.pub'

Download completed (285829 bytes)

~# opkg update
Downloading https://github.com/4IceG/Modem-extras/raw/main/myrepo/Packages.gz
SSL error: SSL - Bad input parameters to function
*** Failed to download the package list from https://github.com/4IceG/Modem-extras/raw/main/myrepo/Packages.gz

Is the problem on my side?

4IceG, 2024-06-15:
Hi,
@AnnaShmykova I'm building repository as I always do. There were no problems before.
The bug is known, you can check this topic https://github.com/4IceG/Modem-extras/issues/2 for more information.

AnnaShmykova, 2024-06-15:
Thank you very much. 
"opkg install wget-ssl" - solves the problem.