4IceG/luci-app-3ginfo-lite #85: Parse error on 5G-SA Network

xhikarishii, 2024-07-23:
Hi @4IceG 

There seems to be a parsing error on some networks here in 🇵🇭 specifically for 5G-SA network, please see my log below
[3ginfo-log.txt](https://github.com/user-attachments/files/16350799/3ginfo-log.txt)

This doesn't seem to be an issue before so could be related to the recent changes made.

f1vefour, 2024-07-24:
I was just about to create an issue, I have the same problem with 5G-SA on T-Mobile in the US.

I'm attaching two logs:

[5G-SA_disabled.txt](https://github.com/user-attachments/files/16356875/5G-SA_disabled.txt) is in 4G with 3ginfo-lite working.

[5G-SA_enabled.txt](https://github.com/user-attachments/files/16356876/5G-SA_enabled.txt) is in 5G--SA  with 3ginfo-lite not working.

4IceG, 2024-07-24:
Hi @xhikarishii @f1vefour,
I'm rarely at my computer these days. 

I'm not promising anything, but I'll try to look at it this weekend.

xhikarishii, 2024-07-24:
No rush :)

KFOzzx, 2024-07-24:
I think my issue is same to you guys 🤔

https://github.com/4IceG/luci-app-3ginfo-lite/issues/86

f1vefour, 2024-07-24:
> I think my issue is same to you guys 🤔
> 
> #86
> 
> samet time ... subject on OpenWrt https://forum.openwrt.org/t/fibocom-quectel-fm190-5g-4g-modem-signals/204468/8

I don't get a parse error on the page like that unfortunately, it simply doesn't display the data when in 5G-SA

obsy, 2024-07-25:
@f1vefour show me
```
sms_tool =D -d /dev/ttyUSB2 at "at+c5greg=2;+c5greg?"
```

f1vefour, 2024-07-25:
@obsy 

> root@OpenWrt:~# sms_tool -d /dev/ttyUSB2 at "at+c5greg=2;+c5greg?"
> at+c5greg=2;+c5greg?
> +C5GREG: 2,1,"B3FE00","109157003",11,1,"01"

obsy, 2024-07-25:
@f1vefour , @xhikarishii 

OK, parser problem. Try this:
```
sed -i "s/{print \$4;exit}')/{print \$4;exit}' | xargs)/g" /usr/share/3ginfo-lite/3ginfo.sh
```

And test again in luci.

f1vefour, 2024-07-25:
@obsy Pure genius, it's working!

KFOzzx, 2024-07-25:
@obsy 
https://github.com/4IceG/luci-app-3ginfo-lite/issues/86

please take look to my case

xhikarishii, 2024-07-29:
> @f1vefour , @xhikarishii
> 
> OK, parser problem. Try this:
> 
> ```
> sed -i "s/{print \$4;exit}')/{print \$4;exit}' | xargs)/g" /usr/share/3ginfo-lite/3ginfo.sh
> ```
> 
> And test again in luci.

Worked for me as well 💯

f1vefour, 2024-08-26:
@4IceG will you be adding a new build?

I run a custom build of openwrt and every time I update I have to patch 3ginfo.sh using the sed snippet above.

4IceG, 2024-08-26:
> @4IceG will you be adding a new build?

Hi @f1vefour 
The new packages/repository compilation will be tomorrow.

f1vefour, 2024-08-26:
> > @4IceG will you be adding a new build?
> 
> Hi @f1vefour The new packages/repository compilation will be tomorrow.

Great to hear, thank you.

f1vefour, 2024-08-28:
This particular issue is resolved, it can be closed now.

closed 2024-08-28 by 4IceG