4IceG/luci-app-3ginfo-lite #86: Fibocom modem -FM190 5G/4G modem Signals

KFOzzx, 2024-07-24:
Hi Dear Sir IceG

**When network connected to 4G only the signal appeared .**
[image]

**After I switched the network to 5G NSA (4G+5G) .... The error came again .**
[image]

**Debug from 3ginfo** 
`{
"conn_time":"0d, 01:12:18",
"conn_time_sec":"4338",
"conn_time_since":"20240724144007",
"rx":"2.9 GiB",
"tx":"301.3 MiB",
"modem":"Fibocom  FM190W-GL",
"mtemp":"",
"firmware":"99103.1000.00.01.06.10",
"cport":"/dev/ttyUSB1",
"protocol":"qmi",
"csq":"21",
"signal":"67",
"operator_name":"stc ksa",
"operator_mcc":"420",
"operator_mnc":"01",
"location":"Saudi Arabia",
"mode":"5G NSA |",
"registration":"1",
"simslot":"",
"imei":"351064272748410",
"imsi":"420012416985880",
"iccid":"8996601240169858806F",
"lac_dec":"65535",
"lac_hex":"FFFF",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"4285",
"tac_d":"17029",
"cid_dec":"7155981",
"cid_hex":"6D310D",
"pci":"78
188",
"earfcn":"4FF
9D0E0",
"pband":"",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"-71",
"sinr":""
}
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB1
+ '[' -z /dev/ttyUSB1 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB1 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='
[... 515 lines cut ...]
+ readlink -f /sys/class/net/br-wan
+ readlink -f /sys/class/net/eth0
+ readlink -f /sys/class/net/eth1
+ readlink -f /sys/class/net/eth2
+ readlink -f /sys/class/net/lan1
+ readlink -f /sys/class/net/lan2
+ readlink -f /sys/class/net/lan3
+ readlink -f /sys/class/net/lo
+ readlink -f /sys/class/net/wan
+ readlink -f /sys/class/net/wwan0
+ NETIF=/sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1/2-1.1:1.4/net/wwan0
+ readlink -f /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1/2-1.1:1.4/net/wwan0/../../driver
+ basename /sys/bus/usb/drivers/qmi_wwan
+ NETDRV=qmi_wwan
+ PROTO=qmi
+ cat
+ exit 0`

Modem Infor 
`"Manufacturer": "QCOM",
"Product": "SDXPINN-IDP _SN:EA9B91F5",
"Vendor": "2cb7",
"ProdID": "0104",
"Bus_speed": "5000",
"Serial_Number": "ea9b91f5"`

obsy, 2024-07-25:
Your case require new parser for FM190. Response for AT command is not fully compatible with FM150 or FM350, so it require new script.

KFOzzx, 2024-07-26:
> Your case require new parser for FM190. Response for AT command is not fully compatible with FM150 or FM350, so it require new script.

I added AT commands file for FM190W and FM350 ..... they are similar (80%) 

[English - Fibocom_AT Commands_FM190.pdf](https://github.com/user-attachments/files/16387412/English.-.Fibocom_AT.Commands_FM190.pdf)
[Fibocom_FM350_AT Commands User Manual_V2.10 (1).pdf](https://github.com/user-attachments/files/16387413/Fibocom_FM350_AT.Commands.User.Manual_V2.10.1.pdf)
[Orginal Chiness Fibocom_AT Commands_ FM190W.pdf](https://github.com/user-attachments/files/16387414/Orginal.Chiness.Fibocom_AT.Commands_.FM190W.pdf)

obsy, 2024-07-26:
Similar. Not the same. And that's the difference from a scripting perspective.

KFOzzx, 2024-07-26:
> Similar. Not the same. And that's the difference from a scripting perspective.

Is there anyway to get it done ? I mean to be added

obsy, 2024-07-26:
Be patient. The script needs to be written.

KFOzzx, 2024-07-28:
Dear @4IceG @obsy 
if there is any file or documents you need me to share it let me know ... 
Really I appreciate for your support

obsy, 2024-09-12:
Update script: https://github.com/obsy/packages/commit/6e46db60d611026129f7019b655702d700dd471d
@4IceG: please adapt it to your project

For 5G SA in FM150 RSSI is not displayed, only RSRP

4IceG, 2024-09-15:
@ghost I don't know if it will work, but you can check it out.

Unpack [test.zip](https://github.com/user-attachments/files/17006859/test.zip) archive and copy files to `usr/share/3ginfo-lite/modem/usb`.

KFOq, 2024-09-23:
Hi , I will try to test it this week . thank you for efforts .

KFOq, 2024-11-14:
> @ghost I don't know if it will work, but you can check it out.
> 
> Unpack [test.zip](https://github.com/user-attachments/files/17006859/test.zip) archive and copy files to `usr/share/3ginfo-lite/modem/usb`.

Hi Boss I tested your package it was not working as I know ... maybe there is something wrong with my setup or the package doesn't match with modem FM190W-GL