4IceG/luci-app-3ginfo-lite #87: Feature Request: Add Cellmapper as Search

frauhottelmann, 2024-07-25:
Hi @4IceG,

Can you add Cellmapper.net as a search for cell ID? It should be straight forward:

https://www.cellmapper.net/enbid?cellid=XXXXXX

Cellmpapper then gives you a results page with an eNB ID that you can use on their map to search your tower.

I would have made a pull request, but I am not sure where I have to add the URL and options for a new search site.

Thanks

4IceG, 2024-07-25:
Hi @frauhottelmann ,
No problem, I'll add this page, I just need to know the rules for creating a link.

frauhottelmann, 2024-07-25:
It's just https://www.cellmapper.net/enbid?cellid= and add the CellID at the end.

4IceG, 2024-07-25:
I understand, I just had to check if XXXXX is a decimal or hex value :).

MiG-41, 2024-07-26:
So in my case: https://www.cellmapper.net/enbid?cellid=3658497
[image]

This informations are not so valuable (and they could be also given in 3ginfo-litle directly i belive).

frauhottelmann, 2024-07-26:
Well, if we can geht the eNB ID from 3G Info directly then we can search an the map directly. But I am not sure if that is possible from a URL. I'll check.

frauhottelmann, 2024-07-26:
Unfortunately you can't directly search on the map. You have to choose your provider and then search for the eNB ID, but the search is not accessible via URL 😔