4IceG/luci-app-3ginfo-lite #95: 从调制解调器读取数据时出现问题

jmyh88, 2024-09-04:
检测语言

中文（简体）

英语

德语

英语

俄语

中文（简体）
你好，编译好固件，进入3ginfo-lite，提示There is a problem reading data from the modem.

Please check:
1. Modem availability in the system.
2. The correct installation of the SIM card in the modem.
3. Port for communication with the modem.
网络诊断

USB调试信息
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.06
S:  Manufacturer=Linux 6.6.48 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=2cb7 ProdID=0104 Rev= 4.14
S:  Manufacturer=Fibocom
S:  Product=Fibocom Modem_SN:A43BDFCC
S:  SerialNumber=a43bdfcc
C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=30 Driver=option
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=83(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=85(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option
E:  Ad=87(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan
E:  Ad=88(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
E:  Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.06
S:  Manufacturer=Linux 6.6.48 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

检查 3ginfo 脚本读取的数据
{
"conn_time":"0d, 00:09:29",
"conn_time_sec":"569",
"conn_time_since":"20240905071819",
"rx":"420.7 KiB",
"tx":"41.7",
"modem":"Fibocom  FM150-AE",
"mtemp":"41.7 &deg;C",
"firmware":"89602.1000.00.03.07.17",
"cport":"/dev/ttyUSB1",
"protocol":"qmi",
"csq":"",
"signal":"",
"operator_name":"CHINA MOBILE",
"operator_mcc":"460",
"operator_mnc":"00",
"location":"China",
"mode":"-",
"registration":"0",
"simslot":"",
"imei":"865697040222645",
"imsi":"460074529238756",
"iccid":"89860000192308753811",
"lac_dec":"0",
"lac_hex":"0",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"0",
"cid_hex":"0",
"pci":"",
"earfcn":"",
"pband":"",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":""
}
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB1
+ '[' -z /dev/ttyUSB1 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB1 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?

+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"CHINA MOBILE",11

+COPS: 0,2,"46000",11

+CREG: 2,0

OK
'
+ CONFIG=modemdefine
+ uci show modemdefine
+ grep -o '@modemdefine\[[0-9]*\]\.modem'
+ wc -l
+ xargs
+ MODEMZ=0
+ '[[' 0 -gt 1 ]]
+ '[[' 0 -eq 0 ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wan
+ '[[' 0 -eq 1 ]]
+ '[' -z wan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wan
+ grep '"up": true'
+ NETUP='	"up": true,'
+ '[' -n '	"up": true,' ]
+ uci -q -P /var/state/ get network.wan.connect_time
+ CT=
+ '[' -z ]
+ ifstatus wan
+ awk '-F[:,]' '/uptime/ {print $2}'
+ xargs
+ CT=569
+ '[' '!' -z 569 ]
+ expr 569 / 60 / 60 / 24
+ D=0
+ expr 569 / 60 / 60 '%' 24
+ H=0
+ expr 569 / 60 '%' 60
+ M=9
+ expr 569 '%' 60
+ S=29
+ printf '%dd, %02d:%02d:%02d' 0 0 9 29
+ CONN_TIME='0d, 00:09:29'
+ date '+%s'
+ date '+%Y%m%d%H%M%S' -d @1725491899
+ CONN_TIME_SINCE=20240905071819
+ ifstatus wan
+ awk '-F"' '/l3_device/ {print $4}'
+ IFACE=wwan0
+ '[' -n wwan0 ]
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='420.7 KiB'
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='2.0 MiB'
+ echo 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?

+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"CHINA MOBILE",11

+COPS: 0,2,"46000",11

+CREG: 2,0

OK
'
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=99
+ '[' x99 '==' x ]
+ '[' 99 -ge 0 -a 99 -le 31 ]
+ CSQ=
+ C
auto_awesome
源语言： 英语
5,000 / 5,000
Hello, after compiling the firmware, I enter 3ginfo-lite and it says There is a problem reading data from the modem.

Please check:
1. Modem availability in the system.
2. The correct installation of the SIM card in the modem.
3. Port for communication with the modem.
Network diagnostics

USB debugging information
T: Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#= 1 Spd=480 MxCh= 1
B: ​​Alloc= 0/800 us ( 0%), #Int= 0, #Iso= 0
D: Ver= 2.00 Cls=09(hub ) Sub=00 Prot=01 MxPS=64 #Cfgs= 1
P: Vendor=1d6b ProdID=0002 Rev= 6.06
S: Manufacturer=Linux 6.6.48 xhci-hcd
S: Product=xHCI Host Controller S: SerialNumber=xhci-hcd.1.auto C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr= 0mA I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub ) Sub=00 Prot=00 Driver=hub E: Ad=81(I) Atr=03(Int.) MxPS= 4 Ivl =256ms T: Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 2 Spd=480 MxCh= 0 D: Ver= 2.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs= 1 P: Vendor=2cb7 ProdID=0104 Rev= 4.14 S: Manufacturer=Fibocom S: Product=Fibocom Modem_SN:A43BDFCC S: SerialNumber=a43bdfcc C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=500mA I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=30 Driver=option E: Ad=81(I ) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 1 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option E: Ad=83(I) Atr=03(Int.) MxPS= 10 Ivl=32ms E: Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option E: Ad=8 5(I) Atr=03(Int.) MxPS= 10 Ivl=32ms E: Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option E: Ad=87(I) Atr=03(Int.) MxPS= 10 Ivl=32ms E: Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If #= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan E: Ad=88(I) Atr=03(Int.) MxPS= 8 Ivl=32ms E: Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms T: Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#= 1 Spd=5000 MxCh= 1 B: Alloc= 0/800 us (0%), #Int= 0, #Iso= 0 D: Ver= 3.00 Cls=09(hub ) Sub=00 Prot=03 MxPS= 9 #Cfgs= 1 P: Vendor=1d6b ProdID=0003 Rev= 6.06 S: Manufacturer=Linux 6.6.48 xhci-hcd S: Product=xHCI Host Controller S: SerialNumber=xhci-hcd.1.auto C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr= 0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub ) Sub=00 Prot=00 Driver=hub
E: Ad=81(I) Atr=03(Int.) MxPS= 4 Ivl=256ms

jmyh88, 2024-09-04:
Check data read by the 3ginfo scripts

{
"conn_time":"0d, 00:15:18",
"conn_time_sec":"918",
"conn_time_since":"20240905071819",
"rx":"595.9 KiB",
"tx":"40.2",
"modem":"Fibocom  FM150-AE",
"mtemp":"40.2 &deg;C",
"firmware":"89602.1000.00.03.07.17",
"cport":"/dev/ttyUSB1",
"protocol":"qmi",
"csq":"",
"signal":"",
"operator_name":"CHINA MOBILE",
"operator_mcc":"460",
"operator_mnc":"00",
"location":"China",
"mode":"-",
"registration":"0",
"simslot":"",
"imei":"865697040222645",
"imsi":"460074529238756",
"iccid":"89860000192308753811",
"lac_dec":"0",
"lac_hex":"0",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"0",
"cid_hex":"0",
"pci":"",
"earfcn":"",
"pband":"",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"",
"rsrq":"",
"rssi":"",
"sinr":""
}
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB1
+ '[' -z /dev/ttyUSB1 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB1 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?

+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"CHINA MOBILE",11

+COPS: 0,2,"46000",11

[... 406 lines cut ...]
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan
E:  Ad=88(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
E:  Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.06
S:  Manufacturer=Linux 6.6.48 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms'
+ awk -F 'Vendor=2cb7 ProdID=0104' '{print $2}'
+ cut -c-1290
+ echo T: 'Bus=01' 'Lev=00' 'Prnt=00' 'Port=00' 'Cnt=00' 'Dev#=' 1 'Spd=480' 'MxCh=' 1 B: 'Alloc=' 0/800 us '(' '0%),' '#Int=' 0, '#Iso=' 0 D: 'Ver=' 2.00 'Cls=09(hub' ')' 'Sub=00' 'Prot=01' 'MxPS=64' '#Cfgs=' 1 P: 'Vendor=1d6b' 'ProdID=0002' 'Rev=' 6.06 S: 'Manufacturer=Linux' 6.6.48 xhci-hcd S: 'Product=xHCI' Host Controller S: 'SerialNumber=xhci-hcd.1.auto' 'C:*' '#Ifs=' 1 'Cfg#=' 1 'Atr=e0' 'MxPwr=' 0mA 'I:*' 'If#=' 0 'Alt=' 0 '#EPs=' 1 'Cls=09(hub' ')' 'Sub=00' 'Prot=00' 'Driver=hub' E: 'Ad=81(I)' 'Atr=03(Int.)' 'MxPS=' 4 'Ivl=256ms' T: 'Bus=01' 'Lev=01' 'Prnt=01' 'Port=00' 'Cnt=01' 'Dev#=' 2 'Spd=480' 'MxCh=' 0 D: 'Ver=' 2.10 'Cls=00(>ifc' ')' 'Sub=00' 'Prot=00' 'MxPS=64' '#Cfgs=' 1 P: 'Vendor=2cb7' 'ProdID=0104' 'Rev=' 4.14 S: 'Manufacturer=Fibocom' S: 'Product=Fibocom' Modem_SN:A43BDFCC S: 'SerialNumber=a43bdfcc' 'C:*' '#Ifs=' 5 'Cfg#=' 1 'Atr=a0' 'MxPwr=500mA' 'I:*' 'If#=' 0 'Alt=' 0 '#EPs=' 2 'Cls=ff(vend.)' 'Sub=ff' 'Prot=30' 'Driver=option' E: 'Ad=81(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=01(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 1 'Alt=' 0 '#EPs=' 3 'Cls=ff(vend.)' 'Sub=00' 'Prot=00' 'Driver=option' E: 'Ad=83(I)' 'Atr=03(Int.)' 'MxPS=' 10 'Ivl=32ms' E: 'Ad=82(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=02(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 2 'Alt=' 0 '#EPs=' 3 'Cls=ff(vend.)' 'Sub=00' 'Prot=00' 'Driver=option' E: 'Ad=85(I)' 'Atr=03(Int.)' 'MxPS=' 10 'Ivl=32ms' E: 'Ad=84(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=03(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 3 'Alt=' 0 '#EPs=' 3 'Cls=ff(vend.)' 'Sub=00' 'Prot=00' 'Driver=option' E: 'Ad=87(I)' 'Atr=03(Int.)' 'MxPS=' 10 'Ivl=32ms' E: 'Ad=86(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=04(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' 'I:*' 'If#=' 4 'Alt=' 0 '#EPs=' 3 'Cls=ff(vend.)' 'Sub=ff' 'Prot=ff' 'Driver=qmi_wwan' E: 'Ad=88(I)' 'Atr=03(Int.)' 'MxPS=' 8 'Ivl=32ms' E: 'Ad=8e(I)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' E: 'Ad=0f(O)' 'Atr=02(Bulk)' 'MxPS=' 512 'Ivl=0ms' T: 'Bus=02' 'Lev=00' 'Prnt=00' 'Port=00' 'Cnt=00' 'Dev#=' 1 'Spd=5000' 'MxCh=' 1 B: 'Alloc=' 0/800 us '(' '0%),' '#Int=' 0, '#Iso=' 0 D: 'Ver=' 3.00 'Cls=09(hub' ')' 'Sub=00' 'Prot=03' 'MxPS=' 9 '#Cfgs=' 1 P: 'Vendor=1d6b' 'ProdID=0003' 'Rev=' 6.06 S: 'Manufacturer=Linux' 6.6.48 xhci-hcd S: 'Product=xHCI' Host Controller S: 'SerialNumber=xhci-hcd.1.auto' 'C:*' '#Ifs=' 1 'Cfg#=' 1 'Atr=e0' 'MxPwr=' 0mA 'I:*' 'If#=' 0 'Alt=' 0 '#EPs=' 1 'Cls=09(hub' ')' 'Sub=00' 'Prot=00' 'Driver=hub' E: 'Ad=81(I)' 'Atr=03(Int.)' 'MxPS=' 4 'Ivl=256ms'
+ PVCUT=' Rev= 4.14 S: Manufacturer=Fibocom S: Product=Fibocom Modem_SN:A43BDFCC S: SerialNumber=a43bdfcc C:* #Ifs= 5 Cfg#= 1 Atr=a0 MxPwr=500mA I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=30 Driver=option E: Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 1 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option E: Ad=83(I) Atr=03(Int.) MxPS= 10 Ivl=32ms E: Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option E: Ad=85(I) Atr=03(Int.) MxPS= 10 Ivl=32ms E: Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=option E: Ad=87(I) Atr=03(Int.) MxPS= 10 Ivl=32ms E: Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E [... cut ...] =0ms E: Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms I:* If#= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan E: Ad=88(I) Atr=03(Int.) MxPS= 8 Ivl=32ms E: Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms E: Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms T: Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#= 1 Spd=5000 MxCh= 1 B: Alloc= 0/800 us ( 0%), #Int= 0, #Iso= 0 D: Ver= 3.00 Cls=09(hub ) Sub=00 Prot=03 MxPS= 9 #Cfgs= 1 P: Vendor=1d6b ProdID=0003 Rev= 6.06 S: ' 'Driver=qmi_wwan'

+ PROTO=qmi
+ cat
+ exit 0

jmyh88, 2024-09-04:
[image]

obsy, 2024-09-08:
Show us
```
sms_tool -d /dev/ttyUSB1 at "AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?"
sms_tool -d /dev/ttyUSB1 at "AT+GTCCINFO?;+GTCAINFO?"
```

jmyh88, 2024-09-08:
***@***.***:~# sms_tool -d /dev/ttyUSB1 at
"AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?"

+CPIN: READY

+CSQ: 99,99

+COPS: 0,0,"CHINA MOBILE",11

+COPS: 0,2,"46000",11

+CREG: 2,0

***@***.***:~# sms_tool -d /dev/ttyUSB1 at "AT+GTCCINFO?;+GTCAINFO?"

+GTCCINFO:
NR service cell:
1,9,460,00,14702C,,7B49E,35C,5041,100,111,71,71,65

obsy, 2024-09-08:
```
sms_tool -d /dev/ttyUSB1 at "at+C5GREG?"
```

jmyh88, 2024-09-09:
***@***.***:~# sms_tool -d /dev/ttyUSB1 at "at+C5GREG?"
 +C5GREG: 0,1

obsy, 2024-09-09:
Very strange. Again
```
sms_tool -d /dev/ttyUSB1 at "at+C5GREG=2;+C5GREG?"
```

jmyh88, 2024-09-09:
***@***.***:~# sms_tool -d /dev/ttyUSB1 at "at+C5GREG=2;+C5GREG?"

+C5GREG: 2,1,"14702C","71B3009",11,1,"01"

obsy, 2024-09-12:
Update script: https://github.com/obsy/packages/commit/6e46db60d611026129f7019b655702d700dd471d
@4IceG: please adapt it to your project

For 5G SA in FM150 RSSI is not displayed, only RSRP

jmyh88, 2024-09-12:
感谢回复，我需要怎么做了，下载替换文件吗？

jmyh88, 2024-09-12:
> 更新脚本：[obsy/packages@ 6e46db6](https://github.com/obsy/packages/commit/6e46db60d611026129f7019b655702d700dd471d) @4IceG：请根据您的项目进行调整
> 
> 对于 FM150 中的 5G SA，不显示 RSSI，仅显示 RSRP

Thanks for your reply, what should I do to download 2cb70104
Replace the file?

obsy, 2024-09-12:
No, it has to be rewritten for this specific project. Mine is different (not for luci)

jmyh88, 2024-09-12:
I don't quite understand how to operate it, or can you help compile the ipk
file for me to install it, thanks

4IceG, 2024-09-12:
@obsy Dzięki Cezary za info, wieczorem sprobuje to dodać.

4IceG, 2024-09-12:
@jmyh88 I don't know if it will work, but you can check it out.

Unpack [test.zip](https://github.com/user-attachments/files/16983963/test.zip) archive and copy files to `usr/share/3ginfo-lite/modem/usb`

jmyh88, 2024-09-12:
> @jmyh88我不知道它是否会起作用，但你可以检查一下。
> 
> 解压[test.zip](https://github.com/user-attachments/files/16984043/test.zip)档案并将文件复制到`usr/share/3ginfo-lite/modem/usb`。

After replacing, the file permissions are 0664, restart openwrt, and enter 3ginfo-lite again
SyntaxError Unexpected end of JSON input
Error: Unexpected end of JSON input
[image]

jmyh88, 2024-09-12:
> sms_tool -d /dev/ttyUSB1 at "AT+GTCCINFO?;+GTCAINFO?"

This is the ssh result after replacing the file
[image]

jmyh88, 2024-09-12:
> @jmyh88我不知道它是否会起作用，但你可以检查一下。
> 
> 解压[test.zip](https://github.com/user-attachments/files/16984043/test.zip)档案并将文件复制到`usr/share/3ginfo-lite/modem/usb`。

root@LibWrt:~# sh -x /usr/share/3ginfo-lite/3ginfo.sh
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
uci: Entry not found
+ DEVICE=/dev/ttyUSB1
+ '[' -z /dev/ttyUSB1 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB1 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ CONFIG=modemdefine
+ uci show modemdefine
+ grep -o '@modemdefine\[[0-9]*\]\.modem'
+ wc -l
+ xargs
uci: Entry not found
+ MODEMZ=0
+ '[[' 0 ]]
+ uci -q get 'modemdefine.@general[0].main_network'
+ SEC=
+ '[[' 0 '=' 0 ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=wan
+ '[[' 0 '=' 1 ]]
+ '[' -z wan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ ifstatus wan
+ grep '"up": true'
+ NETUP='       "up": true,'
+ '[' -n '      "up": true,' ]
+ uci -q -P /var/state/ get network.wan.connect_time
+ CT=
+ '[' -z ]
+ ifstatus wan
+ awk '-F[:,]' '/uptime/ {print $2}'
+ xargs
+ CT=1520
+ '[' '!' -z 1520 ]
+ expr 1520 / 60 / 60 / 24
+ D=0
+ expr 1520 / 60 / 60 '%' 24
+ H=0
+ expr 1520 / 60 '%' 60
+ M=25
+ expr 1520 '%' 60
+ S=20
+ printf '%dd, %02d:%02d:%02d' 0 0 25 20
+ CONN_TIME='0d, 00:25:20'
+ date '+%s'
+ date '+%Y%m%d%H%M%S' -d @1726177608
+ CONN_TIME_SINCE=20240913054648
+ ifstatus wan
+ awk '-F"' '/l3_device/ {print $4}'
+ IFACE=wwan0
+ '[' -n wwan0 ]
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$2}'
+ RX='1.0 MiB'
+ ifconfig wwan0
+ awk '-F[()]' '/bytes/ {printf "%s",$4}'
+ TX='6.2 MiB'
+ echo '
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=26
+ '[' x26 '=' x ]
+ '[' 26 -ge 0 -a 26 -le 31 ]
+ CSQ_PER=83
+ COPS=
+ COPS_MCC=
+ COPS_MNC=
+ echo '
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ awk '-F["]' '/^\+COPS:\s*.,2/ {print $2}'
+ COPS_NUM=46000
+ '[' -n 46000 ]
+ COPS_MCC=460
+ COPS_MNC=00
+ echo '
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ awk '-F["]' '/^\+COPS:\s*.,0/ {print $2}'
+ TCOPS='CHINA MOBILE'
+ '[' 'xCHINA MOBILE' '!=' x ]
+ COPS='CHINA MOBILE'
+ '[' -z 'CHINA MOBILE' ]
+ '[' -z 'CHINA MOBILE' ]
+ '[[' CHINA MOBILE '=~' ' ' ]]
sh: MOBILE: unknown operand
+ sms_tool -d /dev/ttyUSB1 at 'AT+COPS?'
+ sed -n 2p
+ cut -d '"' -f2
+ tr -d '\r'
+ isp=46000
+ isp_num='460 00'
+ isp_numws=46000
+ '[[' CHINA MOBILE '=~' '^[0-9]+$' ]]
sh: MOBILE: unknown operand
+ LOCATIONFILE=/tmp/location
+ '[' -e /tmp/location ]
+ touch /tmp/location
+ cat /tmp/location
+ LOC=China
+ '[' -n China ]
+ cat /tmp/location
+ LOC=China
+ '[[' China '==' - ]]
+ awk '-F[;]' '/^46000;/ {print $2}' /usr/share/3ginfo-lite/mccmnc.dat
+ LOC=China
+ '[' -n China ]
+ echo China
+ echo '
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ awk '-F[, ]' '/^\+CPIN:/ {print $0;exit}'
+ xargs
+ T='+CPIN: READY'
+ '[' -n '+CPIN: READY' ]
+ '[' '+CPIN: READY' '=' '+CPIN: READY' ]
+ echo '
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ awk '-F[, ]' '/^\+CME ERROR:/ {print $0;exit}'
+ T=
+ '[' -n  ]
+ echo '
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ awk '-F[,]' '/^\+CREG/ {gsub(/[[:space:]"]+/,"");printf "T=\"%d\";LAC_HEX=\"%X\";CID_HEX=\"%X\";LAC_DEC=\"%d\";CID_DEC=\"%d\";MODE_NUM=\"%d\"", $2, "0x"$3, "0x"$4, "0x"$3, "0x"$4, $5}'
+ eval 'T="0";LAC_HEX="0";CID_HEX="0";LAC_DEC="0";CID_DEC="0";MODE_NUM="0"'
+ T=0
+ LAC_HEX=0
+ CID_HEX=0
+ LAC_DEC=0
+ CID_DEC=0
+ MODE_NUM=0
+ REG=0
+ '[' -z 0 ]
+ '[' x0 '=' x0 ]
+ echo '
+CPIN: READY

+CSQ: 26,99

+COPS: 0,0,"CHINA MOBILE",7

+COPS: 0,2,"46000",7

+CREG: 2,0

'K
+ awk '-F[,]' '/^\+COPS/ {print $4;exit}'
' MODE_NUM='7
+ MODE=LTE
+ sms_tool -d /dev/ttyUSB1 at at+cereg
' OTX='
' echo '
+ awk '-F[,]' '/^\+CEREG/ {printf "%s", toupper($3)}'
+ sed 's/[^A-F0-9]//g'
+ TAC=
+ '[' x '!=' x ]
+ TAC=-
+ TAC_HEX=-
+ uci -q get '3ginfo.@3ginfo[0].device'
+ CONF_DEVICE=/dev/ttyUSB1
+ echo x/dev/ttyUSB1
+ grep -q 192.168.
+ '[' -e /usr/bin/sms_tool ]
+ REGOK=0
+ '[' x0 '=' x1 ]
+ '[' x0 '=' x5 ]
+ '[' x0 '=' x6 ]
+ '[' x0 '=' x7 ]
+ getdevicevendorproduct /dev/ttyUSB1
+ basename /dev/ttyUSB1
+ devname=ttyUSB1
+ readlink -f /sys/class/tty/ttyUSB1/device
+ devpath=/sys/devices/platform/soc@0/8af8800.usb/8a00000.usb/xhci-hcd.1.auto/usb1/1-1/1-1:1.1/ttyUSB1
+ T=/sys/devices/platform/soc@0/8af8800.usb/8a00000.usb/xhci-hcd.1.auto/usb1/1-1
+ cat /sys/devices/platform/soc@0/8af8800.usb/8a00000.usb/xhci-hcd.1.auto/usb1/1-1/idVendor
+ cat /sys/devices/platform/soc@0/8af8800.usb/8a00000.usb/xhci-hcd.1.auto/usb1/1-1/idProduct
+ echo usb/2cb70104
+ VIDPID=usb/2cb70104
+ '[' -e /usr/share/3ginfo-lite/modem/usb/2cb70104 ]
+ cat /tmp/sysinfo/board_name
+ . /usr/share/3ginfo-lite/modem/usb/2cb70104
+ sms_tool -d /dev/ttyUSB1 at 'AT+CGMM?'
+ O='
+CGMM: "FM150-AE","FM150"
'
+ echo '
+CGMM: "FM150-AE","FM150"
'
+ grep -q FM150
+ . /usr/share/3ginfo-lite/modem/usb/2cb70104.fm150
' '
: not found3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/2cb70104.fm150: line 2: 
+ sms_tool -d /dev/ttyUSB1 at 'AT+MTSM=1'
+ O='
+MTSM: 37.5
'
+ echo '
+MTSM: 37.5
'
+ awk '/\+MTSM:/{print $2}'
' T='37.5
' ][' -n '37.5
'&deg;C'37.5
' '
: not found3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/2cb70104.fm150: line 7: 
/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/2cb70104.fm150: line 17: syntax error: unexpected word (expecting "in")
root@LibWrt:~#