4IceG/luci-app-3ginfo-lite #96: TOZED S12 PRO with Tozed LT70-C V1.0.1-B7 issue

anngit1, 2024-09-08:
When I tried 3ginfo lite on TOZED S12 PRO with Tozed LT70-C V1.0.1-B7, I get following errors

SyntaxError
Bad control character in string literal in JSON at position 446 (line 19 column 18)

Error: Bad control character in string literal in JSON at position 446 (line 19 column 18)

I debug log near "registration":"0 seems to have a new line. that might be the issue. 

Thumbs up for your work :)
[image]

Following is debug log.

`{
"conn_time":"0d, 01:14:31",
"conn_time_sec":"4471",
"conn_time_since":"20240908061517",
"rx":"379.5 MiB",
"tx":"34.1 MiB",
"modem":"Tozed LT70-C V1.0.1-B7",
"mtemp":"",
"firmware":"SL8563-21.36.5-14-DBG-C1BD04B16-61CAA5BB-NCM4-B34C8C1E6CC810D6-61DBFC74_LT70-GA",
"cport":"/dev/ttyUSB0",
"protocol":"ncm",
"csq":"9",
"signal":"29",
"operator_name":"",
"operator_mcc":"413",
"operator_mnc":"12",
"location":"",
"mode":"LTE / ",
"registration":"0
",
"simslot":"",
"imei":"867792051014359",
"imsi":"",
"iccid":"89941202101171109623",
"lac_dec":"",
"lac_hex":"",
"tac_dec":"",
"tac_hex":"-",
"tac_h":"",
"tac_d":"",
"cid_dec":"0",
"cid_hex":"",
"pci":"408",
"earfcn":"39990",
"pband":"41",
"s1band":"",
"s1pci":"",
"s1earfcn":"",
"s2band":"",
"s2pci":"",
"s2earfcn":"",
"s3band":"",
"s3pci":"",
"s3earfcn":"",
"s4band":"",
"s4pci":"",
"s4earfcn":"",
"rsrp":"-93",
"rsrq":"-6",
"rssi":"-73",
"sinr":"17"
}
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB0
+ '[' -z /dev/ttyUSB0 ]
[... 321 lines cut ...]
+ RSRQ=-6
+ echo -93,-6,0,17
+ cut -d, -f4
+ SINR=17
+ echo 41,0-39990,0-408,0--9356,0--693,0--12600,0-1,0-5,255-0,0--32647010,0-20242,0-7,0-0,0-0,0,0-0-0-0-0-0-0-0-0-0--9387,0--9356,0-39990,0-1,0-1280,0-2147483647,0-43822,0--637,0--693,0-0
+ cut -d, -f2
+ cut -d- -f1
+ CA_BAND=0
+ '[' 0 -ne 0 ]
+ echo 41,0-39990,0-408,0--9356,0--693,0--12600,0-1,0-5,255-0,0--32647010,0-20242,0-7,0-0,0-0,0,0-0-0-0-0-0-0-0-0-0--9387,0--9356,0-39990,0-1,0-1280,0-2147483647,0-43822,0--637,0--693,0-0
+ cut -d, -f4
+ cut -d- -f1
+ CA_PCI=0
+ '[' 0 -ne 0 ]
+ echo 41,0-39990,0-408,0--9356,0--693,0--12600,0-1,0-5,255-0,0--32647010,0-20242,0-7,0-0,0-0,0,0-0-0-0-0-0-0-0-0-0--9387,0--9356,0-39990,0-1,0-1280,0-2147483647,0-43822,0--637,0--693,0-0
+ cut -d, -f3
+ cut -d- -f1
+ CA_EARFCN=0
+ '[' 0 -ne 0 ]
+ '[' 0 -ne 0 ]
+ band 41
/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/17824055: line 58: band: not found
+ MODE='LTE / '
+ cat
+ exit 0`

anngit1, 2024-09-08:
This is the debug log in text file

[3ginfo debuglog.txt](https://github.com/user-attachments/files/16921311/3ginfo.debuglog.txt)

4IceG, 2024-09-08:
Hi @anngit1 ,
I don't remember adding support for these USB modems.

The newline error can be resolved by adding "| xargs" to data cut commands.

```
O=$(sms_tool -d $DEVICE at "AT+CREG?;+CCID;+ZRSSI?;+CSQ")
DATA=$(echo "$O" | cut code | xargs)
```
I am more curious why the modem returns `+CREG: 2.0`, which means that it is not logged into network?
Operator name is also missing, the commands `AT+COPS=3,0;+COPS?;` return nothing?

3ginfo-lite checks if the modem is logged into the network and only then allows it to show data.

anngit1, 2024-09-08:
Modem is connected to LTE and receives data. But no CA though. I think roaming is also disabled. but I couldn't find AT command to enable roaming and CA. I'm trying to get your script working as it provides lot of info.

I receive following data to  AT+CREG?;+CCID;+ZRSSI?;+CSQ

+CREG: 2,0
+CCID: "89941202101171109623"
+ZRSSI: -95,-8,0,23
+CSQ: 9,99

I could not figure out how to include your suggestion. does it go into 3ginfo.sh?

Im using attached modem script. (I added txt extension so I can upload it here.

[17824055.txt](https://github.com/user-attachments/files/16922411/17824055.txt)
)

4IceG, 2024-09-08:
@anngit1 Check this file [17824055 v2.txt](https://github.com/user-attachments/files/16922746/17824055.v2.txt).

Which version of 3ginfo-lite are you using? From your chaotic script I can see that you did not follow the scripts from other modems.

You don't need to repeat the `AT+CREG?` and `AT+CSQ` commands (3ginfo-lite already has these commands in scripts).

Update 3ginfo-lite to the latest version and replace the contents of the file `luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/mccmnc.dat` with the contents of the file from github (I made an update for your location).

anngit1, 2024-09-08:
Im on the latest version 3ginfo-lite (1.0.74-20240827)

I did replace mccmnc.dat and used updated modem script you sent. but still no response (Same picture as I posted in last message.)

Following is new debug log.
[debuglog3.txt](https://github.com/user-attachments/files/16922890/debuglog3.txt)

Thank you very much for your time and effort trying to help.

4IceG, 2024-09-08:
@anngit1 OK, check this file [17824055 v3.txt](https://github.com/user-attachments/files/16922940/17824055.v3.txt).

anngit1, 2024-09-08:
Still same result as in last image I posted.

4IceG, 2024-09-08:
@anngit1 Clear browser cache and show Debug result.

anngit1, 2024-09-08:
done. Here is the log. 

[debuglog4_clearedBrowsercache.txt](https://github.com/user-attachments/files/16922966/debuglog4_clearedBrowsercache.txt)

4IceG, 2024-09-08:
@anngit1 Upload again because file does not open properly.

anngit1, 2024-09-08:
Here it is.
[debuglog4_cachecl.txt](https://github.com/user-attachments/files/16922977/debuglog4_cachecl.txt)

4IceG, 2024-09-08:
Invention from this modem..

@anngit1 OK, check this file [17824055 v4.txt](https://github.com/user-attachments/files/16923045/17824055.v4.txt)
My last attempt.

anngit1, 2024-09-08:
Thanks. This didn't work either. same result as my last image.

I noticed that when I issue following command I get response.

sms_tool -D -d /dev/ttyUSB0 at 'AT+CPIN?'

But when I issue similar command with 2 AT commands like follow, it give me +CME ERROR: 3 

sms_tool -D -d /dev/ttyUSB0 at 'AT+CPIN?;+CSQ'

I get same +CME ERROR: 3 for following command extracted from debug log.

sms_tool -D -d /dev/ttyUSB0 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'

Thanks for trying to help. I will keep trying and post here if I managed to fix the issue.

Following is last debug log and above responses from SSH
[debuglog5.txt](https://github.com/user-attachments/files/16923088/debuglog5.txt)
 
[image]

anngit1, 2024-09-09:
I fixed the modem file for the most part. operator name doesn't work yet. So I hardcoded it for now.

[17824055.txt](https://github.com/user-attachments/files/16935318/17824055.txt)

anngit1, 2024-09-10:
Fully working modem template file for Tozed LT70-C cat6 modem. Working as of luci-app-3ginfo-lite version 1.0.74-20240827. Tested on TOZED S12 PRO (on 2024 Sep 10).

(Added txt extension to the file to upload it here.)
[17824055.txt](https://github.com/user-attachments/files/16944819/17824055.txt)

4IceG, 2024-09-10:
@anngit1 Great, glad it's working now.

Can you show debug? Localization is also OK?

anngit1, 2024-09-10:
Here it is.

[DebugLogv6_workingmodemscript.txt](https://github.com/user-attachments/files/16947408/DebugLogv6_workingmodemscript.txt)

(On other note, I think upload and download volume arrow heads in primary information page is exchanged.)

4IceG, 2024-09-10:
Did you maybe replace file [mccmnc.dat](https://github.com/4IceG/luci-app-3ginfo-lite/blob/main/luci-app-3ginfo-lite/root/usr/share/3ginfo-lite/mccmnc.dat)? I'm asking in the context of the missing "location".

[image]

anngit1, 2024-09-10:
Yes I replaced mccmnc.dat. checked again now too.

Following are up down volumes I get.
[image]
[image]

anngit1, 2024-10-06:
@1Conan Thanks for your awesome work.  

Little background of my network and some issue I noticed:
I'm on SLT 4G broadband network in Sri Lanka for internet only. The company name is SLT mobitel and they have voice network for mobile phones separately. SLT 4G is mainly for 4G home broadband. [but they do offer fixed line telephones connected to 4G modem too.](https://slt.lk/en/personal/broadband/lte/new-connection-charges)

I found that the SLT 4G network I'm using (for modems I used, LT70-C and RG500Q-EA) returns +CREG: 2,3 for AT+CREG? command. This says, "Registration denied". However, for AT+CEREG? my RG500Q-EA modem returns +CEREG: 2,1,"AB2E","4F1204",7 meaning I'm attached to 4G network.

My internet connection works fine with both modems, even though both modems return "Registration denied" for AT+CREG? command.

So, it seems like for me to see whether I'm attached to the network, I have to use AT+CEREG? command. (LTE version of command) and AT+CREG? doesn't work ("Registration denied") because I'm not subscribed for voice.

So, may be these scripts need to check network connection with both commands AT+CEREG? and AT+CREG? and if one of them says registered, we can consider as successful network registration.

Again, thanks for all of your work @1Conan @4IceG

VKNemsara, 2026-09-06:
> Fully working modem template file for Tozed LT70-C cat6 modem. Working as of luci-app-3ginfo-lite version 1.0.74-20240827. Tested on TOZED S12 PRO (on 2024 Sep 10). Tested on Sri Lanka, SLTMobitel SLT 4G home broadband network.
> 
> This modem script needs coreutils-base64 package to acquire operator name from modem.
> 
> (Added txt extension to the file to upload it here.) [17824055.txt](https://github.com/user-attachments/files/16944819/17824055.txt)

Hi @anngit1 ,

My S12 pro is from Dialog but have an Airtel SIM. I downloaded your file from here and placed it at /usr/share/3ginfo-lite/modem/usb/17824055. But the following error is there. Nothing loads.

[image]

coreutils-base64 is also installed

[image]