4IceG/luci-app-3ginfo-lite #98: No 3G RSSI on MF286C

victor186, 2024-09-19:
This only show's CSQ on UMTS mode.
But the uqmi show dBm level
[image]
[image]

obsy, 2024-09-19:
MF286C Is not supported by this program. Provide commands that support the modem - find documentation or check if commands from other ZTE routers work.

victor186, 2024-09-25:
> MF286C Is not supported by this program. Provide commands that support the modem - find documentation or check if commands from other ZTE routers work.

Can't show the signal reported by uqmi ?

[image]