4IceG/luci-app-3ginfo-lite #102: JSON.parse: bad control character in string literal at line 18 column 17 of the JSON data

matrock47, 2024-10-30:
I Try compile openwrt 23.05.5 unfortunately there is an error when clicking information 4g/5g with following error. How to fix this? 

DEVICE ZBT Z8102AX M2
JSON.parse: bad control character in string literal at line 18 column 17 of the JSON data

i didnt face this error on wg1608-16, the issue only occured when compile z8102ax m2 - t.

matrock47, 2024-10-30:
[image]

matrock47, 2024-10-30:
MODULE FIBOCOM L850-GL

Ulfila80, 2024-12-10:
HI! Have the same problem.

Analysis of the file output /usr/shared/3ginfo-lite/3ginfo.sh showed that variables EARFCN, SINR may contain a carriage return character (\r). On some routers this corrupts the json output. I added (tr -d '\r') command to the script to remove the \r character and everything works fine.

```
cat <<EOF
{
"conn_time":"$CONN_TIME",
"conn_time_sec":"$CT",
"conn_time_since":"$CONN_TIME_SINCE",
"rx":"$RX",
"tx":"$TX",
"modem":"$MODEL",
"mtemp":"$TEMP",
"firmware":"$FW",
"cport":"$DEVICE",
"protocol":"$PROTO",
"csq":"$CSQ",
"signal":"$CSQ_PER",
"operator_name":"$COPS",
"operator_mcc":"$COPS_MCC",
"operator_mnc":"$COPS_MNC",
"location":"$LOC",
"mode":"$MODE",
"registration":"$REG",
"simslot":"$SSIM",
"imei":"$NR_IMEI",
"imsi":"$NR_IMSI",
"iccid":"$NR_ICCID",
"lac_dec":"$LAC_DEC",
"lac_hex":"$LAC_HEX",
"tac_dec":"$TAC_DEC",
"tac_hex":"$TAC_HEX",
"tac_h":"$T_HEX",
"tac_d":"$T_DEC",
"cid_dec":"$CID_DEC",
"cid_hex":"$CID_HEX",
"pci":"$PCI",
"earfcn":"$(echo $EARFCN | tr -d '\r')",
"pband":"$PBAND",
"s1band":"$S1BAND",
"s1pci":"$S1PCI",
"s1earfcn":"$(echo $S1EARFCN | tr -d '\r')",
"s2band":"$S2BAND",
"s2pci":"$S2PCI",
"s2earfcn":"$(echo $S2EARFCN | tr -d '\r')",
"s3band":"$S3BAND",
"s3pci":"$S3PCI",
"s3earfcn":"$(echo $S3EARFCN | tr -d '\r')",
"s4band":"$S4BAND",
"s4pci":"$S4PCI",
"s4earfcn":"$(echo $S4EARFCN | tr -d '\r')",
"rsrp":"$RSRP",
"rsrq":"$RSRQ",
"rssi":"$RSSI",
"sinr":"$(echo $SINR | tr -d '\r')"
}
EOF

```

SuperMitic, 2025-04-07:
After updating to the latest snapshot I was getting the same issue "JSON bad control character". 
Examining the output of the _3ginfo.sh_ script I found that the TEMP variable contained the carriage return character. So I manually edited the command used to retrieve the TEMP in the modem/usb/modem_ID file and added a pipe to `sed 's/\r//g'`.

Avarem, 2025-04-12:
/usr/shared/3ginfo-lite/3ginfo.sh

``# Function for clearing string values (without escaping quotes)
sanitize_string() {
    [ -z "$1" ] && echo "-" || echo "$1" | tr -d '\r\n'
}

# Function for numeric values (leaves as is if number)
sanitize_number() {
    [ -z "$1" ] && echo "0" || echo "$1"
}

cat <<EOF
{
"conn_time":"$(sanitize_string "$CONN_TIME")",
"conn_time_sec":"$(sanitize_number "$CT")",
"conn_time_since":"$(sanitize_string "$CONN_TIME_SINCE")",
"rx":"$(sanitize_number "$RX")",
"tx":"$(sanitize_number "$TX")",
"modem":"$(sanitize_string "$MODEL")",
"mtemp":"$(sanitize_string "$TEMP")",
"firmware":"$(sanitize_string "$FW")",
"cport":"$(sanitize_string "$DEVICE")",
"protocol":"$(sanitize_string "$PROTO")",
"csq":"$(sanitize_number "$CSQ")",
"signal":"$(sanitize_number "$CSQ_PER")",
"operator_name":"$(sanitize_string "$COPS")",
"operator_mcc":"$(sanitize_string "$COPS_MCC")",
"operator_mnc":"$(sanitize_string "$COPS_MNC")",
"location":"$(sanitize_string "$LOC")",
"mode":"$(sanitize_string "$MODE")",
"registration":"$(sanitize_string "$REG")",
"simslot":"$(sanitize_string "$SSIM")",
"imei":"$(sanitize_string "$NR_IMEI")",
"imsi":"$(sanitize_string "$NR_IMSI")",
"iccid":"$(sanitize_string "$NR_ICCID")",
"lac_dec":"$(sanitize_number "$LAC_DEC")",
"lac_hex":"$(sanitize_string "$LAC_HEX")",
"tac_dec":"$(sanitize_number "$TAC_DEC")",
"tac_hex":"$(sanitize_string "$TAC_HEX")",
"tac_h":"$(sanitize_string "$T_HEX")",
"tac_d":"$(sanitize_number "$T_DEC")",
"cid_dec":"$(sanitize_number "$CID_DEC")",
"cid_hex":"$(sanitize_string "$CID_HEX")",
"pci":"$(sanitize_number "$PCI")",
"earfcn":"$(sanitize_number "$EARFCN")",
"pband":"$(sanitize_string "$PBAND")",
"s1band":"$(sanitize_string "$S1BAND")",
"s1pci":"$(sanitize_number "$S1PCI")",
"s1earfcn":"$(sanitize_number "$S1EARFCN")",
"s2band":"$(sanitize_string "$S2BAND")",
"s2pci":"$(sanitize_number "$S2PCI")",
"s2earfcn":"$(sanitize_number "$S2EARFCN")",
"s3band":"$(sanitize_string "$S3BAND")",
"s3pci":"$(sanitize_number "$S3PCI")",
"s3earfcn":"$(sanitize_number "$S3EARFCN")",
"s4band":"$(sanitize_string "$S4BAND")",
"s4pci":"$(sanitize_number "$S4PCI")",
"s4earfcn":"$(sanitize_number "$S4EARFCN")",
"rsrp":"$(sanitize_number "$RSRP")",
"rsrq":"$(sanitize_number "$RSRQ")",
"rssi":"$(sanitize_number "$RSSI")",
"sinr":"$(sanitize_number "$SINR")"
}
EOF
exit 0
``