4IceG/luci-app-3ginfo-lite #105: The dreaded switch to APK packages

f1vefour, 2024-11-20:
Will you be creating APK packages for your modem extras? As it stands your packages are the only ones I use outside of the standard included packages and I can't enable APK manager and can't use opkg any longer on my nightly builds.

Not your responsibility just wondering if you will be creating APK packages.

4IceG, 2024-11-20:
Please be patient…

I'm planning an apk repository, I just need to know how to create it.

KFOq, 2024-11-21:
Thank you @4IceG  for your efforts 

related topic for apk tool by OpenWRT 
https://github.com/openwrt/packages/issues/23706

apk tool :
https://gitlab.alpinelinux.org/alpine/apk-tools

4IceG, 2024-11-21:
@KFOq I am not happy with the APK implementation at the moment, because there is a lot of confusion and lack of good documentation (I have to learn everything from scratch at the moment).

I hope I will be able to solve the repository issue relatively quickly.

closed 2025-05-16 by 4IceG